espruino/Espruino #1041: NRF51/52 - Alpha - very early in development (GPIO, but not peripherals, no sleep or save)

OwenBrotherwood, 2017-01-29:
I am not sure if the README.md is correct and I have to find out what sleep or save means in context
[image]

gfwilliams, 2017-01-30:
The readme needs updating - the nRF52 port is fine now

OwenBrotherwood, 2017-01-30:
My take at making an Espurino/docs  @ https://github.com/OwenBrotherwood/Espruino/tree/doc
In the first instance, I will use the information to aid myself in connection with the https://github.com/OwenBrotherwood/Espruino/blob/doc/docs/NRF52832DK.md and when I get the puck.js make the same for that and probably make a doc for general NRF52832.md otherwise there will be a lot of repeats in board information that would include processor information.
About a week and a PR, I ping again when I have a reasonable documentation for making developement easy
I am unsure as to how information in the Espruino repo relates to EspruinoDocs and the Espruino web site: there could be many overlaps.

In general, if the board is not Espruino, the community should maintain the information in docs for boards: therefore the subject Activity in the md file for NRF52832DK.md

gfwilliams, 2017-01-30:
Thanks - however please could you put the docs with the ones on other boards - they're all here: https://github.com/espruino/EspruinoDocs/tree/master/boards

Which appear on the Espruino site, for instance:
http://www.espruino.com/Puck.js
http://www.espruino.com/EspruinoESP8266
http://www.espruino.com/MicroBit
 
There are even ways of auto-making the pin diagrams for all the boards from the BOARD.py file (see the other `.py` files for examples)

And if you want to put info on building, stick it here: https://github.com/espruino/Espruino/blob/master/README_Building.md

OwenBrotherwood, 2017-01-30:
will do 
the info on placement was what I needed, thanks.

commit bf9a3508 (2017-02-03): Update README.md

Update status of nRF devkits, fix #1041
closed 2017-02-03 by gfwilliams