espruino/Espruino #1050: Espruino on ESP32 resets again and again, HELP!

aifer2007, 2017-02-05:
reference to:
1) https://github.com/espruino/Espruino/blob/ESP32/targets/esp32/docs/README.md
2) https://github.com/espruino/Espruino/blob/ESP32/targets/esp32/docs/ESP32_and_Git.md

I follow the steps:

$ mkdir esp32
$ cd esp32
$ git clone --recursive https://github.com/espressif/esp-idf.git
$ cd esp-idf
$ git submodule update --init
$ cd ..
$ git clone https://github.com/espressif/esp-idf-template.git template
$ nano setenv.sh
>>>
#!/bin/bash
export ESP_IDF_PATH=$(pwd)/esp-idf
export IDF_PATH=$(pwd)/esp-idf
export ESP_APP_TEMPLATE_PATH=$(pwd)/template
export ESP32=1
[[ ":$PATH:" != *":/opt/xtensa-esp32-elf/bin:"* ]] && PATH="/opt/xtensa-esp32-elf/bin:${PATH}"
<<<
$ chmod u+x setenv.sh
$ source ./setenv.sh
$ cd template
$ mkdir components
$ cd components
$ git clone https://github.com/espressif/arduino-esp32.git
$ nano arduino-esp32/cores/esp32/main.cpp

>>> Comment out app_main
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "Arduino.h"

#if CONFIG_AUTOSTART_ARDUINO

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
        loop();
    }
}
/*
extern "C" void app_main()
{
    initArduino();
    xTaskCreatePinnedToCore(loopTask, "loopTask", 4096, NULL, 1, NULL, 1);
}
*/
#endif
<<<

$ cd ..
$ make menuconfig
>>>Change some of the settings necessary for Espruino:
Component config -> LWIP -> Enable SO_REUSEADDR option [Enable]
Component config -> ESP32-specific config ->  Task watchdog [Disable]
>>>
$ make
$ cd ..
$ git clone https://github.com/MYGITHUBID/Espruino.git
$ cd Espruino
$ git checkout ESP32
$ git remote add upstream https://github.com/espruino/Espruino.git
$ git remote -v
>>>see something like this
origin  https://github.com/MYGITHUBID/Espruino.git (fetch)
origin  https://github.com/MYGITHUBID/Espruino.git (push)
upstream https://github.com/espruino/Espruino.git (fetch)
upstream https://github.com/espruino/Espruino.git (push)
>>>
$ git pull upstream ESP32
$ make
$ nano Makefile
>>> edit the flash label, change --port settings to my usb-uart port
flash:
        python $(ESP_IDF_PATH)/components/esptool_py/esptool/esptool.py \
        --chip esp32 \
        --port "/dev/tty.SLAB_USBtoUART" \
        --baud 921600 \
        write_flash \
        -z \
        --flash_mode "dio" \
        --flash_freq "40m" \
        0x1000 $(ESP_APP_TEMPLATE_PATH)/build/bootloader/bootloader.bin \
        0x10000 espruino_esp32.bin \
        0x4000 $(ESP_APP_TEMPLATE_PATH)/build/partitions_singleapp.bin
>>>
$ make flash
>>> result:
python ~/HomeDir/esp32/esp-idf/components/esptool_py/esptool/esptool.py \
	--chip esp32 \
	--port "/dev/tty.SLAB_USBtoUART" \
	--baud 921600 \
	write_flash \
	-z \
	--flash_mode "dio" \
	--flash_freq "40m" \
	0x1000 ~/HomeDir/esp32/template/build/bootloader/bootloader.bin \
	0x10000 espruino_esp32.bin \
	0x4000 ~/HomeDir/esp32/template/build/partitions_singleapp.bin
esptool.py v2.0-beta1
Connecting........................
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Attaching SPI flash...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 9504 bytes to 5655...
Wrote 9504 bytes (5655 compressed) at 0x00001000 in 0.1 seconds (effective 883.9 kbit/s)...
Hash of data verified.
Compressed 858288 bytes to 494430...
Wrote 858288 bytes (494430 compressed) at 0x00010000 in 9.0 seconds (effective 766.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00004000 in 0.0 seconds (effective 8492.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

It seems that everything is OK!

But, when I connect the ESP32 module in EsrpuinoWebIDE,

>>>
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2388
load:0x40078000,len:6788
load:0x40080000,len:252
entry 0x40080034
;32mI (1343) heap_alloc_caps: Initializing. RAM available for dynamic allocation:m
;32mI (1343) heap_alloc_caps: At 3FFCB5A8 len 00014A58 (82 KiB): DRAMm
;32mI (1353) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAMm
;32mI (1375) heap_alloc_caps: At 4009E3A0 len 00001C60 (7 KiB): IRAMm
;32mI (1395) cpu_start: Pro cpu up.m
;32mI (1407) cpu_start: Single core modem
;32mI (1420) cpu_start: Pro cpu start user codem
;32mI (1662) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0m
;32mI (1818) cpu_start: Starting scheduler on PRO CPU.m
timer EspruinoTimer : 0.0
timer 1 free
Untested FreeRTOS function ulTaskNotifyTake
/Users/aifer/Documents/DevSpace/esp-sdk/esp32/esp-idf/components/freertos/./tasks.c:4431 (ulTaskNotifyTake)- assert failed!
abort() was called at PC 0x40085d26
Guru Meditation Error: Core  0 panic'ed (abort)
Backtrace: 0x40008155:0x3ffd3710 0x40007d16:0x3ffd3730 0x40116531:0x3ffd3750 0x40131c30:0x3ffd3770
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2388
load:0x40078000,len:6788
load:0x40080000,len:252
entry 0x40080034
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2388
load:0x40078000,len:6788
load:0x40080000,len:252
entry 0x40080034
;32mI (1342) heap_alloc_caps: Initializing. RAM available for dynamic allocation:m
;32mI (1342) heap_alloc_caps: At 3FFCB5A8 len 00014A58 (82 KiB): DRAMm
;32mI (1352) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAMm
;32mI (1374) heap_alloc_caps: At 4009E3A0 len 00001C60 (7 KiB): IRAMm
;32mI (1394) cpu_start: Pro cpu up.m
;32mI (1406) cpu_start: Single core modem
;32mI (1419) cpu_start: Pro cpu start user codem
;32mI (1658) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0m
;32mI (2745) cpu_start: Starting scheduler on PRO CPU.m
timer EspruinoTimer : 0.0
timer 1 free
Untested FreeRTOS function ulTaskNotifyTake
/Users/aifer/Documents/DevSpace/esp-sdk/esp32/esp-idf/components/freertos/./tasks.c:4431 (ulTaskNotifyTake)- assert failed!
abort() was called at PC 0x40085d26
Guru Meditation Error: Core  0 panic'ed (abort)
Backtrace: 0x40008155:0x3ffd3730 0x40007d16:0x3ffd3750 0x40116531:0x3ffd3770 0x40131c30:0x3ffd3790
......

It seems that the ESP32 module resets again and again!

HELP!

wilberforce, 2017-02-07:
Update this
`0x4000 $(ESP_APP_TEMPLATE_PATH)/build/partitions_singleapp.bin`

To 0x8000

The Partition location was updated, it looks like the Makefile needs updating too.

You should not need to edit the Makefile to set the port - its read from an environment variable..

Also the docs are not quite current, we no longer are using the arduino libs.

I have been putting the build into a set of scripts here:

https://github.com/wilberforce/espruino-esp32

Join the chat in espruino/esp32
https://gitter.im/espruino/esp32

There are only 2 of us working on the esp32 port, so most of the dialogue is on the gitter channel.

FYI @jumjum123

wilberforce, 2017-02-07:
Oh .... saw this...

Untested FreeRTOS function ulTaskNotifyTake

You need to run:
`make menuconfig`

And under esp specific there is a RTOS menu.... there is an option there to stop on an untested function... that needs to be turned off and a make of the template folder again

jumjum123, 2017-02-07:
Wrong adress in https://github.com/espruino/Espruino/blob/ESP32/targets/esp32/docs/Install.md is changed.
Please see https://github.com/espruino/Espruino/blob/ESP32/targets/esp32/docs/README.md
Description of installation for Arduino is only mentioned for those, who want to use functions which are not yet available in esp-idf. Espruino does not use those anymore.
Please have in mind, ESP32 port of Espruino is under heavy construction (by 2 people, whenever their family gives ok ;-)

closed 2017-03-01 by wilberforce