espruino/Espruino #1051: `NRF.setScan(callback)` crashes on nRF51822

urish, 2017-02-06:
It seems to crash on this line:

`JsVar *s = jsvVarPrintf("%04x", *(uint16_t*)&dPtr[i+2]);` 

[View in context](https://github.com/urish/Espruino/blob/ecf0017b695a4c088bd812dac7c60316191543c1/libs/bluetooth/jswrap_bluetooth.c#L934)

Here is the stack trace, register values + dump of what it was trying to parse:

```
(gdb) where
#0  0x000340c6 in HardFault_Handler ()
#1  <signal handler called>
#2  jswrap_nrf_bluetooth_setScan_cb (callback=0x1, adv=<optimized out>) at libs/bluetooth/jswrap_bluetooth.c:934
#3  0x00027b90 in jsnCallFunction (function=0x228c1 <jswrap_nrf_bluetooth_setScan_cb>,
    argumentSpecifier=JSWAT_FINISH, thisParam=<optimized out>, paramData=0x20003f30, paramCount=1)
    at src/jsnative.c:210
#4  0x00021436 in jspeFunctionCall (function=<optimized out>, functionName=<optimized out>, thisArg=<optimized out>,
    isParsing=<optimized out>, argCount=<optimized out>, argPtr=<optimized out>) at src/jsparse.c:618
#5  0x000329f8 in jspExecuteFunction (func=0x20002408 <jsVars+276>, thisArg=0x200022f4 <jsVars>, argCount=1,
    argPtr=0x20003f30) at src/jsparse.c:2769
#6  0x0001f064 in jsiExecuteEventCallback (thisVar=0x200022f4 <jsVars>, callbackVar=<optimized out>, argCount=1,
    argPtr=0x20003f30) at src/jsinteractive.c:1637
#7  0x0001f15a in jsiExecuteEventCallbackArgsArray (thisVar=0x200024b0 <jsVars+444>, callbackVar=0x200022f4 <jsVars>,
    argsArray=<optimized out>) at src/jsinteractive.c:1616
#8  0x0001d260 in jsiExecuteEvents () at src/jsinteractive.c:1592
#9  jsiIdle () at src/jsinteractive.c:2029
#10 jsiLoop () at src/jsinteractive.c:2110
#11 main () at targets/nrf5x/main.c:32

(gdb) info registers
r0             0x229ea  141802
r1             0x31b43  203587
r2             0x1b000  110592
r3             0x340c7  213191
r4             0x20003cd8       536886488
r5             0x3      3
r6             0x20002480       536880256
r7             0x20003cf8       536886520
r8             0xffffffff       -1
r9             0xffffffff       -1
r10            0x1fff4000       536821760
r11            0x0      0
r12            0x0      0
sp             0x20003cb8       0x20003cb8
lr             0xfffffff9       -7
pc             0x340c6  0x340c6 <HardFault_Handler>
xPSR           0x21000003       553648131

(gdb) print dPtr
$16 = 0x20003cd8 "\002\001\006\003\003x\376\021\026x\376 "

(gdb) print i
$17 = 3

(gdb) print *(uint16_t*)&dPtr[i+2]
$19 = 65194
```

gfwilliams, 2017-02-06:
Hmm - interesting. I just committed a change that might help... Not sure - but it could be to do with 'varargs' expecting a 32 bit value and only actually getting a 16 bit one. Strange that it works on other boards but it's possible that when I've been scanning I haven't actually found anything advertising a 16 bit UUID so that code never got tested?

gfwilliams, 2017-02-06:
Would be interesting to see if it fixes it for you?

urish, 2017-02-06:
I am checking right now

urish, 2017-02-06:
Seems like it still crashes on this line:
`unsigned int uuid = *(uint16_t*)&dPtr[i+2];`

I asked GDB for the disassembly, this is what I got:

```
(gdb) frame
#2  jswrap_nrf_bluetooth_setScan_cb (callback=0x1, adv=<optimized out>) at libs/bluetooth/jswrap_bluetooth.c:934
934     in libs/bluetooth/jswrap_bluetooth.c
=> 0x000229ea <jswrap_nrf_bluetooth_setScan_cb+298>:    51 88   ldrh    r1, [r2, #2]
(gdb) print $r1
$3 = 1
(gdb) print $r2
$4 = 536886491
(gdb) print (char*)$r2
$5 = 0x20003cdb "\003\003x\376\f\026x\376\020\376\002urish\b\024"
(gdb)
```

any idea how to investigate further?

urish, 2017-02-06:
Furthermore, if I change the line to read:

`unsigned int uuid = 0x8080;`

it works perfectly - no crash, I get the scan response in JS (albeit all the services appear as "8080")

gfwilliams, 2017-02-06:
Really strange - could the issue be that it's an unaligned read?

What about:

```
unsigned int uuid = dPtr[i+2] | (dPtr[i+3]<<8);
```

I can't remember - it's totally possible I have the endianness wrong though.

urish, 2017-02-06:
Seems like this could be the case, quoting from [the docs](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52%2Fdita%2Fnrf52%2Fmigration%2Ffunctional.html):

`Cortex-M4F allows unaligned reads/writes, but on Cortex-M0 this leads to a HardFault.`

Do you know if we can get some compiler help with that?

gfwilliams, 2017-02-06:
Wow, interesting... I never knew that about the M0. I bet that'll be an issue for other things.

You'd think that there would be a compiler option wouldn't you? I guess worst case we could create a `#define UNALIGNED_UINT16` in jsutils.h - we already have to do that kind of hack for ESP8266 to work around the fact that it'll *only* read 32 bits from flash with `READ_FLASH_UINT16`

urish, 2017-02-06:
There seems to be a `-mno-unaligned-access` compiler option, though, I couldn't get it to work for me - it still produces the same assembly. Going to try the `jsutils.h` approach now

gfwilliams, 2017-02-06:
Ok, thanks! Hopefully you can check some definition to see if it's targeting M0 or not?

It could be useful as I'm sure other platforms will have alignment issues too.

urish, 2017-02-06:
What do you think about the following?

```
#define UNALIGNED_UINT16(addr) ((((uint16_t)*((uint8_t*)(addr)+1)) << 8) | (*(uint8_t*)(addr)))
```

(I checked and it seems to be working correctly)

gfwilliams, 2017-02-06:
Looks great - thanks!

commit 6a2eb0d9 (2017-02-06): attempt to fix #1051
closed 2017-02-06 by gfwilliams