espruino/Espruino #1054: Wi-Fi Protected Setup for ESP8266

MaBecker, 2017-02-06:
Hi,

@gfwilliams please flag this issue as ESP8266 and enhancement

Passing SSI and Password is what we all do to integrate our ESPs into the local Network
What about this: Just push a button on your ESP and on your Router to join ?

The SDK offers four functions WPS :

* wifi_wps_enable 
* wifi_wps_disable 
* wifi_wps_start
* wifi_set_wps_cb
 
Will try to find some samples and than start to implement WPS  into module Wifi for ESP

wilberforce, 2017-04-17:
Now in esp-ide for esp32. Not sure if in v2.0

https://github.com/espressif/esp-idf/blob/master/examples/wifi/wps/main/wps.c

MaBecker, 2019-04-13:
No activity after two years, so lets close this.

closed 2019-04-13 by MaBecker