espruino/Espruino #1055: Add TEA to lib crypto

MaBecker, 2017-02-07:
like discussed in conversation [267841](http://forum.espruino.com/conversations/267841/)

found libs/crypto/mbedtls/include/mbedtls/xtea.h and  libs/crypto/mbedtls/library/mbedtls/xtea.c

but took the source from xxtea.zip and started with this:

jswrap_crypto.h
```
#ifdef USE_TEA
JsVar* jswrap_crypto_TEA_encrypt(JsVar*, JsVar*);
JsVar* jswrap_crypto_TEA_decrypt(JsVar*, JsVar*);
#endif
```

jswrap_crypto.c
```
/*JSON{
  "type" : "library",
  "class" : "crypto",
  "name" : "TEA",
  "generate_full" : "jspNewBuiltin(\"TEA\");",
  "return" : ["JsVar"],
  "return_object" : "TEA",
  "ifdef" : "USE_TEA"
}
Class containing TEA encryption/decryption
*/

#ifdef USE_TEA

/*JSON{
  "type" : "staticmethod",
  "class" : "TEA",
  "name" : "encrypt",
  "generate" : "jswrap_crypto_TEA_encrypt",
  "params" : [
    ["data","JsVar","message to encrypt"],
    ["key","JsVar","key for encryption"]
  ],
  "return" : ["JsVar","The encrypted string"],
  "ifdef" : "USE_TEA"
}*/

JsVar* jswrap_crypto_TEA_encrypt(JsVar *v, JsVar *k)
{
// coding
}

/*JSON{
  "type" : "staticmethod",
  "class" : "TEA",
  "name" : "decrypt",
  "generate" : "jswrap_crypto_TEA_decrypt",
  "params" : [
    ["data","JsVar","message to decrypt"],
    ["key","JsVar","key for decryption"]
  ],
  "return" : ["JsVar","The decrypted string"],
  "ifdef" : "USE_TEA"
}*/
JsVar* jswrap_crypto_TEA_decrypt(JsVar *v, JsVar *k)
{
// coding
}

#endif
```

USE_TEA=1 make

 and then checking gen/jswapper.c for TEA - found none

@gfwilliams  any hints was is missing or wrong ?

MaBecker, 2017-02-08:
added this to, no change, had no idea that this is more than a copy paste thing :-)

```
/*JSON{
  "type" : "library",
  "class" : "crypto",
  "name" : "TEA",
  "generate_full" : "jspNewBuiltin(\"TEA\");",
  "return" : ["JsVar"],
  "return_object" : "TEA",
  "ifdef" : "USE_TEA"
}
Class containing TEA encryption/decryption
*/
````

gfwilliams, 2017-02-08:
Presumably it's the ` "ifdef" : "USE_TEA"` thing. You could try removing that to get something working - but if it's in your Makefile then it should pick it up?

gfwilliams, 2017-02-08:
Just to add - if there's a way to use mbedtls's XXTEA that would be ideal. It's crazy to have two separate libraries that do the same thing in there.

wilberforce, 2017-02-08:
As mentioned in #1057 these are the lookup tables that ideally would be kept in Rom so that more ram does not get eaten:

https://github.com/espruino/Espruino/blob/master/libs/crypto/mbedtls/library/xtea.c#L201

MaBecker, 2017-02-08:
> You'd need DEFINES+=-DUSE_TEA in it

Yep - TEA is availabel

* USE_CRYPT=1 make

    user1.bin uses  475044 bytes

* USE_CRYPT=1 USE_TEA=1 make

   user1.bin uses  476100 byte
 
Next is to check how to implement XXTEA

hungryforcodes, 2017-04-26:
Just curious if this ever made it in -- I didn't see it in the docs or change log... :)

wilberforce, 2017-04-27:
unlesss @MaBecker  has done something in his fork, nothing has been done.

hungryforcodes, 2017-04-27:
Ok thanks for responding -- much appreciated :)