espruino/Espruino #1056: adding a bitrate to I2C1.setup() cause a crash on ESP8266

MaBecker, 2017-02-07:
source that crashes the ESP

- https://github.com/espruino/Espruino/blob/master/src/jshardware.h#L307

source that does not crash

- https://github.com/MaBecker/Espruino/blob/master/src/jshardware.h#L309

Why is the different order of element in a structure an issue, or is this a compiler bug (using xtensa-lx106-elf-gcc-4.8.2) ?

Btw no issues on Pico's  ;-) 
```
> I2C1.setup({scl:B8,sda:B9,bitrate:50000});
> I2C1._options
={ "scl": B8, "sda": B9, "bitrate": 50000 }
> 
```

MaBecker, 2017-02-07:
@gfwilliams  please label it ESP8266

wilberforce, 2017-02-08:
typedef struct {
  Ok so you moved this to the start of the struct....

  int bitrate;  ///< bits per second

I think you need the elements to be on 4 byte boundaries if the code is read from flash.

I seems to recall @tve had padded some other structures to make them align.

I think I had tried other speeds before - setting to 100,000 as the default is strangly 50,000 so perhaps something else has changed?

MaBecker, 2017-02-27:
@wilberforce 

looks like the same issue https://github.com/espruino/Espruino/blob/master/src/jshardware.h#L230

closed 2017-04-22 by MaBecker