espruino/Espruino #1057: has lib crypto cleanup potential  ?

MaBecker, 2017-02-08:
Hi @gfwilliams,

**options to include crypto into firmware**  

USE_CRYPTO=1 make

    includes SHA1, SHA224, SHA256, SHA384, SHA512

USE_CRYPTO=1 USE_TLS=1 make     

    includes AES, PBKDF2,SHA1 SHA224 SHA256, SHA384, SHA512

**What is needed and what can be removed**

* module  ws.js is using hasher SHA1

* function crypto.PBKDF2(passphrase, salt, options) can use one SHA hasher

**Questions**

* what is used by https ?

* can some of the SHA hashers be removed ?

gfwilliams, 2017-02-08:
HTTPS uses AES I believe? But not PBKDF2 - although that's very useful encryption-wise it seems.

Some hashers could be removed, but as far as I know they mostly share the same code so the savings are pretty minimal. Worth checking though - I doubt anyone uses 224 at least.

wilberforce, 2017-02-08:
The issue is that some of these other use large look up tables, that default to be in ram.

I had to do some linker tricks to get the tables used for sha1 and sha256 into rom, otherwise you get even less heap space. 

The other issue is that there isn't enough ram in the esp8266 for the buffers - Gordon has noted you might need 16k and that is never going to happen!

So even if you got it to compile I don't think there will be enough ram left...

Https is working in esp32 build as there is enough ram, and these modules compile without any linker tricks... it also has hardware acceleration for sha256 etc, so I'll look at using that one day considering  this as an optimization for later.

MaBecker, 2017-02-08:
Thanks for your input.

Have to dig more into this to understand how this works ...

wilberforce, 2017-06-09:
@MaBecker 

Can this be closed?

MaBecker, 2017-06-09:
yes

closed 2017-06-09 by MaBecker