espruino/Espruino #1060: No such file or directory: 'bootloader_espruino_1v91.60_puckjs.hex

OwenBrotherwood, 2017-02-09:
Unable to make puck on Ubuntu 16.x
```
GEN espruino_1v91.60_puckjs.lst
GEN espruino_1v91.60_puckjs.hex
Merging SoftDevice and Bootloader
# nrfutil settings generate --family NRF52 --application espruino_1v91.60_puckjs.hex --application-version 0xff --bootloader-version 0xff --bl-settings-version 1 dfu_settings.hex
FIXME - had to set --overlap=replace
python scripts/hexmerge.py --overlap=replace /home/owen/source/repos/github/espruino/Espruino/targetlibs/nrf5x/components/softdevice/s132/hex/s132_nrf52_3.0.0_softdevice.hex bootloader_espruino_1v91.60_puckjs.hex espruino_1v91.60_puckjs.hex -o tmp.hex
Traceback (most recent call last):
  File "scripts/hexmerge.py", line 178, in <module>
    sys.exit(main())
  File "scripts/hexmerge.py", line 158, in main
    ih = intelhex.IntelHex(fname)
  File "/home/owen/source/repos/github/espruino/Espruino/scripts/intelhex/__init__.py", line 93, in __init__
    self.loadhex(source)
  File "/home/owen/source/repos/github/espruino/Espruino/scripts/intelhex/__init__.py", line 202, in loadhex
    fobj = open(fobj, "r")
IOError: [Errno 2] No such file or directory: 'bootloader_espruino_1v91.60_puckjs.hex'
Makefile:2146: recipe for target 'espruino_1v91.60_puckjs.hex' failed
make: *** [espruino_1v91.60_puckjs.hex] Error 1
make: *** Waiting for unfinished jobs....
owen@owen-VirtualBox:~/source/repos/github/espruino/Espruino$ 

```

gfwilliams, 2017-02-09:
type `BOOTLOADER=1` before make along with the other stuff, then try it again

closed 2017-02-09 by gfwilliams

OwenBrotherwood, 2017-02-09:
I knew I was missing something.
Would you like a PR for https://github.com/espruino/Espruino/blob/master/README_Building.md#for-nordic-semiconductors-nrf51nrf52-series-devices
where puck.js is given a mention in  the PR for this area

gfwilliams, 2017-02-09:
That'd be great - thanks!