espruino/Espruino #1062: UDP Support

opichals, 2017-02-11:
# UDP Support

The purpose of this ticket is to track Espruino native UDP support.
The intention is to loosely follow the node.js `dgram` (https://nodejs.org/api/dgram.html) module API .

The development is currently in progress and can be reviewed at https://github.com/espruino/Espruino/compare/master...opichals:espruino-udp

## Phase 1

Minimal effort to get UDP working.

- [ ] Introduce ST_UDP flag
- [ ] Drop NetCreateFlags in favor of SocketType
- [ ] Add SocketType argument to net->createsocket()
- [ ] Implement ST_UDP handling (network_linux.c)
- [ ] Validate feasibility for ESP8266 (network_esp8266.c)
- [ ] Validate this being enough for mDNS.js

```js
// server
let srv = dgram.createSocket('udp4');
srv.bind(41234);
srv.on('data', function(msg, info) { // TBD: temporary, eventually srv.on('message')
    info = { port: 41235, host: 'localhost' }; // faking info, TBD: use the callback param instead

    console.log('s:message', msg);
    // echo
    srv.send('echo1: '+msg, info.port, info.host);
    // echo 2 // TBD: using setTimeout() to separate the message packets
    setTimeout(() => srv.send('echo2: '+msg, info.port, info.host));
});

// client
let sckt = dgram.createSocket('udp4');
sckt.bind(41235); // TBD: till the 'data' event info structure is provided
console.log('sckt', sckt);
sckt.on('data', function(msg) { // TBD: temporary, eventually srv.on('message')
    console.log('c:message', msg);
});
sckt.send('Espruino UDP Message!', 41234, '127.0.0.1');
```

## Phase 2

Making the API more node.js-like (separate pull request).

- [ ] Provide the second 'message' event callback argument to be able to reply (message origin host/port combination)
- [ ] Create a message data object {data, host, port} and pass to clientRequestWrite
- [ ] Adapt eventing. Change 'data' to 'message' and make it send every send() call separately, not concatenating to a stream like pushData does

```js
// server
let srv = dgram.createSocket('udp4');
srv.bind(41234);
srv.on('message', function(msg, info) {
    console.log('s:message', msg);
    // echo (2x)
    srv.send('echo1: '+msg, info.port, info.host);
    srv.send('echo2: '+msg, info.port, info.host);
});

// client
let sckt = dgram.createSocket('udp4');
sckt.on('message', function(msg, info) {
    console.log('c:message', msg, info.host, info.port);
});
sckt.send('Espruino UDP Message!', 41234, '127.0.0.1');
```

wilberforce, 2017-02-20:
@opichals 

I'm interested in following as we'll need to bring the code into the esp32 branch. The esp32 network code is cloned off the Linux base, so what every is working for Linux code be ported...

opichals, 2017-02-20:
I assume the ESP32 related changes are way more complex so I would wait for it to be merged to master and I will rebase on top. What do you think?

wilberforce, 2017-02-23:
@opichals 
Actually the esp32 network code is very similar to the Linux network code, so it should be a simple update.

opichals, 2017-02-24:
For UDP there are updates to the JsNetwork interface which applies and have to be adjusted/synced for all boards netowrk stacks. My hunch is to leave this and add UDP support after the ESP32 stuff is merged.

hungryforcodes, 2017-04-20:
Was there any updates on this by the way? I'm curious about some of the new things this might offer, including mDNS.

opichals, 2017-04-20:
@hungryforcodes I am struggling to understand the Espruino's networking stack in detail as the changes affect all boards. Unfortunately I wasn't able to work on this recently.

On the positive side I got it working with some limitations including a basic mDNS.js responder (Linux, ESP8266 so far). The mDNS support could prove to be quite limited though due to the small amount of RAM available.

wilberforce, 2017-04-20:
The esp32 branch has been merged into mainstream, and the network code was cloned from Linux so should be easy to merge. There is more memory on this board so you would not have the same memory limitations as the esp8266

hungryforcodes, 2017-04-22:
@opichals Thanks for the update :) 

@wilberforce Yes we understand your position on the ESP32 :)

gfwilliams, 2017-08-29:
I think this is probably closable now?

opichals, 2017-08-29:
Sounds all right, I have my minor TODOs elsewhere.

closed 2017-08-29 by opichals