espruino/Espruino #1065: Save failing on ESP8266

chevdor, 2017-02-12:
Hello, I am using an ESP-8266. I flashed 1v91 and get the following issue:

```
>reset()
=undefined
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v91 Copyright 2016 G.Williams
Espruino is Open Source. Our work is supported
only by sales of official boards and donations:
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>save()
=undefined
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)
wdt reset
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
ãìNì|{N|ìlàd ã{²pþInvalid saved code in flash!
Uncaught SyntaxError: Got '@' expected EOF
 at line 1 col 1
@` _ðXQðP0X@
^
> 
```

MaBecker, 2017-02-12:
Hi @chevdor 

you first have to erase your flash, because saved code is not compatible between versions.

    esptool.py --port [/dev/ttyUSB0|COM1] --baud 115200 erase_flash

and then flash new version

chevdor, 2017-02-12:
Wooo @MaBecker! Right on. That was the issue that was bugging me for a while. Thanks a lot. Issue closed!

closed 2017-02-12 by chevdor

chevdor, 2017-02-12:
I won´t get caught again.

It is helps someone:

```
#!/bin/bash

BAUD=115200
FREQ=80m

esptool.py --port /dev/cu.SLAB_USBtoUART --baud $BAUD erase_flash
esptool.py --port /dev/cu.SLAB_USBtoUART --baud $BAUD write_flash --flash_freq $FREQ --flash_mode qio --flash_size 32m 0x0000 boot_v1.6.bin 0x1000 espruino_esp8266_user1.bin 0x3FC000 esp_init_data_default.bin 0x37E000 blank.bin

```

MaBecker, 2017-02-12:
Cool, this will help !

Have you thought about a [donation](http://www.espruino.com/Donate) for Gordon?