espruino/Espruino #1070: How to update esp32 branch to current

wilberforce, 2017-02-15:
Hi Gordon,

@jumjum123 and I have got the esp32 branch to a point that it is reasonably stable.

As we are quite a way behind the master branch, I was wonderingring if you had any recommendation on how to bring it up to date?

There will be files in comflict - e.g. The Makefile and other common files.

Can you recommend an approach?

gfwilliams, 2017-02-15:
Great!

I'd suggest that you do a 'git merge' to bring your branch up to date with master (to be honest it helps if you do this every so often anyway). You can then resolve any conflicts and check it all still works ok, then it'll be easy to pull it back into master.

There should be a bunch of tutorials on this online - it's quite a common thing.

wilberforce, 2017-02-16:
Ok - I applied a rebase from the master back into ESP32.

Gosh - it was awful.  @gfwilliams  I see why you hate extra spaces at the end of lines - any code with extra spaces can't be merged cleanly and required edits.

https://github.com/espruino/Espruino/pull/1071

closed 2017-02-17 by wilberforce