espruino/Espruino #1073: save( ) can't work well on ESP32

aifer2007, 2017-02-16:
@wilberforce
var wifi = require("Wifi");
wifi.connect("HG", {password:"12345678"}, function(err){
console.log("connected? err=", err, "info=", wifi.getIP());
wifi.save();
var a = 1;
var b = a+1;
console.log("wifi connect");
console.log(b);
});

send , and the codes work well, then save() left, ok.
after reboot, the wifi is reconnected,
BUT, the var a and b are undefined, and the console.log codes are not executed.
none of the 2 console.log codes r exe.

aifer2007 @aifer2007 22:52
@wilberforce
Sometimes, after save() ok, when reboot, the ESP32 crashed. I have to reflah it.
BUT, the same code can be save() and load on ESP8266(ESP-12F), reboot is ok.
It seems that save() can't work well on ESP32 now.

MaBecker, 2017-02-16:
Hi @aifer2007,

var a and b are not defined as global, use dump() to list saved code.

aifer2007, 2017-02-17:
Thanks,
var start = function () {
// user codes here
};
function onInit () {
start();
}
save();
ESP32.reboot();
Such , the user codes can be saved and loaded when ESP32 starts/restrars.

MaBecker, 2017-02-17:
Cool, so it works as designed !

. . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . .
 You should think about a  [donation](http://www.espruino.com/Donate) for Gordon

wilberforce, 2017-02-17:
Thanks. Please close this issue

closed 2017-03-01 by wilberforce