espruino/Espruino #1078: TypeError: Cannot read property 'length' of undefined (Caused by arrow functions in module)

PaddeK, 2017-02-20:
If i use arrow functions in a module i get an error while uploading the code via send button from WebIDE to espruino (wifi).

The error is:
`TypeError: Cannot read property 'length' of undefined`

Minimal test module (test.js):
```
'use strict';

let Test;

Test = function () {
    this.mtFn = () => {};
};
module.exports = Test;
```

Simply require the module and upload code to espruino will trigger the error.

Right hand side of WebIDE:
```
let Test = require('test');
```

If i replace the arrow function `this.mtFn` with a 'normal' function `this.mtFn = function () {};` i get no error.

gfwilliams, 2017-02-21:
Where does this error come from? The Web IDE? I guess it's not from Espruino itself as there is no use of `.length` in the code

Unfortunately Espruino doesn't support all of ES6 so `let` won't work correctly, but that shouldn't actually be your problem.

PaddeK, 2017-02-21:
Yes it comes from the Web IDE and as far as i can tell the uploaded code works fine so far.

gfwilliams, 2017-02-21:
Looks like it's a bug inside the esprima minification library that's being used.

If you turn off module minification in settings, or turn it to the online minification, it should work

PaddeK, 2017-02-21:
Both suggestions work. I will use the online closure minifier for now.
Thank you!

gfwilliams, 2017-02-27:
Closing this - as moved to EspruinoTools now: https://github.com/espruino/EspruinoTools/issues/58

closed 2017-02-27 by gfwilliams