espruino/Espruino #1081: E.setBootCode("")  let ESP32 reboot again and again

aifer2007, 2017-02-22:
for example, 
E.setBootCode("var = 1;") or E.setBootCode("var = 1;", true)
enter, after writing flash, reboot ESP32, it will reboot again and again.

wilberforce, 2017-03-08:
To reset the js area on the board use:

`esptool.py  --chip esp32 --port COM3 --baud 921600 --before esp32r0 --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x100000 blank.bin`

wilberforce, 2017-03-08:
Tried to use core dump tool - can't get get command to work:

`./esp-idf/components/espcoredump/espcoredump.py -c esp32  info_corefile core-dump.txt -t b64`

core-dump.txt:

```
================= CORE DUMP START =================
lB0AAAsAAAB0AQAA
VB79P0Ac/T9IHv0/
4Bv9P+Ad/T8IBAAAeEz7P3hM+z9UHv0/cEz7PxQAAABqEUNV12WgnlQe/T8AAAAA
BQAAADz3/D9lc3BydWlub1Rhc2sAcSMAAAAAAEge/T8AAAAAIAEGAAUAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGC7/D/Iu/w/MLz8PwAAAAAAAAAA
AQAAAAAAAACUaEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAsNTg
AAAAAP0UAEAwDwYAOmMQgAAd/T8IABAABAAQAP8AAAAA/wAAAAD/AAAAAP8AAAAA
gBz9PwAAAABAHAhABAAAAAAAAAAAAEAAJAAAAAwAAAAcAAAACAAQAP0UAEANFQBA
/////wQAAAAAAAAAaBAIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAABId/s/AgAAABEAAAB8oBKAEB39PwgAEAABAAAA
w30QgDAd/T8AAAAAAAAAABkAAAAAHgAAAQAAAAAAAAA7fxCAUB39PwAAAAAAAAAA
AAAAAEh3+z8AAAAAAAAAAHp/EICAHf0/AQAAAAEAAAB6fxCAgB39PwEAAAABAAAA
AAAAAAAAAAABAAAAAAAAAIA2E4CgHf0/AQAAAAAQAABrZghAEOH8PyMABQAAAAAA
AAAAAMAd/T8AAAAAAAAAACMJBgBUHv0/AQAAAAAAAAAAAAAA4B39PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAOwd/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
oOL8PzDh/D+U4vw/
MOH8PzDi/D/OXoY5FEz7PxRM+z+g4vw/DEz7PxkAAADg+dxwOjEwhaDi/D8AAAAA
AAAAAFjd/D9JRExFAE6ffUWLRfLKmCkAAAAAAJTi/D8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGC7/D/Iu/w/MLz8PwAAAAAAAAAA
AQAAAAAAAACUaEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAd9Jc
uBAIQOiSFEAwCQYAPFgIgPDh/D8IAAAAAQAAACMCBgCg4vw/AQAAAAAAAAAUYPs/
0OH8PwAAAAAgCQYAbFQIgLDZ/D8DAAAAIwgGAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAALkSCECw2fw/oGUIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAGxUCICw2fw/AwAAACMIBgAAAAAAEOL8PwAAAAAAAAAA
IwIGAKDi/D8BAAAAAAAAAAAAAAAw4vw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAPOL8PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
LP39PzCm/j/cp/4/
MKb+P3Cn/j9tBAAAAEz7P8j1/D8s/f0/+Ev7PwUAAAD8rf4//K3+Pyz9/T/0rf4/
FAAAABCg/j91YXJ0VGFzawAAAACABBEAAAAAANyn/j8AAAAAIAcGABQAAAABAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGC7/D/Iu/w/MLz8PwAAAAAAAAAA
AQAAAAAAAACUaEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAADNLCIAgBQYAAAAAAA1vCICQpv4/OKn+PwAAAAAAAAAACQQAAAAAAAC4X/s/
ZAAAAICm/j8sqf4/AAAAABiu/j8AAAAAAQAAAAMAAADSjBSAwKb+Pzip/j/wpv4/
8Kb+P+Cm/j8AAAAA4LX8P2QAAAAAAAAAdKn+P3Sp/j/6aBGA4Kb+PwAAAADgtfw/
ZAAAAAAAAAAAAAAAAAAAAFE2E4Awp/4/AAAAAAAAAAAAAAAAUKf+PwAAAAAAAAAA
AMIBAAMAAAAAAAAAAQAAAAAAAAB6AAAA//////////8AAQAAZAAAAAAAAACo/v0/
AAAAAFCn/j8AAAAAAAAAACMCBgAs/f0/AQAAAAAAAAAAAAAAcKf+PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHyn/j8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
wPX8P3D0/D+09fw/
cPT8P1D1/D9MBAAANP39PwBM+z/A9fw/+Ev7PwcAAAC87Pw/vOz8P8D1/D+07Pw/
EgAAALjt/D90aVQAUm48VveFICdU6/cA////f7T1/D8AAAAAIAUGABIAAAABAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGC7/D/Iu/w/MLz8PwAAAAAAAAAA
AQAAAAAAAACUaEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
[... 87 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAYzwZ
AAAAADNLCIAgBwYAAAAAAAAAAABgn/4/pGz7P2Aw+z8AAAAApQMAAKRs+z9gMPs/
/////2Cf/j+kbPs/YDD7P3Tl/T8AAAAAAQAAAAAAAAAAAAAAoJ/+PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAA8AAAAAAAAAAAAAAAAAAACobPs/GAAAABymCUCXbPs/
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAArJ/+PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
TMP8P2DC/D9Aw/w/
YML8P+DC/D9zd5AqNPb9P6RL+z9Mw/w/nEv7PwEAAADcvfw/3L38P0zD/D/Uvfw/
GAAAAAS+/D9pcGMwAPY+xY+47cH1I0YAAAAAAEDD/D8AAAAAIAAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGC7/D/Iu/w/MLz8PwAAAAAAAAAA
AQAAAAAAAACUaEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAlajR
AAAAADNLCIAgAAYAAAAAAAAAAADAwvw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
/////wAAAAAAAAAAAAAAAPi9/D8AAAAAAQAAAAAAAAAAAAAA4ML8PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAB4MCIAQPP4/jEv7P0RO+z8AAAAAAAAAAOzC/D8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
================= CORE DUMP END =================
```

wilberforce, 2017-03-09:
https://github.com/espressif/esp-idf/issues/409 Coredump needs elf file

wilberforce, 2017-03-14:
Appears to be reading from flash ins the issue... copying to ram first works...

Which defeats the purpose... so just next 4 byte aligned reads like the flash read code.... and ESP 8266 code

https://gitter.im/espruino/esp32?at=58b02bce7ceae5376a32c1d6

wilberforce, 2017-03-30:
ESP8266 code adjusts the flash read address:

https://github.com/espruino/Espruino/blob/9ec75f55ce4391550f8895128be9f22d77c6f9c1/src/jswrap_flash.c#L555-L560

Possible ESP32 equivalent...

https://esp-idf.readthedocs.io/en/v1.0/api/spi_flash.html

wilberforce, 2017-04-03:
@aifer2007 

This is fixed in https://github.com/espruino/Espruino/pull/1129

A new partition table is required that defines where the js_code is stored.

commit a44e3cbb (2017-04-03): Fix #1081 E.setBootCode();
commit 011042e2 (2017-04-03): Fix #1081 E.setBootCode();
closed 2017-04-11 by wilberforce