espruino/Espruino #1085: ESP32 merge into master

gfwilliams, 2017-02-23:
Hi everyone,

I thought a while back there had been a push to merge ESP32 into master, and all the recent changes had been pulled in with `git merge`.

Looking at it, it's still miles behind - so either the merge wasn't done or something has failed horribly. Any ideas what's happened?

It'd be nice to get this merged properly as it looks like there's a bunch of interest in it now

jumjum123, 2017-02-23:
Hello Gordon,

should have been done with this
https://github.com/espruino/Espruino/commit/425c2c82b783fcb8a34ab0a738c5d605bbb6cdc9
Wilberforce did the merge.
I've seen this:https://gitter.im/espruino/esp32?at=58a54f011465c46a56701d7e

After comiling we get a file named espruino_1v91.2321_esp32.tgz
I compared actual ESP32 version against an 10 days old commit from MaBecker
for ESP8266, and both are the same.
Do you have any other file to check ?
I'm away from home for this week, therefore I cannot check with ESP32
itself.

There must be something crazy.
As a first overview, we have running, but not fully tested with a lot of
devices
- Espruino itself
- GPIO
- SPI
- I2C
- ADC
- DAC
- OneWire
- PWM
- Pulse
- Telnet
- HTTP (and https, AFAIK)
- microsec delay

BTW, where is the bunch of interest comin from ?

regards from Krefeld(stormy and 10 degrees)
Jürgen

gfwilliams, 2017-02-23:
Thanks! The strangeness is really from: https://github.com/espruino/Espruino/compare/master...ESP32

Maybe I'm using GitHub wrong but as I understand it, that should be comparing the main branch with the ESP32 branch - but there are quite a few differences that shouldn't be in there.

Having said that it seems to be implying that the ESP32 branch is newer than master, so maybe it's just that GitHub has got confused.

gfwilliams, 2017-02-23:
Not sure about the interest - there just seemed to be a few more bug/forum posts mentioning ESP32 now

jumjum123, 2017-02-24:
> Thanks! The strangeness is really from: master...ESP32

Hmm,

could you please give an example for differences ? May be, we can find the
reason.
ESP32 branch will always be newer after merging actual master. There is no
way to avoid that.

Worst case would be, Github could not handle this correctly.
In that case we would have to merge manually to master.
We tried to have all ESP32 specifics in ESP32 related directorys.
So this should be doable. I would like to avoid this, obviously.

Regards from Krefeld(sunny and carneval)
Jürgen

gfwilliams, 2017-02-24:
Well, just take a look at https://github.com/espruino/Espruino/compare/master...ESP32 and click on `Files changed`

ChangeLog is a good example - you'd expect it to be more or less the same. However the 'new' lines are on the right which should be the ESP32 side.

Just looked at this - it's fine. GitHub appears to be diffing ESP32 with master at the point ESP32 forked, not the current master. Diffing on my PC shows it's fine.

I'll leave this open until I've properly merged though.

gfwilliams, 2017-02-24:
Just looked through - it's some amazing work. You managed to just slot support in without having to make any real changes to the main Espruino codebase  - thanks everyone!

Shall I remove the ESP32 branch - or do you want to keep it open as a staging area for other ESP32 changes?

jumjum123, 2017-02-24:
Hello Gordon,

thanks for the feedback. We would like to wait for release of esp-idf V2.0.
This should arrive end of this month.
Usually there are some "minor last minute changes".
Last not least there are some changes on my todo list, that should be done.
For example jshReset need to be filled with sense, and some files will be
renamed to have a common naming.

Personally, I would prefer to keep ESP32. This is good for me to avoid
changes by accident on master.

Only common file changed that I know about is makefile.
BTW, its good to know we have been successful by seperating ESP32 from core
Espruino.

Regards from Krefeld(its hot time of carneval in this part of the world)
Jürgen

gfwilliams, 2017-02-24:
Sure, no problem!

By the way, I'll remove `ESP32.neopixelWrite` soon - I'm finally giving up and adding `E.neopixelWrite` :)

wilberforce, 2017-02-27:
@gfwilliams @jumjum123 

I have done a new pull request that should update the ESP32 branch up to the current master
https://github.com/espruino/Espruino/pull/1087

Hopefully after this the number of changes should be minimal

jumjum123, 2017-02-27:
I would like to add my 2 cents for neopixelWrite.
This looks like an extension for special hardware, the neopixel.
Main reason for creating this for ESP8266 was a less powerful SPI driver.
In Espruino we already used SPI for writing to WS2812B. And for ESP32 the
ESP8266 solution was used as a starting point.

Anyway, I can imagine using some other special hardware where Javascript is
not the right solution.
For example, I would like to control one of these 64*32 rgb panels, which
are very cheap.
If an Arduino has the power to support them, ESP32, ARM and others should
be able to do the same.

Other hardware extensions are looking around the corner, like CAN bus.
And at the end aren't BT, VGA or  SD cards already a special kind of
hardware ?

How could we use them inside Espruino in a general way?
jshardware.h was and is a very big help in porting Espruino to ESP32.
On the other hand, its a restriction if you want to add something which is
not implemented yet.
Another location for device specific source are targetlibs and libs.
At the end, I would not know exactly where/how to add more driver. I dont
see a general direction, like it is available with jshardware.

Comin back to neopixel, I wonder if E.neopixel is the best way we should go
for the future.
Especially the E bothers me. Would it be an option to have a class NeoPixel
with method write ?
BTW, there is no source available for ESP32 yet. Sooner or later I will add
this, there is already a nice solution available, based on a "give me more
data"-interrupt.

Right now I am wondering, if somebody would follow my 2 cents, they may be
more confusing then helpful ;-)

gfwilliams, 2017-02-27:
I know what you mean... It would probably make more sense to add its own library - I just wanted something that would entice people away from `ESP8266.neopixelWrite` :)

I have things like `tv` in the `libs` folder. Perhaps I should just do that and have - as you say - `require("neopixel")`? Since I haven't done a release yet, now would be the time to change it

gfwilliams, 2017-02-27:
Anyone else have any strong feelings about this?

wilberforce, 2017-02-27:
> I have things like tv in the libs folder. Perhaps I should just do that and have - as you say - require("neopixel")? Since I haven't done a release yet, now would be the time to change it

This sounds less hacky than `E.neopixelWrite`. If it is a class in it's own right, then at least it can have different methods - perhaps for setup of hardware - making it a bit more extensible.

gfwilliams, 2017-02-28:
Ok, any thoughts about the naming? 'neopixel' is ok I guess, but neopixel is just Adafruit's brand name for them. In reality it encompasses loads of different devices, WS281x, APA10x, etc

gfwilliams, 2017-03-01:
Have you checked out the code? That's almost exactly what I did yesterday 

https://github.com/espruino/Espruino/commit/7928db756ecc182846fdada287eeb484339245b1

I forgot to commit jswrap_neopixel at that point though. I just shoved the ESP8266-specific code in there. It's not huge and I'm not sure it warrants its own file - potentially it could be used for some other platforms too.

jumjum123, 2017-03-01:
> 7928db7

I checked out this morning, and neopixel did not exist at that time (?)

As far as I can see, there are 2 major differences between yours and mine
- splitted into 2 functions (init and write)
- source code for neopixel is located in targets/*board*/libs. where board
in my case is esp32

My idea was to seperate board specific source in board specific folder. At
least for ESP32, the source is complex.
Interrupt handler are used to minimize memory usage. To keep this together
with specific source for each board could end up in a huge file.
On the other hand, this moves some complexity to makefile.

gfwilliams, 2017-03-01:
I forgot to commit it. Everything else was there though - including removal of the neopixel code from jshardware.c

I have a separate file for nrf52 - I put that directly in `targets/nrf52` (no libs directory) so you could do the same (or not)... Up to you.

Also, do you need an init function? It just confuses things. Are you going to add a deinit one as well?

closed 2017-03-01 by gfwilliams

jumjum123, 2017-03-01:
I could live without init, no problem.
Same for working without target/.../libs
If my understanding is correct, you prefer to add #elif for ESP32 in
libs/neopixel/jswrap_neopixel
Please correct me if I'm wrong.

Personally, I'm a bit afraid to change something in core Espruino, and
would avoid it whenever I can.

gfwilliams, 2017-03-02:
Just add an `#elif` that calls to a function you've defined in `target/esp32/whatever` - I think that's safest

My hope is that at some point a lot of boards will be able to use the STM32's SPI method (which uses all jsHardware functions) - but until then we'll need platform specific code

PR #1086 (2017-02-24): ESP32 master merge
fixes #1085