espruino/Espruino #1093: Travis builds - location of headers and libs - esp32

wilberforce, 2017-02-28:
@gfwilliams @jumjum123 

The esp32 builds require and /app and /esp-idf folder.

Currently I have set up a project to build the libraries and required headers:

https://github.com/wilberforce/espruino-esp32

Part of this build will be to generate the tgz files needed by the Travis build, and the trial Travis build is referring to:

https://raw.githubusercontent.com/wilberforce/espruino-esp32/master/esp-idf
https://raw.githubusercontent.com/wilberforce/espruino-esp32/master/app.tgz

When the upstream esp-idf files changes, these will need to be rebuilt.

Would the best place for these files to be under the main Espruino project as:
Espruino/espruino-esp32 

Or should the tar files go under:

https://github.com/espruino/Espruino/tree/master/targetlibs/esp32
?

I'm suggesting the tar files rather than the .a and .h and bin files as there are quite a few.

gfwilliams, 2017-02-28:
Roughly how big are they? if it's less than 1MB total I'd stick them in targetlibs/esp32 but if not, I think a separate repo is probably better

gfwilliams, 2017-02-28:
Is this mainly for the travis build?

How about we create an `espruino-build` or similar and put *everything* we need in there - right now we're still dragging files off @tve's server for ESP8266 builds and we could put them in there too

wilberforce, 2017-02-28:
Yes - for the Travis builds. So have a top level with board specific sub folders. That would work.

Although it could also be useful for people to build custom firmware without having to build the framework too - I guess they just pull the .tgz files like Travis does.

wilberforce, 2017-02-28:
There is already this (outdated) - this could be repurposed?
https://github.com/espruino/EspruinoBuilds

gfwilliams, 2017-02-28:
> Yes - for the Travis builds. So have a top level with board specific sub folders. That would work.

Sounds good - for files that *could* be built I guess you could stick instructions/scripts in there too if you wanted?

> I guess they just pull the .tgz files like Travis does.

yeah, that should be fine?

I'd leave EspruinoBuilds - but maybe something like `EspruinoBuildTools`?

> Here is the Travis file to date if you would like to eyeball it

Looks good - I think we should also put those files from the EspressIf servers on GitHub though - I bet they'll disappear in a year or two :)

> I have commented out the other boards to speed up the build

Just a thought here - but how about we have some script like `provision.sh` where you say `provision.sh ESP8266` and it drags in all the code you need to compile for ESP8266?

That way we move the code that's currently duplicated in Vagrant/Travis/Docker/etc into one common place.

AND we could then make sure we only provision for the board we're building - which would drastically speed up all builds

wilberforce, 2017-02-28:
Ok, if you set up the new repository, I can  clone and add esp32 stuff. Once we are happy with the patterns etc, esp8266 could be added.

Are there any other boards that need toolchains and libs?

Let me know if you have an idea for Folder structure.

wilberforce, 2017-02-28:
> provision.sh ESP8266 and it drags in all the code you need to compile for ESP8266?

How about 
ESP8266/provision.sh
ESP32/provision.sh

Then the provision scripts would be independent, and not switch statements in the script, easier to maintain.

gfwilliams, 2017-02-28:
Sounds good. Just added: https://github.com/espruino/EspruinoBuildTools

Maybe go lowercase with the paths, as that's what we have in the main Espruino repo

wilberforce, 2017-03-07:
https://github.com/espruino/Espruino/pull/1095

wilberforce, 2017-03-07:
https://github.com/espruino/Espruino/issues/1105

closed 2017-03-07 by wilberforce