espruino/Espruino #1096: I2C clock stretch on esp8266

MaBecker, 2017-03-03:
Hi Gordon,

can you please do another cherry picking to have I2C clock stretching for ESP8266?

https://github.com/MaBecker/Espruino/commit/c6eb1ba8b16b31456b277701921cd75a810143a4

Or is there a way for me to do this by my own, what I would prefer.

gfwilliams, 2017-03-03:
You should be able to do:

```
git checkout master
git checkout -B clockstretch
git cherry-pick c6eb1ba8b16b31456b277701921cd75a810143a4
git checkout master
git push origin clockstretch
```

And then create a PR from the clockstretch branch that should then be on your GitHub?

MaBecker, 2017-03-03:
Thanks - works , but cherry-pick c6eb1ba8b16b31456b277701921cd75a810143a4 is also containing all the stuff before. 

As this is only a single file I use copy and paste

MaBecker, 2017-03-03:
add clock stretch #1097

gfwilliams, 2017-03-03:
Closed with https://github.com/espruino/Espruino/pull/1097

closed 2017-03-03 by gfwilliams