espruino/Espruino #1102: ESP8266.deepSleep(micros, option) to discuss

MaBecker, 2017-03-05:
@gfwilliams  please add ESP8266 and enhancement label.

Asking for a possible way to control power consumption on wakeup by adding a second parameter  to ESP8266.deepSleep().

Any comment, suggestion or advise? 

**Call type:**

ESP8266.deepSleep(micros, option)

**Description**

Put the ESP8266 into 'deep sleep' for the given number of microseconds, reducing power consumption drastically.

meaning of option values:

0 - the 108th Byte of init parameter decides whether RF calibration will be performed or not.

1 - run RF calibration after waking up. Power consumption is high.

2 - no RF calibration after waking up. Power consumption is low.

4 - no RF after waking up. Power consumption is the lowest.

Note: unlike normal Espruino boards' 'deep sleep' mode, ESP8266 deep sleep actually turns off the processor. After the given number of microseconds have elapsed, the ESP8266 will restart as if power had been turned off and then back on. All contents of RAM will be lost.

Special: 0 microseconds cause sleep forever until external wakeup RST pull down occurs.

**Parameters**

micros - Number of microseconds to sleep.

option - possible values are 0, 1, 2 or 4

MaBecker, 2017-03-17:
Great, so I will create a PR in the next days

TrevorJTClarke, 2017-03-17:
@MaBecker this would be excellent!! I have been trying to get deepsleep of any kind working on esp8266... It keeps locking or needing a forced reset. Looking forward to your work!

MaBecker, 2017-03-17:
Hi @TrevorJTClarke,

did you connect gpio16 to rest? 

I guess it is not part of of the reference, will plan to add this.

MaBecker, 2017-03-17:
Made a decision: 

I will add "bool system_deep_sleep_set_option(uint8 option)" as additional function

TrevorJTClarke, 2017-03-17:
@MaBecker thanks for the reply!!
No, i didn't realize I needed to do this! Is there documentation somewhere about this? Im not familiar enough with the esp side of things i suppose. 
So if i connect 16 to reset, it should allow for `ESP8266.deepSleep(micros)` correct? (I understand the second param is the discussion here).

MaBecker, 2017-03-17:
>So if i connect 16 to reset, it should allow for ESP8266.deepSleep(micros) correct?

Yes

TrevorJTClarke, 2017-03-17:
@MaBecker working now! Appreciate the help!

MaBecker, 2017-04-18:
#1142

closed 2017-04-19 by gfwilliams