espruino/Espruino #1103: ESP32 D21.reset(); fails also D21.digitalPulse(rst,0,10)

wilberforce, 2017-03-06:
See http://forum.espruino.com/comments/13506090/ raised by @user73896
Coonect lead between GND and D21

`D21.set(); ` lights up - `D21.reset();`  doesn't reset.

digitalPulse possible related, @jumjum123 not sure if the timer code for this has been done?

wilberforce, 2017-03-06:
Also need to check `D21.toggle()`

jumjum123, 2017-03-06:
Works fine on my board with:
- D21.set() --> led on
- D21.reset() --> led off
- D21.toggle() --> led on
- D21.toggle() --> led off
No idea whats going wrong on your board.
Only to make sure, compiled from ESP32 branch

wilberforce, 2017-03-06:
set reset works for me too.

Seems to be issue with:

>digitalPulse(D22,0,10);
=undefined
>digitalPulse(D22,1,10);

jumjum123, 2017-03-06:
Just tested toggle one more time directly after reset without doing any command.
-     --> led is off
- D21.toggle() --> led stays off
- D21.toggle() --> led on
- D21.toggle() --> led off
- D21.toggle() --> led on
So first toggle only has a problem, if pin is not used before.
Will check digitalPulse later today

wilberforce, 2017-03-06:
I get different output! ` 1v91.390`
 Toggle only works after a reset()!

Then won't toggle off.

For the pulse:

https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardware.c#L396
 ` int duration = (int)pulseTime * 1000; //from millisecs to microsecs`

https://github.com/espruino/Espruino/blob/master/targets/esp32/jshardwarePulse.c#L92
```
//pin to be pulsed. value to be pulsed into the pin. duration in milliseconds to hold the pin.
void sendPulse(Pin pin, bool pulsePolarity, int duration){
```

Could be a factor of 1000 out - comments don't match up.

jumjum123, 2017-03-06:
Correct, comments don't match. But thats not the problem.
Looks like duration is not used. There is only one pulse.
So one pulse is working fine.
Have to find a nice way to stop.

jumjum123, 2017-03-06:
Tested with digitalPulse(D21,0,[4,5,6,7,8]);  --> works fine
next test with digitalPulse(D21,0,[4,50,6,7,8]); --> 2nd pulse (50ms) is only 17 ms long

jumjum123, 2017-03-06:
RMT timer has a base frequency of 80Mhz.
This can be divided by 1 ...  255
Duration of pulse is 1... 32768
So longest pulse time would be 104ms, at least in theory.
Some more investigation is needed now from my side.
Actually, a divider of 80 is used in Espruino, therefore we use recalculation to usec

jumjum123, 2017-03-07:
Found the reason, why D21.reset() did not work.
Config of rmt connects a pin to rmt directly.
Due to that, GPIO cannot set/reset D21 anymore, its not connect to the pin anymore.
According to espressif we would need a gpio_matrix_out with SIG_GPIO_OUT_IDX
We could do this after each pulse, it would take some CPU time.
Any comments to this ?

wilberforce, 2017-03-07:
Can we set the config for RMT only to occur when digitalPulse is used?

I wonder of this is also making Onewire inconsistent?

jumjum123, 2017-03-07:
The only way, I know, is to config rmt before pulse and to reset to GPIO after the pulse.
Have to find a way to recognize end of pulse (?)
Or the other way, we could switch to GPIO, whenever we set/reset/toggle a pin.

wilberforce, 2017-03-07:
>Or the other way, we could switch to GPIO, whenever we set/reset/toggle a pin.

That might the cleanest way

jumjum123, 2017-03-07:
Is done in ESP32 branch.
Had to reset to rmt in jshardwarePulse too.
Hopefully, this did not create conflicts somewhere else (SPI or I2C for example).
Same problem could appear, if you _PIN_.set() for an already for SPI assigned pin

aifer2007, 2017-03-07:
Great

发自我的 iPhone

wilberforce, 2017-03-09:
@jumjum123 

Is this all sorted now - can we close, or does it need a pull request to bring into master?

wilberforce, 2017-04-02:
@jumjum123 

Can we close this now?

jumjum123, 2017-04-02:
Works as expected,
as long as nobody finds something different ;-)

closed 2017-04-02 by jumjum123