espruino/Espruino #1105: Improving the build process

gfwilliams, 2017-03-07:
Hi,

@wilberforce  @jumjum123  you've been looking at the build system a bit recently... Please could you take a look at:

https://github.com/espruino/Espruino/compare/NEW_BUILD?expand=1

Given our recent issues with `BOARD=` in https://github.com/espruino/Espruino/pull/1104 it does seem like by far the most sensible way of doing builds, as a typo will be able to generate an error rather than blindly doing a linux build.

What I've done is replaced the individual board-specific makefile bits with a script file `scripts/get_makefile_decls.py` which creates the relevant definitions (`CURRENT_BOARD.make`) and then loads them into the Makefile. It should reduce a lot of duplication. I've currently done it for the 4 main Espruino boards and Linux as a proof of concept.

I'm also planning on breaking out each of the `USE_XYZ` bits of Makefile into separate Makefiles that would then also get included, and doing the same for the chip family - it should make the Makefile a lot more managable.

Any thoughts?

@jumjum123 I guess this may break your cool build website (it may not) - *however* it should be an easy fix, and has the benefit that you can basically just analyse and modify `CURRENT_BOARD.make` to do pretty much all of the build config options? It should make your life a lot easier.

gfwilliams, 2017-03-07:
I've just moved the majority of platform-specific stuff into different Makefiles as well. It shouldn't affect the build much but it makes the Makefile way easier to work with. ESP8266 and ESP32 now seem to work, but maybe you could check that those all seem to be working ok.

Then it's just the nasty job of going through all the rest of the boards

gfwilliams, 2017-03-07:
Ok, all of Travis now builds ok on that branch now. Any reasons I shouldn't drag this in? Should be quite an improvement - travis builds are way faster for ARM now too.

wilberforce, 2017-03-07:
Gosh - you have been busy!
https://github.com/espruino/Espruino/compare/NEW_BUILD#diff-a8bf4bc96799a9c3df0a143fdc63aa3eR28

a. One thing I did notice that might effect ESP8266 users for the next release.
boards/ESP8266_BOARD.py
#'GRAPHICS',

Was this turned back off on purpose?

b. 
> Ok, all of Travis now builds ok on that branch now. Any reasons I shouldn't drag this in? Should be quite an improvement - travis builds are way faster for ARM now too.

@MaBecker  might like to pull the NEW_BUILD branch and check the ESP8266?

I can or @jumjum123 can test the ESP32.

I guess before updating master there would be quite a few building docs that need to update? Perhaps we could list here?

jumjum123, 2017-03-07:
> a8bf4bc96799a9c3df0a143fdc63aa3eR28

@wilberforce
I'm just fighting with some problems around I2C. Could you please take over
test for ESP32 ?

@gordon
I've to take a closer look to see how my build website will work with this.
Time is limited, next week could be a good guess.
Since feedback was poor, and due to working on ESP32 port, I did'nt spend
much time on that. May be, your changes are a good reason to start again :-)

wilberforce, 2017-03-08:
I've found a bit of weirdness.

a. un-commented the GRAPHICS in `boards/ESP8266_BOARD.py`
and run:  `python scripts/get_makefile_decls.py ESP8266_BOARD`

```
# Generated with scripts/get_makefile_decls.py ESP8266_BOARD
BOARD=ESP8266_BOARD
PROJ_NAME=espruino_1v91.617_esp8266
EMBEDDED=1
FAMILY=ESP8266
CHIP=ESP8266
USE_NET=1
USE_TELNET=1
USE_CRYPTO=1
USE_NEOPIXEL=1
```

GRAPHICS is not present??. 

Changes to `boards\*.py` does not seem tp update?

b. `make clean`

`include make/sanitycheck.make` spits out....  

```
*************************************************************
*           To build, use BOARD=my_board make               *
*************************************************************
```
There are other targets that could put out this warning too such as `flash` and `wiflash`.

Or is this required now in all cases?

MaBecker, 2017-03-08:
Just a hint for ESP8266:

* if adding GRAPHICS=1 also add RELEASE=1  otherwise firmware is too large 

* and set vars to 1600 otherwise heap is to small

wilberforce, 2017-03-08:
> Just a hint for ESP8266:
people are currently expecting graphics to be compiled in right?

MaBecker, 2017-03-08:
> @MaBecker might like to pull the NEW_BUILD branch and check the ESP8266?

sure, I can do.

>Just a hint for ESP8266:
people are currently expecting graphics to be compiled in right?

I would say yes.

gfwilliams, 2017-03-08:
> Graphics

Odd - I tried to copy everything as-is, I might have missed it. Did you add the correct commas in when you uncommented it? Also, did you make clean? If you mess up the python file it won't build a new one - but that should really have broken the build process.

> Docs

Good point about the docs - should be pretty easy to change, and sanitycheck.make tries to lend a helping hand if someone uses the wrong command

> Build website

@jumjum123 I wouldn't be too disheartened. Do you know how much it's being used?

I find with Espruino often no news is good news. People rarely say anything when it's working - you only hear back when something breaks! Just yesterday I got an email from someone who explicitly mentioned your website (although they wanted a custom Puck.js build!). 

I'm also more than happy to add a link to the Espruino building page - sorry, I should have done that before :(

gfwilliams, 2017-03-08:
Also, ESP8266 **never had graphics built in?**

This is from the makefile:

```
-else ifdef ESP8266_BOARD		
-EMBEDDED=1		
-USE_NET=1		
-USE_TELNET=1		
-#USE_GRAPHICS=1		
-USE_CRYPTO=1		
-USE_NEOPIXEL=1		
-BOARD=ESP8266_BOARD
```

Those fields in the python file never actually did anything (although they do now!)

MaBecker, 2017-03-08:
> Also, ESP8266 never had graphics built in?

True, but it would be a nice to have 😄

MaBecker, 2017-03-08:
set  'variables'       : 1600 in boards/ESP8266_BOARD.py

first run

    BOARD=ESP8266_BOARD make 

    user1.bin uses  478580 bytes of 479232 available

    # Generated with scripts/get_makefile_decls.py ESP8266_BOARD
    BOARD=ESP8266_BOARD
    PROJ_NAME=espruino_1v91.2748_esp8266
    EMBEDDED=1
    FAMILY=ESP8266
    CHIP=ESP8266
    USE_NET=1
    USE_TELNET=1
    USE_CRYPTO=1
    USE_NEOPIXEL=1

second run - add  'GRAPHICS' in boards/ESP8266_BOARD.py

    RELEASE=1 BOARD=ESP8266_BOARD make 

    user1.bin uses  475908 bytes of 479232 available

    # Generated with scripts/get_makefile_decls.py ESP8266_BOARD
    BOARD=ESP8266_BOARD
    PROJ_NAME=espruino_1v91.2748_esp8266
    EMBEDDED=1
    FAMILY=ESP8266
    CHIP=ESP8266
    USE_NET=1
    USE_TELNET=1
    USE_GRAPHICS=1
    USE_CRYPTO=1
    USE_NEOPIXEL=1

Works great.

Is there a chance  to add  RELEASE=1  ?

wilberforce, 2017-03-08:
@MaBecker 
Please post the provision script you are using...

MaBecker, 2017-03-08:
> Please post the provision script you are using...

no script, just make calls ....

or what do you mean ?

wilberforce, 2017-03-08:
how are you calling `make`?
 Are you using preset env vars or putting on the make command line?

MaBecker, 2017-03-08:
env setting

    export ESP8266_BOARD=1
    export FLASH_4MB=1
    export ESP8266_SDK_ROOT=/esptools/esp_iot_sdk/sdk
    export PATH=$PATH:/esptools/esp-open-sdk/xtensa-lx106-elf/bin
    export COMPORT=/dev/cu.SLAB_USBtoUART

manual make calls

    BOARD=ESP8266_BOARD make

or 

    RELEASE=1 BOARD=ESP8266_BOARD make

wilberforce, 2017-03-08:
Thanks.

@gfwilliams 
@MaBecker has suggested making RELEASE=1 the default.

The trouble with this is I don't think there is a way to turn this off if you wanted to make a debug version?

wilberforce, 2017-03-09:
@gfwilliams 

I got to the bottom of why editing a board/*.py did not reflect the change.

It turns out that the complied files .pyc don't always get refreshed on changing the .py files.

http://stackoverflow.com/questions/15839555/when-are-pyc-files-refreshed

I've got a change here to delete the .pyc files and also put the ESP8266 settings to USE_GRAPHICS and 1600 vars
https://github.com/wilberforce/Espruino/commit/b524b629003b8ecfdcdd468e9785ae25c1d3e4da

MaBecker, 2017-03-09:
you need to call `make clean` after changing a board/*.py

MaBecker, 2017-03-09:
tried to add RELEASE, but  section of `Get info out of BOARDNAME.py` is below section where RELEASE is pared.

suggestion to add  build flags to board files

     'build' : {
       'flags': [
         'RELEASE',
       ],
       'libraries' : [
         'NET',
         'TELNET',
         'GRAPHICS',
         'CRYPTO',
         'NEOPIXEL',
       ]
     }

and parse them in scripts/get_makefile_decls.py with

     60 if "flags" in board.info["build"]:
     61     for flag in board.info["build"]["flags"]:
     62        print(flag+"=1")

wilberforce, 2017-03-09:
> you need to call make clean after changing a board/*.py
Still does not clear the cached  .pyc files

MaBecker, 2017-03-09:
ok, so this is what I do 

if 'libraries' entries change

    BOARD=ESP8266_BOARD make clean

if .py is change

    nothing, creation of pyc is handled by make becaue of timestamps

gfwilliams, 2017-03-09:
Hi, thanks for checking, but I really don't want to add RELEASE at the moment (and definitely not on a per-board basis) - maybe later but for development it's a big help as the assertions that get inserted catch a lot of problems before they become serious.

Also with Graphics - let's add that in a PR after merging. I've spent almost 2 days doing this branch and want to restrict it just to build changes - it's huge enough as it is.

But otherwise, this looks fine now, right? Shall I merge?

gfwilliams, 2017-03-09:
Seriously, let's not change the var count in this either - again, later - but if I keep adding random non-build stuff it'll never get merged. Also did't @MaBecker actually *increase* the var count, and now you want to drop it?

With the pyc files - I've *never* had an issue with this - the stackoverflow post seems to imply it just depends on the date of the files? Shall we just add it to `make clean`? Seems more sensible really as it's where all the other compiled stuff gets cleaned out?

MaBecker, 2017-03-09:
@gfwilliams  fine with me, you are the driver, so it is all up to you. 

>  @MaBecker actually increase the var count
sorry , not me ... https://github.com/espruino/Espruino/commits/master/boards/ESP8266_BOARD.py

At the end this is all about having GRAPHICS or not like explained in #1035

gfwilliams, 2017-03-09:
Ahh, sorry about that - I must have got confused with something earlier. I should have checked :)

I'll merge and then pull in your other changes. *However* I guess you need fewer vars because the fonts in the Graphics lib haven't been moved into Flash? If you did that you could probably keep going with 1700 vars?

gfwilliams, 2017-03-09:
Closing this - think I've added everything now

closed 2017-03-09 by gfwilliams

OwenBrotherwood, 2017-03-09:
A quick "low noise" comment about travis.
One thing travis doesn't do is actually check that the build works on the hardware.
One thing that I have thought was cool was that a travis build could trigger a pull of the firmware over onto a physical device that has a pre defined test program with a status back to the build.
The test device would be monitored by  another device that controlled firmware load, reset, load test program and check of result with push back to travis. (would increase travis time).

A "cool" thing if it could **reduce test time on actual hardware**: it is left up to the reader if this could be a "cool" thing to do "when one had time".

gfwilliams, 2017-03-09:
Yes - this is actually done for the original Espruino board, although it hasn't been used properly in a while. There were some discussions about it a while back as well.

I'm actually in the process of setting up a Patreon page where this will be one of the targets if I get enough funding to support it :)

OwenBrotherwood, 2017-03-09:
https://www.patreon.com/ - go for it :)