espruino/Espruino #1110: Create a additional ESP8266 board with 4MB:1024/1024

MaBecker, 2017-03-11:
Inspired by @pedro [4MB: 1024/1024](Proposed, and implemented, some enhancements (1Mb + 1Mb firmware, resizable jsvars size, ...)) it is time to create a ESP8266_4MB_BOARD.py to have more space for firmware, modules and flash and keep ESP8266_BOARD.py as is.

* ESP8266_4M_BOARD.py
* add Graphics and Filesystem   
* use 1MB for save()
* 1MB to use with "Flash" 
* 1600 variables
* and about 9500 freeHeap 

@gfwilliams please add label ESP8266

gfwilliams, 2017-03-13:
When you do this, please can you also make a `1M_combined` image? It'd save a lot of pain for beginners, as then you just have one single image to upload which *also* wipes the saved code.

wilberforce, 2017-03-13:
@MaBecker 

ESP8266_4MB_BOARD.py

Perhaps should be ESP8266_4MB.py

1mb for save area seems a lot!

MaBecker, 2017-03-13:
> make a 1M_combined image

Good point!  Yes of course. 
Note: wiflash.sh can only handle single bin files

> Perhaps should be ESP8266_4MB.py

ok so it is  ESP8266_4MB.py

gfwilliams, 2017-03-13:
Good point. 1MB does seem like overkill. I guess even 50k would be more than enough.

Personally I'd say you should pack everything into the first 1MB (including saved code), and then you can leave a full 3MB for the user to do what they want (including 1MB for wifi firmware updates *and* still 2MB free for user data).

MaBecker, 2017-03-13:
> 1mb for save area seems a lot!

>1MB does seem like overkill. I guess even 50k would be more than enough.

so we have 13 pages for save

using 4MB: 1024/1024 gives you 2 times 1 MB  user flash

````
  chip['saved_code'] = {
    'address' : 0x201000, # first page is used for wifi save 
    'page_size' : 4096,
    'pages' : 13,
    'flash_available' : 912, # firmware can be up to this size 
   };
````

and add the rest can be added as free Flash Areas, not sure if it makes sense to have to big areas

````
  .....
  uint32_t map = system_get_flash_size_map();
  if ( map == 6 ) {
     addFlashArea(jsFreeFlash, 0x214000, 0x200000-0x14000);
     addFlashArea(jsFreeFlash, 0x300000, 0x300000-0x4000);
    return jsFreeFlash;
  }
  .....  
````

gfwilliams, 2017-03-13:
Just wondering - could we not just create a 1MB image? It would then be compatible with all 1MB devices too - so pretty much everything, it just wouldn't allow OTA firmware updates on 1MB parts?

MaBecker, 2017-03-17:
ok, fs why not, just have to check that there is enough heap space left.

gfwilliams, 2017-03-17:
fs *shouldn't* take up heap. I guess if it does, it might be a code page table that does - but it should be easy enough to put in flash.

MaBecker, 2017-03-17:
>fs shouldn't take up heap 

perfect, so module fs set too.

MaBecker, 2017-03-18:
#1118

Kyklas, 2017-03-27:
I am looking for a greater save() size. I use 1183 JSVars and have 9k of code so 12KB of storage is not enough.

opichals, 2017-03-27:
Hi @MaBecker! Would you mind sharing the board file and other required changes before it gets to master?

MaBecker, 2017-03-27:
@opichals check https://github.com/MaBecker/Espruino

    FLASH_4MB_C1=1 make
    FLASH_4MB_C1=1 make flash

MaBecker, 2017-03-27:
@Kyklas  ok lets start with  16 pages for save()

MaBecker, 2017-03-27:
@gfwilliams  continue working on this issue - please let me know if this is the correct way.

My understanding is that  

    BOARD=ESP8266_4MB  make 
 
will create a `4MB: 1024/1024` firmware

* new ESP8266_4MB.py file with 4MB settings 
    ...
   'package' : "4MB",
    ...

* adds to Makefile

    ...
    else ifeq ($(FAMILY),ESP8266)
    USE_ESP8266=1
    ifeq ($(PACKAGE),4MB)
    FLASH_4MB=1
    endif
    ...

* make/targets/ESP8266.make

    ...
    ifdef FLASH_4MB
        LD_SCRIPT1  = ./targets/esp8266/eagle.app.v6.new.2048.app1.ld
        LD_SCRIPT2  = ./targets/esp8266/eagle.app.v6.new.2048.app2.ld
    else 
        LD_SCRIPT1  = ./targets/esp8266/eagle.app.v6.new.1024.app1.ld
        LD_SCRIPT2  = ./targets/esp8266/eagle.app.v6.new.1024.app2.ld
    endif
    ...

and so on....

gfwilliams, 2017-03-27:
Sounds good - but I'd ditch the change to the Makefile and would put `FLASH_4MB=1` straight in the board file, like is done here:

https://github.com/espruino/Espruino/blob/master/boards/PUCKJS.py#L43

MaBecker, 2017-03-27:
THANKS, that makes it very simple!

MaBecker, 2017-03-28:
Next steps:

* adopt flash areas

* try to create a single 1MB flash file

Kyklas, 2017-04-03:
I would like to test a firmware with greater storage.
I looked at the build instruction to compile something myself but I run into issues. Perhaps I can work with someone to go thru the process and update the instructions : https://github.com/espruino/Espruino/blob/master/README_Building.md. Should I open a new ticket ?

@MaBecker if you have a image available I would like to try. I have ESP-12F devices i.e 4MB flash.

MaBecker, 2017-04-03:
Hi @Kyklas,  

sorry not now, at the moment I am very busy with other things.

lancernet, 2017-04-04:
@MaBecker  hi, I follow the method you say, compile, but the system_get_flash_size_map function always returns 4, what is the problem?

MaBecker, 2017-04-14:
> Just wondering - could we not just create a 1MB image? 

No because default and blank are not in first partition for a 4MB  1024/1024 

[image]

but it is possible to add  a file to blank save and wifi section, which will have the same effect.

MaBecker, 2017-04-14:
Or, send a 1MB file plus esp_init_data_default.bin and blank.bin, will try that

lancernet, 2017-04-14:
thx，but where is esp_init_data_default.bin and blank.bin downloading？

MaBecker, 2017-04-14:
@gfwilliams not sure if this is what you have in mind

`make flash`  will always flash a combined file plus addons for a 4MB board like this:

```
flash: $(ESP_COMBINED)
ifndef COMPORT
	$(error "In order to flash, we need to have the COMPORT variable defined")
endif
	-$(ESPTOOL) --port $(COMPORT) --baud $(FLASH_BAUD) write_flash --flash_freq $(ET_FF) --flash_mode qio --flash_size $(ET_FS) 0x0000 $(ESP_COMBINED) $(ESP_FLASH_ADDONS)

```

MaBecker, 2017-04-14:
Splitting 2x1MB up in 4x256k on second partition and 1x1008k on forth partition  - any comments?

MaBecker, 2017-04-14:
For those who like to sneak in, just  committed  to my Espruino fork.

wilberforce, 2017-04-14:
@MaBecker - please add:
https://github.com/MaBecker/Espruino/blob/master/boards/ESP8266_4MB.py#L25-L31

```
     'FILESYSTEM',
     'FLASHFS'	
```

This will use the last 1Mb, at location 3Mb as a file system.

MaBecker, 2017-04-15:
@wilberforce - fyi:  the third partition is not completely free on a 4MB 1024/1024, but second is totaly free, so I will change the addFlashArea settings

like this:

    addFlashArea(jsFreeFlash, 0x200000, 0x10000);
    addFlashArea(jsFreeFlash, 0x300000, 0x40000);
    addFlashArea(jsFreeFlash, 0x340000, 0x40000);
    addFlashArea(jsFreeFlash, 0x380000, 0x40000);
    addFlashArea(jsFreeFlash, 0x3C0000, 0x3C000);

MaBecker, 2017-04-15:
adding FILESYSTEM and FLASHFS eats to much heap space, is there optimization potential in this libraries?

````
'libraries' : ['NET','TELNET','GRAPHICS','NEOPIXEL']

    "cpuFrequency": 160, "freeHeap": 10536, "maxCon": 10, <- nice

'libraries' : ['NET','TELNET','GRAPHICS','NEOPIXEL', 'FILESYSTEM']

    "cpuFrequency": 160, "freeHeap": 8688, "maxCon": 10, <- possible

'libraries' : ['NET','TELNET','GRAPHICS','NEOPIXEL', 'FILESYSTEM','FLASHFS']

    "cpuFrequency": 160, "freeHeap": 5168, "maxCon": 10,  <- this might be too small
````

wilberforce, 2017-04-15:
@MaBecker 

Unfortunately with 4096 flash page size,. it also requires 4096 of RAM as the current buffer for read and write operations. So it will always require at least this additional space. Possibly some of the strings could be moved to ROM, however I'm not sure it is worth the effort.

MaBecker, 2017-04-16:
Summary:
- [x] new board ESP8266_4M with map mode 6 : 1024/1024
  - [x] use partition 1 and 2 for firmware 
  - [x] allow firmware up to 940KB
  - [x] save source code in partition 1
  - [x] save wifi setting  in partition 1
  - [x] create combined 1024K firmware image to wipe source and wifi settings
  - [x] add flash areas 
    - [x] partition 3: 1 x 1MB 
    - [x] partition 4: 3 x 256K and 1 x 240K
  - [x] define 1600 vars
  - [x] define 16 pages for save code
  - [x] add flash_combined and flash_erase to make/targets/ESP8266.make
  - [ ] wiflash is breaking save sections, no idea.....
- [ ] additional libraries
  - [x] GRAPHICS, freeHeap: 9512
  - [ ] ...
- [ ] build option RELEASE=1
  - [ ] GRAPHICS, freeHeap: 102
  - [ ] ...

MaBecker, 2017-04-17:
Have to give up to save  wifi and code in first partition, because `make wiflash` wipes that.

* Moving to forth partition, to keep third as 1MB.
* build a 4MB combined image, so everything can be  included

MaBecker, 2017-04-17:
adding 'FILESYSTEM' and 'FLASHFS' make no sense, because it is using to much heap space.

I think [Flash storage Module, any js type , web pages, css, js and images](http://forum.espruino.com/conversations/283045/ ) or something similar would be nice to have for systems with poor resources like the ESP8266.

MaBecker, 2017-04-18:
@gfwilliams  please take a brief look and give some advice to make sure that this is still going in the right direction - thanks

MaBecker, 2017-04-18:
#1142

gfwilliams, 2017-04-19:
Looks fine to me...

closed 2017-04-19 by MaBecker

Kyklas, 2017-05-09:
@MaBecker : Thank you for the release of the 64k flash save memory.

@ALL : I have downloaded and flash the new firmware on my ESP8266 yesterday. Unfortunately I am still facing an "Out of Memory" errors. I think i may have just too much JS Code/Vars for the ESP to handle. I would like to know how much JS Code I can have in the ESP RAM. The IDE reports around 9k of JS Code sent. I am using the MQTT module and using it with so much code leads to errors. I am using close to 700+ JS Vars as well. How is the RAM Memory of the ESP8266 split between JS Code and JS Var ?

Could http://www.espruino.com/EspruinoESP8266 be updated to bring information about the limitation of the ESP8266 ? 

I may go for the ESP32 since the ESP8266 limitations are reach with program with moderate complexity.