espruino/Espruino #1116: Allow FS (fatfs) API to work on ESP8266's flash

gfwilliams, 2017-03-14:
This would make a lot of sense for the 4M parts (https://github.com/espruino/Espruino/issues/1110)

I think this got shot down a while ago with concerns about repeated writes to flash, but I think realistically the use it would be to everyone would far outweigh any problems.

So either the fatfs library could be tweaked such that the block read/write functionality wrote flash blocks (which I imagine would be easy), or SPIFFS could be ported - but that'd mean re-writing every file IO function to use SPIFFS :(

wilberforce, 2017-03-14:
Hi Gordon,
If I recall when @tve looked at this it was to due with the 4096 byte pages vs 512 bytes sector size.

I have been thinking about storage recently and looking at options.

The esp32 has a number of options that could be of use...

1. There is a nvs storage - 
Non volatile storage that uses a key and stores a value. All sorts of types are supported, however I thought the best thing to use would Ben the binary blob...

So the idea is to take a js object - json stringy, compress with heat shrink, and then store the blob against the key.
For retrieval, it would be lookup, decompress, and restore into a js object. This way we can support, all sorts of complex js types - nested structures etc, and also functions and modules....

2. The esp32 also supports vfs. A virtual file system. This is cunning in that if links into newlib, so that the standard posix `fopen` hook into this transparently.

They have an sd card fat example that uses this.

This means that the Lunix  implemention of the `fs` Library could be made to use this, and the layer of mapping to sd card and fat done via vfs. There is some talk of a spiffs implementation, however have not seen that yet. Cesanta have a spiffs library on github for the esp32/8266. The lua fold have added too.

3. There is a read only file system https://github.com/Spritetm/libesphttpd/tree/master/espfs
This uses a tools to make an gpio file and copy to to flash. The API then provides read only access.

4. what every file system is implemented, and easy way of updating would be required. As we already have a webserver implementation, I think a http PUT method would be a good way of getting files onto the device, or even a page that files could be dargged and dropped onto..

wilberforce, 2017-03-14:
I was looking `fs` libs last night and the implemention is #ifdef Linux.... #else 

If this was abstracted the same way as the network library, with a function table it would be more extensible and easier to add new layers

wilberforce, 2017-03-14:
Here is the nvs stuff on the esp32

https://gitter.im/espruino/esp32?at=58c66411dd08b4b859c2c560

At this stage just working with i32

gfwilliams, 2017-03-14:
Yes, some kind of function table could work. I guess it depends whether we have to support different filesystem types *with the same build*? If not we could do like jsHardware - create a common API and then just compile in the implementation that's needed? Swapping to the function table from that could be easy to do later though.

Personally, I don't think adding ESP32-specific stuff is a good idea at the moment - 99% of people with this problem right now are using ESP8266, so ideally we want some solution which handles both.

It's fine just to set the block size to 4096b in FatFS - which would solve this? http://elm-chan.org/fsw/ff/en/appnote.html

It would require 4k of RAM temporarily to access files, but it's not the end of the world. Or there's so much free flash available it wouldn't be the end of the world to just ignore the other 3500 bytes.

I kind of like that approach, because we can do it using the existing Espruino flash APIs. The filesystem could then be supported on *any* device that Espruino runs on if it has spare pages of flash - not just ESP8266/32.

For writing, a simple JS webserver that handles posting data should be fine - maybe as a library. I'd posted some example code on the forum a year or so ago.

Also at some point I'll be looking at adding a built-in server to allow console over websocket, and I guess at that point we could look at adding a way to upload files with it.

wilberforce, 2017-03-14:
Here is a js post server that emulates the esp32 ota server that is in firmware.

https://github.com/wilberforce/Espruino/blob/ota/targets/esp32/tests/ota-server.js

I was thinking something like this could be extended, so not only handle ota updates, but update files in a storage system.

wilberforce, 2017-03-14:
So it looks like these set of functions would need to be set up to use the inbuild flash libs:

https://github.com/espruino/Espruino/blob/master/libs/filesystem/fat_sd/diskio.h#L34-L38

wilberforce, 2017-03-15:
http://elm-chan.org/fsw/ff/en/appnote.html#port

Function    Required
- [x] get_fattime 
- [ ] disk_status
- [ ] disk_initialize
- [ ] disk_read	Always	
- [ ] disk_write
- [ ] disk_ioctl (CTRL_SYNC)	             
- [ ] disk_ioctl (GET_SECTOR_COUNT)  
- [ ] disk_ioctl (GET_BLOCK_SIZE)	     
- [ ] disk_ioctl (GET_SECTOR_SIZE)     	
```
DSTATUS disk_initialize (BYTE pdrv);
DSTATUS disk_status (BYTE pdrv);
DRESULT disk_read (BYTE pdrv, BYTE* buff, DWORD sector, UINT count);
DRESULT disk_write (BYTE pdrv, const BYTE* buff, DWORD sector, UINT count);
DRESULT disk_ioctl (BYTE pdrv, BYTE cmd, void* buff);
```

Use as template for starting point
https://github.com/espruino/Espruino/blob/master/libs/filesystem/fat_sd/sdio_diskio.c

gotchas:
- [ ]  Write will need to pad to 4 bytes?

wilberforce, 2017-03-19:
It doesn't look like much, however have made a start here:
https://github.com/espruino/Espruino/compare/master...wilberforce:flash_fs

gfwilliams, 2017-03-20:
Nice - does it work?

I didn't realise there was a `mkfs` command - I guess we could expose that as a special `format` function?

Probably for the final version I'd add a new `flash_diskio.c` file, since on the majority of platforms it's actually `spi_diskio.c` that gets included, not the spi one (iirc). 

It sounds like a good idea to force people to initialise it, yes. To stick with `E.connectSDCard` maybe do `E.connectFlashFS`? I guess we could make it check what areas of memory are free (jshardware.c has jshFlashGetFree or something similar) if nothing is specified? It'd have to know how much memory it could use as well :)

wilberforce, 2017-03-20:
No - it doesn't work ;-(
I'm getting crashes at the moment when I try to mount the card.

I have set up a test harness that attempts to make the file system. It's going to be pretty slow as all of the pages get set to zero as part of the format - only to clear them back to FF before the next write..

It is spi_diskio.c I'm changing at the moment - Ideally want to make it switchable so depending on what call you make, you can connect to flash or an SD card.

BTW - what command do you use to format the SD card in linux? I've been trying to make an img, and then flash that and then try to read from it..

``
dd if=/dev/zero of=fat.fs bs=1024 count=1024
mkfs.vfat -v -F 16 -S 4096 -s 1 fat.fs.img
```
This spits out errors:
```
WARNING: Not enough clusters for a 16 bit FAT! The filesystem will be
misinterpreted as having a 12 bit FAT without mount option "fat=16".
fat.fs.img has 64 heads and 32 sectors per track,
hidden sectors 0x0000;
logical sector size is 4096,
using 0xf8 media descriptor, with 256 sectors;
drive number 0x80;
filesystem has 2 16-bit FATs and 1 sector per cluster.
FAT size is 1 sector, and provides 249 clusters.
There is 1 reserved sector.
Root directory contains 512 slots and uses 4 sectors.
Volume ID is a849e3a6, no volume label.
```

So I really want to start from a known state.

gfwilliams, 2017-03-20:
That command looks ok to me as far as I know (however I guess fat.fs and fat.fs.img should be the same?).

I guess another option would be to implement jshFlash* on Linux (just writing to a file) and then you could run and debug much more easily?

wilberforce, 2017-03-20:
Yay - success (of sorts).
https://github.com/espruino/Espruino/compare/master...wilberforce:flash_fs?expand=1

The files system can be formatted, and files written.

```
>var files = require("fs").readdirSync();
WARNING: Flash Init - disk_initialize 0
WARNING: Flash disk_read sector: 0, buff: 1073526768 len: 4096
WARNING: jsfsInit - appears to f_mount!
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
WARNING: Flash disk_read sector: 2, buff: 1073526768 len: 4096
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0

=[
  "hello.txt",
  "hello2.txt",
  "data.csv"
 ]
```

however, it is crashing on file read - which is weird since the folder structures etc can be read fine.

I thought it might be a 4 bytes boundary thing, but the jshFlashRead takes care of single byte reads.... I guess I could check to see if the reads are starting on a boundary?

gfwilliams, 2017-03-20:
I don't know - maybe print out exactly what addresses and sizes you're reading and see if anything looks odd about the one that crashes.

Good sign though!

wilberforce, 2017-03-21:
I've spent quite a bit of time on this today - it's very frustrating.

Reading back the flash:

```
f=require("Flash");
b=0x300000;
function rSect(s){
uintArray=f.read(64,b+s*4096);
+console.log(String.fromCharCode.apply(null, uintArray));
}
rSect(34);
```

It looks like all the bits are getting put in the write places... however still crashes on file reads on the ESP32 -so it could be be some other issue?

I guess I could see it compiles on the esp8266 and if the same problem .

Downloading the 1M block and trying to read the img file does not work either - so it could be some sort of corruption with what is written.

Anyway, it's a starting point if anyone is keen for the challenge.

gfwilliams, 2017-03-21:
I just committed some flash emulation functions to the Linux port - so if you're able to compile your code for linux then you should be able to see how it works on the PC (might need some bodging in jswrap_fs to force it to use the SPI FS functions). 

It'll be much easier to debug and it should even trap segfaults/etc :)

wilberforce, 2017-03-21:
It's interesting how comments trigger thought processes... following what you said about fs_wrap lead to here... I think this read is the issue...

https://github.com/espruino/Espruino/blob/e120ce1a7f86639623edfed460c1fdaa12e9db1e/libs/filesystem/jswrap_fs.c#L251

gfwilliams, 2017-03-21:
That read itself *should* be safe since you don't want the user top be able to segfault by calling `File.read(0x7fffffff)` - but it could still be causing you some trouble for another reason I guess :) Maybe something like `ftell` is broken?

wilberforce, 2017-03-21:
Found it. argg,h. It was a buffer overflow and no safety checks in the ff.c

```
#define	_MIN_SS		512
#define	_MAX_SS		512
```

to
```
#define	_MIN_SS		512
#define	_MAX_SS		4096
```

https://github.com/espruino/Espruino/blob/master/libs/filesystem/fat_sd/ffconf.h#L174-L181

According to the comment - get sector size is implemented so we should be able to leave _MAX_SS		4096 - unless you want it in an ifdef?

here is some output anyway...

```
>E.flashFatFs(0x300000,2);
WARNING: E.flashFatFs addr: 3145728 format: 2
WARNING: Flash Init - disk_initialize 0
WARNING: Flash disk_ioctl 2
WARNING: disk_ioctl FS_SECTOR_SIZE 4096
WARNING: Flash disk_read sector: 0, buff: mem: 3145728 buff: 1073527048 len: 4096
WARNING: after read

ERROR: Unable to mount SD card : NO_FILESYSTEM
WARNING: jsfsInit: 0
=undefined
>E.flashFatFs(0x200000,1);
WARNING: E.flashFatFs addr: 2097152 format: 1
ERROR: E.flashFatFs formatting...
WARNING: Flash Init - disk_initialize 0
WARNING: Flash disk_ioctl 2
WARNING: disk_ioctl FS_SECTOR_SIZE 4096
WARNING: Flash disk_ioctl 1
WARNING: disk_ioctl FS_SECTOR_COUNT 256
WARNING: Flash disk_ioctl 3
WARNING: disk_ioctl FS_BLOCK_SIZE 1
WARNING: Flash disk_write sector:  0, buff: mem: 2097152 buff: 1073527048 len: 4096
WARNING: Flash disk_write sector:  1, buff: mem: 2101248 buff: 1073527048 len: 4096
WARNING: Flash disk_write sector:  2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: Flash disk_write sector:  3, buff: mem: 2109440 buff: 1073527048 len: 4096
WARNING: Flash disk_write sector:  4, buff: mem: 2113536 buff: 1073527048 len: 4096
WARNING: Flash disk_write sector:  5, buff: mem: 2117632 buff: 1073527048 len: 4096
WARNING: Flash disk_ioctl 0
WARNING: Flash disk_ioctl CTRL_SYNC
=undefined
>require("fs").writeFileSync("hello.txt", "Hello World\nHello World\nNew line\n");
WARNING: Flash Init - disk_initialize 0
WARNING: Flash disk_ioctl 2
WARNING: disk_ioctl FS_SECTOR_SIZE 4096
WARNING: Flash disk_read sector: 0, buff: mem: 2097152 buff: 1073527048 len: 4096
WARNING: after read

WARNING: jsfsInit - appears to f_mount!
WARNING: Flash Init - disk_status 0
WARNING: Flash disk_read sector: 2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: after read

WARNING: Flash Init - disk_status 0
WARNING: Flash disk_write sector:  2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: Flash disk_read sector: 1, buff: mem: 2101248 buff: 1073527048 len: 4096
WARNING: after read

WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
WARNING: Flash disk_write sector:  6, buff: mem: 2121728 buff: 1073550728 len: 4096
WARNING: Flash disk_write sector:  1, buff: mem: 2101248 buff: 1073527048 len: 4096
WARNING: Flash disk_read sector: 2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: after read

WARNING: Flash disk_write sector:  2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: Flash disk_ioctl 0
WARNING: Flash disk_ioctl CTRL_SYNC
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
=true
>var files = require("fs").readdirSync();
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
=[
  "hello.txt"
 ]
>console.log(require("fs").readFileSync("hello.txt")); // prints "Hello World"
WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
WARNING: Flash disk_read sector: 6, buff: mem: 2121728 buff: 1073550284 len: 4096
WARNING: after read

WARNING: Flash Init - disk_status 0
WARNING: Flash Init - disk_status 0
Hello World
Hello World
New line

=undefined

```

wilberforce, 2017-03-21:
further good news....
```
..\esp-idf\components\esptool_py\esptool\esptool.py --chip esp32 --port COM3 --baud 921600  read_flash 0x200000 0x100000 fs.img
```

And the on windows opened with `OSFMount`. This mounts it as a drive... copied the README.md

```
..\esp-idf\components\esptool_py\esptool\esptool.py  --chip esp32 --port COM3 --baud 921600 --before esp32r0 --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect  0x200000 fs.img
```

```
>var files = require("fs").readdirSync();

=[
  "hello.txt",
  "README.md"
 ]
>console.log(require("fs").readFileSync("README.md"));
Espruino JavaScript for Microcontrollers
========================================
<pre>
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_|
</pre>
http://www.espruino.com &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; )
```

wilberforce, 2017-03-21:
I copied a module onto the SDcard:
```
>console.log(require("fs").readFileSync("/DS18B20.js"));
var C={CONVERT_T:0x44,COPY:0x48,READ:0xBE,WRITE:0x4E};function DS18B20(oneWire,}
if(!this.sCode)throw new Error("No DS18B20 found");this.type=parseInt(this.sCod}
DS18B20.prototype._r=function(){var b=this.bus;b.select(this.sCode);b.write(C.R)
c=(c>>1)^0x8C*(c&1);}
var temp=null;if(c==s[8]){temp=s[0]+(s[1]<<8);if(temp&32768)temp-=65536;temp=te}
if(callback)callback(temp);return temp;}
this.bus.select(this.sCode);this.bus.write(C.CONVERT_T,true);if(!callback)retur}
DS18B20.prototype.searchAlarm=function(){return this.bus.search(0xEC);};DS18B20;
=undefined

>d=require("DS18B20");
WARNING: Flash Init - disk_status 0
WARNING: Flash disk_read sector: 2, buff: mem: 2105344 buff: 1073527048 len: 4096
WARNING: after read

ERROR: Could not open file : NO_PATH
WARNING: Module "DS18B20" not found
=undefined

```

No sure what the path or syntax is meant to be for this?

wilberforce, 2017-03-21:
@MaBecker 
I'll clean this up and then you can try it out on the ESP8266!

gfwilliams, 2017-03-21:
I think it looks in a 'node_modules' folder?

That's great news though! 
`_MAX_SS` would definitely need ifdefing - it's used here afaik: https://github.com/espruino/Espruino/blob/master/libs/filesystem/fat_sd/ff.h#L105

Which is statically allocated in jswrap_file.c - so by increasing it you're using 3.5kB of RAM all the time.

I guess potentially:

* We could look at only allocating that buffer out of jsVars when needed, which could be good for all platforms
* Maybe on devices where you care about the RAM (everywhere apart from ESP32?) you could just use 512 byte blocks and waste 3.5k of each flash block
* If we know we'll always have 4k of free RAM on the stack then we could still write 512 byte blocks but could 'emulate' block erase by  reading 4k, erasing, then writing a full 4k again

wilberforce, 2017-03-22:
I have used #ifdef for the sector side, so the other builds are not affected.

https://github.com/espruino/Espruino/compare/master...wilberforce:flash_fs?expand=1

The code is getting close to ready to merge.

- [ ]  move the flash*.bat to scripts - these are windows flashing scripts - move to /scripts/eps32/ ?

`E.flashFatFs( { addr: 0x200000, sectors : 100, readonly: true } );`

- the flash fs is auto mounted on demand and formatted on first use, at the default address. This can be overidden using  `E.flashFatFs`
- [ ] readonly not yet implemented.

The Fatfs also supports disks.... `0:disk0_file.txt` `1:disk0_file.txt` so we have the potential of using system and user volumes - like modules on 0: and user files on another....

gfwilliams, 2017-03-22:
Nice, thanks! Looks to me like `disk_ioctl` might need to be aware of the different sector sizes too?

It might be an idea to just disable it by default or at least make it read-only? as-is, I imagine just trying to use a module that doesn't exist would cause 1MB of flash memory to get formatted?

I'd steer clear of multiple disks - that just sounds like pain (i mean, that's what directories are for?). If we did do that, IMO we should focus on allowing normal SD cards and the flash memory to coexist on separate volumes first as that could be really useful.

Please could we move jsDebug to just commented out jsWarns for now? seems like overkill to add a new global debug macro just for this?

wilberforce, 2017-03-22:
I made the jsDebug macro global, so it can be used anywhere - not just with this code.
It is useful when you need to see the details, and is removed when `RELEASE=1`

I don't have any gdb debugging, so was using this. I can make it local to these files if required.

The `fat_initialised` flag is off at startup, so a missing module should not trigger a format. I'll check this.

I just wanted to make it easy to use, by turning on if you used fs or file...

wilberforce, 2017-03-23:
To Do:

- [ ] Fix create fs if missing module, init call required
- [ ] readonly mode - seems to be compile only option?
- [ ] mount command - format param
- [ ] turn off other E.SD stuff ifndef   E.connectSDCard  E.unmountSD
- [ ] new File("hello world.txt").pipe(res); fails  http://www.espruino.com/File+IO#line=34,49,59   No File.pipe method
- [ ] add mkdir
- [ ] test mkdir on linux

Pull request https://github.com/espruino/Espruino/pull/1119

wilberforce, 2017-03-23:
Working on the ESP8266!

For this build:
https://github.com/espruino/Espruino/issues/1110

Add to the BOARD.py:
     'NEOPIXEL',
     'FILESYSTEM',
     'FLASHFS'		 
   ]

```
http://espruino.com/Donate
Flash map 4MB:512/512, manuf 0xe0 chip 0x4016
>fs=require("fs");
=function () { [native code] }
=undefined
>if ( typeof(fs.readdirSync())==="undefined" ) {
:  console.log("Formatting FS");
:  E.flashFatFS({format:true});
:}
=undefined
=undefined
>fs.readdirSync();
=[
  "hello world.txt",
  "node_modules"
 ]
>require("fs").writeFileSync("hello world.txt", "This is the way the world ends\nHello World\nnot with a bang but a whimper.\n");
=true
=undefined
>fs.readdirSync();
=[
  "hello world.txt",
  "node_modules"
 ]
```

gfwilliams, 2017-03-23:
Nice, thanks! So do you think that PR is ready to go?

Yeah, I'm just a bit iffy about adding the jsDebug thing for what's one thing when the rest of Espruino doesn't use it. I think if *everything* used it it'd just spam the console constantly, when realistically what you want is just to enable debugging for one specific feature at a time? Please could you add it as a local decl?

Agreed about making it easy to use, but it's worth checking as I think the module loader might try and auto-mount. Maybe we don't care, or maybe, before formatting automatically you could check if the area is all 0xFF - so that it wouldn't overwrite anything if someone had put it there?

wilberforce, 2017-03-23:
> Nice, thanks! So do you think that PR is ready to go?

yup

> Yeah, I'm just a bit iffy about adding the jsDebug thing
It should be local and commented out now. 

> module loader might try and auto-mount.
Yes it does auto-mount - however, now on failure it does not format. 
This is the same behaviour as the linux build. e.g if the file system is valid you can just use it.
This keeps it simple.

If you want to auto format, this js code will do it:

```
fs=require("fs");
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("Formatting FS");
  E.flashFatFS({format:true});
}
```

I have tested the build on ESP32 and ESP2866 on the hardware. And also tried the linux build as there is now a `fs.mkdir`.  I used this to create a `node_modules` folder and have confirmed a module can be loaded from there - there is a /eps32/test script which demos getting a module and storing it in flash.

I think this is going to be a game-changer on the EPS32. The recent Webserver module and the ability to store files and serve web pages will be quite powerful.

I've still had issues getting the File.pipe to work - I'm not sure if the recent change you did for the http request solved the issue? I could not get the http pipe file example on the reference pages to work. I didn't try on linux mind.

gfwilliams, 2017-03-27:
Sorry - only just got around to this. Looks great - just merged.

hungryforcodes, 2017-04-04:
I managed to build some binaries for the ESP8266  on a 4MB flash board (1v91.736 as mentioned on Gitter), and for the most part the basic file stuff seems to work (fs), though I'm still testing it with larger and larger files. But the File Class stuff seems dodgy or not working. When I do something like (simplified from my code): 

<pre>
`
function pageHandler(req,res) {
     res.writeHead(200);
     require("fs").pipe("htmlTest.html",res);
     res.end()
}
</pre>`
 I get "Uncaught Error: Source object does not implement the required read(buffer, length, position) method.". 

If I use the other way of piping (according to the docs):

<pre>
`
function pageHandler(req, res) {
      res.writeHead(200);
      html=E.openFile("htmlTest.html","r");
      html.pipe(res);
      res.end();
}
`
</pre>

I get "Uncaught Error: Expecting a function to call, got Number".

If I do

`console.log(require("fs").readFileSync("htmlTest.html"));`

it displays fine. 

So pipe just doesn't seem to work at this point. Its a big improvement from before though! :)

Also now I'm getting the occasional reset after "save()". That didn't happen before, but it doesn't seem to occur again after reset for the moment. It may or may not be related to this or other modifications that have been made recently to Espruino, obviously. And again it seems to be only after a "save()" and it is not consistent. 

<pre>
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)
load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8
2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
ãìgìsûo<äld`ârdþLoading 6523 bytes from flash...
`
</pre>

Also just for my information, maybe I missed it -- on a 4MB module it will allocate a filesystem of 1MB in size? Thanks! ;)

wilberforce, 2017-04-04:
I remember having issues with the docs..  take a look at the post here:

http://forum.espruino.com/conversations/301910/

For the pipe to work, end has to be called...

wilberforce, 2017-04-05:
Here are some useful snippets that might be of use for documentation of read and write flash FS image files:

From web ide  - format FS if needed
=======================
```
fs=require("fs");
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("Formatting FS");
  E.flashFatFS({format:true});
}

Read and write images from flash
=====================
```
Read the flash file system (previously formatted) and copy as local disc image
`esptool.py --chip esp32 --port COM3 --baud 921600  read_flash 0x300000 0x100000 fatfs.img`

Write a disc image
`esptool.py  --chip esp32 --port COM3 --baud 921600 --before esp32r0 --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x300000 fatfs.img`install tools to allow img manipulation with mounting the disc image - also works on windows 10 bash on linux

For web ide for testing:
===============

```
fs=require("fs");
fs.readdirSync();

require("fs").readFileSync("hello world.txt");
require("fs").readFileSync("flash-erase.bat");

fs=require("fs");
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("Formatting FS");
  E.flashFatFS({format:true});
}

fs.readdirSync();
```

Linux tools to manipulate images
=====================
```
apt-get install dosfstools  mtools
echo MTOOLS_SKIP_CHECK=1 >  ~/.mtoolsrc
```

```
# list files
mdir -i fatfs.img

# Copy local file README.md  onto flash FS image
mcopy -i fatfs.img README.md  ::

# research - option of a recursive copy...
```

```
fsck.fat -v fatfs.img

fsck.fat 4.1 (2017-01-24)
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "MSDOS5.0"
Media byte 0xf0 (5.25" or 3.5" HD floppy)
      4096 bytes per logical sector
      4096 bytes per cluster
         1 reserved sector
First FAT starts at byte 4096 (sector 1)
         1 FATs, 12 bit entries
      4096 bytes per FAT (= 1 sectors)
Root directory starts at byte 8192 (sector 2)
       512 root directory entries
Data area starts at byte 24576 (sector 6)
       250 data clusters (1024000 bytes)
63 sectors/track, 255 heads
         0 hidden sectors
       256 sectors total
Checking for unused clusters.
fatfs.img: 2 files, 2/250 clusters
```

```
minfo -i fatfs.img
device information:
filename="fatfs.img"
sectors per track: 63
heads: 255
cylinders: 1

mformat command line: mformat -T 256 -i fatfs.img  -h 255 -s 63 -H 0 -S 5 ::
```

Create a new disc image
================

```
dd if=/dev/zero of=espruino.img bs=1024 count=1024
mformat -T 256 -i espruino.img  -h 255 -s 63 -H 0 -S 5 ::
minfo -i espruino.img

bootsector information
banner:"MSDOS5.0"
sector size: 4096 bytes
cluster size: 1 sectors
reserved (boot) sectors: 1
fats: 1
max available root directory slots: 512
small size: 256 sectors
media descriptor byte: 0xf0
sectors per fat: 1
sectors per track: 63
heads: 255
hidden sectors: 0
big size: 0 sectors
physical drive id: 0x80
reserved=0x0
dos4=0x29
serial number: C2210000
disk label="NO NAME    "
disk type="FAT     "
```

hungryforcodes, 2017-04-05:
Wow thanks! This is very useful - especially reading the fs and bringing it in locally. :)  I've been testing out the linux espruino image as well on my desktop. I will have a look at the new pipe code from the discussion you linked to later tonight and post if I have success! :)

gfwilliams, 2017-04-06:
The pipe error is fixed now - however I'm pretty sure pipe doesn't need `end` called - it does it automatically when the file is finished - so I think the example here is correct: http://www.espruino.com/Internet#transferring-files

hungryforcodes, 2017-04-09:
So some encouraging news on the pipe feature, though there remain issues overall. Lets start with the bad news. If I do a simple: 
<pre>
 require("fs").writeFileSync("hello.txt", "Hello World\nHello World\nNew line\n");
</pre>
and then after: 
<pre>
console.log(require("fs").readFileSync("hello.txt"));
</pre>

this works as expected. However when I use the above code example to get the file, the request starts but no data is sent. All I get in the network section of Chrome is "pending" and no data. So it would SEEM like pipe doesn't work. However I decided instead to create my own large example on the 8266. I used this code to kind of stress test the file system: 
<pre>
function writeTest() {
  console.log(process.memory());
  for (i=0; i<10000; i++) {
  var f = E.openFile("data.csv","a");
    var readings = [
      i,
      i,
      i,
      i
    ];
    f.write(readings.join(", ")+"\n");

    f.close();
  }
    console.log("done writing");
    console.log(process.memory())
}
</pre>

This was a modification of Gordon's code elsewhere. So first some good news about this -- it worked perfectly, and the memory overhead was minimal -- so its a useful "append feature" for large files if you split them up into chunks and do this. Big win :) 

The resulting file was over 200K in length. It seemed to overflow the console buffer (memory errors) when doing a standard read:
<pre>
  console.log(require("fs").readFileSync("data.csv"));
</pre>
So I wrote the following code using setTimeout to give it some space to display, and it worked flawlessly. There were no read errors, and obviously no write errors as the data was not garbled in any way: 
<pre>
function consoleLongFile() {
   var f = E.openFile("data.csv","r");
    if (typeof f === 'undefined') {
      console.log("no file found");
      return;
    }
    var data;
    readInterval=setInterval(function() {
     
      data = f.read(32);
      if (data) console.log(data);
      if (!data) {
        clearInterval(readInterval);
        f.close();
        console.log("\/\/\/ ::: READ ENDING ::: \\\\\\");
      }
    },100);
}
</pre>

So this was progress. And interestingly NOW when I ran the above code from Gordon using pipe, it worked nearly flawlessly (I say this because I had tried to write my own code using fs.read() and res.write to do my own "pipe" to the browser, and ran into all sorts of problems).

So pipe transfered 238K from the file system in 49 seconds. OK a bit slow, but perfectly usable. Memory dropped from 1529 to 1081 during the transfer. 

That said the pipe never finished from the browsers point of view and stayed in pending, even though the data showed up in the browser, and stopped at the end of the file. Also memory would stay at 1081 until you pressed on the "X" in Chrome at which point it would return to 1529 on the 8266. So I suspect a bad interaction between the FS and Pipe? 

So to summarize: 

- When I write files using writeFileSync() pipe does not pipe them. Though you can read them back normally, from the file system, so the file system seems ok there.
- When I write files "manually" using E.openFile() and then fs.write() etc, I can pipe it, and it even sends 200OK, which appears correctly in Chrome. 
- However, in either case the browser is left in a pending state that has to be manually canceled by the user.

wilberforce, 2017-04-10:
I modified you write test to be less extreme.....

function writeTest() {
  console.log(process.memory());
  for (i=0; i<100; i++) {
  var f = E.openFile("www/data.csv","a");
    var readings = [
      i,
      i,
      i,
      i
    ];
    f.write(readings.join(", ")+"\n");

    f.close();
  }
    console.log("done writing");
    console.log(process.memory())
}

and tested on the linux build and esp32 build with this code:

```
pipe=1;
function onPageRequest(req, res) {

	var a = url.parse(req.url, true);
	console.log(a);

    file=a.pathname;
	if (file == '/') {
		file='/index.htm';
	}
  
  

	var f = E.openFile('www' + file, "r");
    //var f = require("fs").readFileSync('www' + file);
	if (f !== undefined) {
      mime='text/html';
      if ( file.substr(-2)  == 'js' ) mime='application/javascript';
      if ( file.substr(-2)  == 'css' ) mime='text/css';
		res.writeHead(200, {
			'Content-Type': mime
		});
     if ( pipe === 0 ) {
      do{
        data = f.read(128);
        if (data) res.write(data);
      } while(data);
      res.end();
      f.close();
     }
    
      if ( pipe === 1 ) {
          f.pipe(res,{chunkSize:32, end:true, complete:function(){
            console.log("Complete");
            f.close();
            res.end();
            } 
          } );
      } 
      
      if ( pipe === 2 ) {
		f.pipe(res); // streams the file to the HTTP response
      }
	} else {
		res.writeHead(404, {
			'Content-Type': 'text/plain'
		});
		res.end("404: Page " + a.pathname + " not found");
	}
}

var ws_outer;
  
  function sendTime() {
    ws_outer.send(JSON.stringify({data:new Date().ms}));
  }
 var server = require('http').createServer(onPageRequest);
 server.listen(8080);
```

With `pipe=0` it does the read which works fine.

with `pipe=1` it tries to used the pipe complete function.

This works on the linux build.

on the esp32 it looks up after getting to here:

```
0, 0, 0, 0 1, 1, 1, 1 2, 2, 2, 2 3, 3, 3, 3 4, 4, 4, 4 5, 5, 5, 5 6, 6, 6, 6 7, 7, 7, 7 8, 8, 8, 8 9, 9, 9, 9 10, 10, 10, 10 ..........
93, 93 94, 94, 94, 94 95, 95, 95, 95 96, 96, 96, 96 97, 97, 97, 97 98, 98, 98, 98 99, 99, 99, 99
```

If you press the esc key in chrome, it then calls the complete function.

So it appears that the file close is not triggering the complete event.  It might be something to do the status of the number of bytes left to read or to do with the close.

It would be interesting to try this on an Espruino board with an SD card, unfortunately I don't have one to test with.

hungryforcodes, 2017-04-10:
So I liked the way you organized your code to test out the different pipe options. :)

So I would be extremely surprised if you could get pipe=0 to work with the 8266 on large files. Funny enough I started with the "stress test" example with just 100. But once you increase it to say 10000, and start outputting largish data we see which "pipes" work and which don't. The 8266 or Espruino/8266 seem to have buffer issues where once you start outputting something over say 100K it runs into memory problems and fails. Even more more annoying if you even out put just small files and hit reload a few times, it stops out putting data at all. So there is something fundamentally wrong with this pattern on Espruino/8266 based on fs.read() ->  res.write().

The good news is that I experienced the same behaviors as you on pipe=1, and pipe=2. Pipe=2 was obviously faster. So if we could just get it to properly END the pipe, then we'd be in business. 

Oh and now I can read and output all types of files, including the ones that I wrote with:

<pre>
require("fs").writeFileSync("hello.txt", "Hello World\nHello World\nNew line\n");
</pre>

I'll look into why this is the case. Sometimes you need to power the 8266 after a save to get things to work properly -- I suspect this is the case here. 

I have not tried WRITING from a pipe yet. @gfwilliams Do you have a nice compact example like the one you did above for testing out the pipe writing? That you know works? It would be much appreciated :)

wilberforce, 2017-04-11:
here is an example of piping out from an http upload into a file:
http://forum.espruino.com/comments/13546291/

hungryforcodes, 2017-04-11:
Yeah I've been working with that example for a few days now. I can't get it to work -- there might be a few problems, which I'm trying to work through, but I posted something in the Espruino forums about my basic problem. Once I get that clear that will be better for me (for example when I write a file and try and read it back I get connection resets etc). Fundamentally though the piping either only partially writes or just doesn't write at all. 

I am currently testing these snippits to see if I can get enough information to post something useful back: 
<pre>
function postUploadPipeEndFalse(req,res){
    console.log("postUploadPipe() --> req");
    console.log(req);
    var f = E.openFile("testUploadPipe.txt","w");
    req.pipe(f,{chunkSize:32, end:false,
      complete:function(){
          f.close();
          res.end();
          console.log("write Ended");
      }
    });
  }

function postUploadPipeEndTrue(req,res){
    console.log("postUploadPipe() --> req");
    console.log(req);
    var f = E.openFile("testUploadPipe.txt","w");
    req.pipe(f,{chunkSize:32, end:true,
      complete:function(){
          f.close();
          res.end();
          console.log("write Ended");
      }
    });
  }

function postUploadPipeNotFancy(req,res){
    console.log("postUploadPipe() --> req");
    console.log(req);
    var f = E.openFile("testUploadPipe.txt","w");
    req.pipe(f);
  }

function postUploadPipeNotFancyWithEnd(req,res){
    console.log("postUploadPipe() --> req");
    console.log(req);
    var f = E.openFile("testUploadPipe.txt","w");
    req.pipe(f);
    req.end();
  }

function postUploadFSWrite(req,res){
    console.log("postUploadFSWrite() --> req");
    console.log(req);
    var f = E.openFile("testUploadPipe.txt","w");
    req.on("data",function(data){
      f.write(data);
      console.log(data);});
    req.on("close",function(){
      f.close();
      res.end();
      console.log("write Ended");
    });
  }
</pre>

I'll update you when I have more info. For what its worth, once I get remove the connection resets from the equation, most of the problems seem to be that "completed" doesn't call and the page doesn't end and maybe also f.close() doesn't get called.

hungryforcodes, 2017-04-11:
Does it work for you on the ESP32, btw?

wilberforce, 2017-04-11:
Pipe does not work on esp32 however everything in the comment above:

> I modified you write test to be less extreme.....

Works.

With the esp8266 you are battling with such little memory.

The post above with pipe=0 and if you set the read buffer to 32, it should work for you.

hungryforcodes, 2017-04-11:
Right -- we both agree on that. And its great work!  :) Aside from the "closing" bug we also both agree on. It already makes things more usable even just for basic file I/O, and this is amazing, and honestly I'm very thankful for it. :)

However now I'm trying to write using pipe (or even fs.write()) from a browser. So I was curious if you had tried writing (say a large file from a browser) to the ESP32. Anyways its not so important I'll probably find a way around it sometime. I'll continue testing and see if I can find anything more concrete for you to work with.

wilberforce, 2017-04-11:
This should work for you on the ESP8266:

if ( pipe === 0 ) {
      do{
        data = f.read(32);
        if (data) res.write(data);
      } while(data);
      res.end();
      f.close();
     }

Please try that.

hungryforcodes, 2017-04-11:
Yes I'm sorry I wasn't clear -- I agree some variation of that will work on the ESP8266. I tried that extensively on the weekend --  the do / while construct will fail with out of memory on very large files like in my original example. Your modified example of mine is not large enough to trigger this. Its only a few K not a few 100K. However, this WILL work generally:

<PRE>

 readInterval=setInterval(function() {
      data = f.read(32);
      if (data) {res.write(data);}
      if (!data) {
        clearInterval(readInterval);
        f.close();
        res.end(page);
        console.log("\/\/\/ ::: READ ENDING ::: \\\\\\");
        return;
        }
  },100);

</PRE>

though occasionally will not detect the end of data and will start outputting again. Since Gordon's pipe() works more or less fine, I didn't pursue it too much further.  It would seem that buffers are not being cleared fast enough for the do / while construct. If I reduce the setTimeout below 100ms, or increase the f.read() to 64 or higher and keep the setTimeout at 100ms, it eventually does run out of memory about halfway through my example (the 10000 iteration one). 

But we agree that pipe works for sending data to the browser :)

What I am trying to do is WRITE to the file system, using pipe(). My understanding is that if I want to upload stuff to the ESP8266, this would be the best way if the file is large. 

Again I wasn't clear -- I believe we are confusing "res.write()" with a "write to the file system". Its the second I would like to do. So again my question would be have you tried writing to the file system on the ESP32 this way, using pipe()? If you can, then its clearly a problem on the ESP8266 side, is more or less what I'm trying to figure out. :)

Thanks!
hfc

wilberforce, 2017-04-11:
ok. I  Found the issue for reading a file wth File.pipe.  It is the read:

```
JsVar *jswrap_file_read(JsVar* parent, int length) {
  jsWarn( ">> jswrap_file_read len %d",length);
  if (length<0) length=0;
  JsVar *buffer = 0;
  JsvStringIterator it;
  FRESULT res = 0;
  size_t bytesRead = 0;
  if (jsfsInit()) {
    JsFile file;
    if (fileGetFromVar(&file, parent)) {
      if(file.data.mode == FM_READ || file.data.mode == FM_READ_WRITE) {
        size_t actual = 0;
#ifndef LINUX
        // if we're able to load this into a flat string, do it!
        size_t len = f_size(&file.data.handle)-f_tell(&file.data.handle);
        
        jsWarn( ">> jswrap_file_read  flat string %d",len);
        if (len>(size_t)length) len=(size_t)length;
        if ( len == 0 ) {
          jsError( ">> jswrap_file_read returning empty string");
          buffer = jsvNewFromEmptyString();
          return buffer;
        }
        buffer = jsvNewFlatStringOfLength(len);
        jsWarn( ">> jswrap_file_read  buffer is now  %d",buffer);
        if (buffer) {
          res = f_read(&file.data.handle, jsvGetFlatStringPointer(buffer), len, &actual);
          if (res) jsfsReportError("Unable to read file", res);
          fileSetVar(&file);
           jsWarn( ">> jswrap_file_read  returning flat string buffer %d",res );
          return buffer;
        }
        jsError( ">> jswrap_file_read  flat string fall through ... %d",len);
#endif
        char buf[32];

        while (bytesRead < (size_t)length) {
          size_t requested = (size_t)length - bytesRead;
          jsWarn( ">> jswrap_file_read requested %d",requested);
          if (requested > sizeof( buf ))
            requested = sizeof( buf );
          actual = 0;
          jsWarn( ">> jswrap_file_read requested now %d",requested);
    #ifndef LINUX
          res = f_read(&file.data.handle, buf, requested, &actual);
          if(res) break;
    #else
          actual = fread(buf, 1, requested, file.data.handle);
          jsWarn( ">> jswrap_file_read actual %d, got %s",requested, buf);
    #endif
          if (actual>0) {
            if (!buffer) {
              buffer = jsvNewFromEmptyString();
              if (!buffer) return 0; // out of memory
              jsvStringIteratorNew(&it, buffer, 0);
            }
            size_t i;
            for (i=0;i<actual;i++)
              jsvStringIteratorAppend(&it, buf[i]);
          }
          bytesRead += actual;
          if(actual != requested) break;
        }
        fileSetVar(&file);
      }
    }
  }
  if (res) jsfsReportError("Unable to read file", res);

  if (buffer)
    jsvStringIteratorFree(&it);

  return buffer;
}
```

This goes into a loop:

```
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
WARNING: >> jswrap_file_read  flat string 0
ERROR: >> jswrap_file_read returning empty string
WARNING: >> jswrap_file_read len 64
```

If this part is changed  `#ifndef LINUX` to `#ifdef XXXLINUX` to comment out the block, and the fall through code is called, it works...

So it seems the code that optimisation for a flat string returns something different so the receiving pipe does does recognise this as the end of file.

hungryforcodes, 2017-04-11:
OK great! I think there has been progress then on both sides. I don't know quite enough about the Espruino code to fully get whats going on there, but I get the gist of it. :) 

On my side I rethought about the problem and realized that the setTimeout was really awkward, and so after going through the docs thought to redo our snippet ("pipe==0"), with 

<PRE>
res.on("drain", function () {....});
</PRE>

which really seems like the *correct* way to do it. And sure enough it works, even detecting the end of data and does not leave the browser hanging. Here is the snippet I ended up with: 

<PRE>
f = E.openFile("data.csv","r");

    res.on("drain", function () {

      data = f.read(32);
          if (data) {res.write(data);}
          if (!data) {
              f.close();
              res.end(page);
              console.log("\/\/\/ ::: READ ENDING (DRAIN) ::: \\\\\\");
              return;
            }
        });
</PRE>

This of course is not the actual pipe() function, but is functionally identical (as far as I can tell), though a bit slower by maybe 20 seconds for my "large file" example. 

I would imagine this would easily work on the ESP32 as well. :)

Cheers,
hfc

hungryforcodes, 2017-04-11:
Yes so the above snippet seems to work after over 20 requests, without any problems, and drops the JsVars by only about only 90 vars while transferring, and then it returns to baseline. Which on the ESP8266 is pretty good! My results from using pipe() dropped available memory by almost 500. Not sure why pipe() would do that, but if you're working on an extremely limited board like mine, the code above seems more efficient.

hungryforcodes, 2017-04-11:
Oh and finally I'm happy to announce that the reset after save (see my original post), seems to have totally disappeared since I changed boards. So I'd say its not an issue at this point.

wilberforce, 2017-04-11:
@gfwilliams 
I found the cause issue of the pipe not closing after a File.read

https://github.com/espruino/Espruino/blob/master/src/jswrap_pipe.c#L17-L18

 * Piping works as follows:
 *   On idle, data is read from the source stream
 *    * If this returns undefined, the stream is considered finished and is closed
 *    * If this returns "" we just assume that it's waiting for more data

https://github.com/espruino/Espruino/blob/master/libs/filesystem/jswrap_file.c#L462-L472

```
 // if we're able to load this into a flat string, do it!
        size_t len = f_size(&file.data.handle)-f_tell(&file.data.handle);
        if (len>(size_t)length) len=(size_t)length;
        buffer = jsvNewFlatStringOfLength(len);
        if (buffer) {
          res = f_read(&file.data.handle, jsvGetFlatStringPointer(buffer), len, &actual);
          if (res) jsfsReportError("Unable to read file", res);
          fileSetVar(&file);
          return buffer;
        }
```
So when the len is 0, a buffer of "" is returned, so it waits for more data and then re-reads, resulting in a loop.

Here is the fix, to make it behave like the non flat string code:

```
        // if we're able to load this into a flat string, do it!
        size_t len = f_size(&file.data.handle)-f_tell(&file.data.handle);
        if ( len == 0 ) { // nothing left
          return 0; // if called from a pipe singal end callback
        }
```

gfwilliams, 2017-04-11:
Great, thanks for finding that! Looks like it might have been a regression due to my addition of flat strings?

Do you want to issue a PR for it? Sorry for the lack of replies but I'm supposed to be on holiday this week :)

wilberforce, 2017-04-11:
@gfwilliams 

I can add to the existing PR I have in the ESP32 branch - Are you happy for me to apply that?

gfwilliams, 2017-04-11:
Yes, that's fine. On the ESP32 PR the `romdata_jscode`stuff looks a bit dodgy though - it's still executing all the code before and is then just overwriting the result of it

wilberforce, 2017-04-11:
Ah - you mean here:
https://github.com/espruino/Espruino/blob/d87fe185b7e0c894beabfaa5d3deeabcbefc2bc2/src/jswrap_flash.c#L555-L569

It's executing this line which is not necessary:
  ```code = (char *)(FLASH_DATA_LOCATION);```

wilberforce, 2017-04-11:
Ok - tidied that up:
https://github.com/espruino/Espruino/pull/1129/commits/1c793f2bf5bd18c49133f41598a426881fbd25

But now it is showing as a conflict? I think you might have changed permissions?

I know see:
You're not authorized to merge this pull request.

MaBecker, 2017-04-15:
Started to work with this on the ESP8266_4MB board

What am I missing, after a reboot the file is gone?

````
// test fs and E.flashFatFS

fs=require("fs");
freeFlash=require("ESP8266").getFreeFlash();
index=0;
fname="Hello.txt";

content="Hello Espruino\n";
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("Formatting FS");
  E.flashFatFS({ addr:freeFlash[index].addr,
                 sectors:freeFlash[index].length/4096,
                 readonly:false,
                 format:true });
  require("fs").writeFileSync(fname,content );
}
else {
  require("fs").readFileSync(fname);
}
````

MaBecker, 2017-04-15:
Found the missing part, works as designed, great way to use 1MB.

````
var fs = require("fs");
var ESP8266 = require('ESP8266');
var freeFlash = ESP8266.getFreeFlash();
var index = 0;  // 1MB  flash area

var fname = "Hello.txt";
var content = "Hello Espruino";
var rc;

// attach flash as Fat FS
rc = E.flashFatFS({ addr:freeFlash[index].addr,
                    sectors:freeFlash[index].length/4096,
                    readonly:false,
                    format:false });

// format and write sample file 
if ( typeof(fs.readdirSync())==="undefined" ) {
  console.log("I: attach and format FS, rc="+
    E.flashFatFS({ addr:freeFlash[index].addr,
                   sectors:freeFlash[index].length/4096,
                   readonly:false,
                   format:true }));
  // write sample file
  fs.writeFileSync(fname,content);
}

// read sample file 
console.log("I: Content of File "+fname+":["+fs.readFileSync(fname)+"]");

````

MaBecker, 2017-04-22:
Adding 'FILESYSTEM' and 'FLASHFS' to ESP8266 is using to much heap space:

    'libraries' : ['NET','TELNET','GRAPHICS','CRYPTO','NEOPIXEL', 'FILESYSTEM','FLASHFS']
    
        "cpuFrequency": 160, "freeHeap": 5040, "maxCon": 10,

I think a combination of [Flash storage Module, any js type , web pages, css, js and images](http://forum.espruino.com/conversations/283045/) and module [FlashEEPROM](http://www.espruino.com/FlashEEPROM) would be nice to have for systems with poor resources like the ESP8266.

I am pretty sure this works nicely for ESP32.

wilberforce, 2017-04-22:
I had a looks this again recently and the module nesting a sub classing no longer works and I'm not sure why.

I fixed the compiler warnings in the fatfs code,  and made sure on the esp8266 that the 4 pages of flash are not used.

MaBecker, 2017-04-22:
I started to work on module  I name "simpleFlashStore", hope to share soon.

Concept: Use some pages as directory and the other to store data.

[8 more replies]

closed 2017-06-20 by wilberforce