espruino/Espruino #1117: pin handling in JSON.stringify returns empty object

jumjum123, 2017-03-16:
Using JSON stringify for pins creates a string, that cannot be parsed by JSON parse anymore.
After running attached lines, y is undefined. If we have more complex objects, its still undefined.
var x = {port:C5};
var j = JSON.stringify(x);
var y = JSON.parse(j);
If definition of objects like pins for JSON is not has no description in JSON definitions, we could 
- return an error message like 'port cannot be parsed'
- return an object like {port:undefined}
- extend parse to recognize pins
- some more ideas ?

gfwilliams, 2017-03-16:
As I'd said on that forum post (http://forum.espruino.com/conversations/301697/#comment13523682), if I make `parse` actually parse those pins then someone will complain - because it's used for parsing data from the net, it needs to be as compliant as possible.

It should really throw an exception.

If this is a problem, maybe JSON.stringify should also be made more compliant by refusing to output anything for the Pin's value, but then that could get annoying if you're trying to use stringify for debugging

gfwilliams, 2017-03-16:
Just done :) If you need to parse 'JSON' with pins, use `eval`

closed 2017-03-16 by gfwilliams

jumjum123, 2017-03-17:
Great, just checked changes and .... still learning ;-)

gfwilliams, 2017-03-17:
:) as it happened, JSON.parse *did* produce exceptions, but just not for that particular class of error

commit eb587571 (2017-03-16): JSON.parse now throws an exception if an incorrect value type is found (fix #1117)