espruino/Espruino #1130: RFE: Add setConnectionInterval (for BLE MIDI)

joebowbeer, 2017-04-03:
Please add `setConnectionInterval(min, max)` for BLE services such as MIDI. The BLE MIDI spec requires/suggests a connection interval between 7.25 and 15 ms.

Sample implementations:

https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/src/BLEPeripheral.h
https://github.com/01org/corelibs-arduino101/blob/master/libraries/CurieBLE/src/BLEPeripheral.h

I have a BLE MIDI controller up and running on Puck.js, without this enhancement, but I need this enhancement in order to achieve better performance.

As I understand the flow, the peripheral sends a`Connection Parameters Update Request` after the connection is established by the host.

FWIW, below are some references that I found useful.

https://devzone.nordicsemi.com/question/60/what-is-connection-parameters/
https://punchthrough.com/blog/posts/maximizing-ble-throughput-on-ios-and-android
http://community.silabs.com/t5/Bluetooth-Wi-Fi-Knowledge-Base/Midi-over-BLE/ta-p/170804
https://blog.felipetonello.com/2017/01/13/midi-over-bluetooth-low-energy-on-linux-finally-accepted/

gfwilliams, 2017-04-04:
At it happens there is already the ability to coursely control the connection interval: `NRF.setLowPowerConnection`.

Implementation here if you're interested: https://github.com/espruino/Espruino/blob/master/targets/nrf5x/bluetooth.c#L1098

When I did it I figured nobody would want to control the interval *that* exactly.

However, the default is already about as fast as it will go - 7.5 -> 20ms. You're not going to see any noticeable improvement by changing it to 7.5 -> 15ms I'm afraid - it's just the speed that BLE runs at.

joebowbeer, 2017-04-04:
Thanks for the explanation! I saw the setLowPowerConnection function but I didn't see that code that configures the connection interval range. I'll perform some more tests to determine if more functionality is needed.

gfwilliams, 2017-05-05:
any thoughts on this, or shall I just close it?

joebowbeer, 2017-05-05:
I haven't been able (yet) to demonstrate that this is needed, so I'll close this for now.

According to spec, the BLE MIDI peripheral is supposed to request an interval of 15ms or less, but the central device has the final say. In theory, the Espruino connection interval range of 7.5ms - 20ms (6, 16) should be acceptable.

On iOS, I've read that the host device will try to request an interval appropriate for the advertised services. HID and MIDI, for example, need shorter intervals than other services.

In practice, there have been reports that the advertised connection interval can make a significant difference:

https://groups.google.com/d/msg/android-midi/awtanDwxaCQ/7mkT2BojBgAJ

On Nexus 9, for example, advertising (6, 9) resulted in a shorter connection interval than advertising (6,16).

```
request 6,16 awarded 15
request 6,12 awarded 12
request 6,9 awarded 9
```

But the Android BLE MIDI implementation is still a work in progress, and this report is over a year old.

closed 2017-05-05 by joebowbeer

gfwilliams, 2023-05-03:
Just to add, this has existed for a whole now `NRF.setConnectionInterval(...)`