espruino/Espruino #1131: ESP8266 - Wifi.scan() causing "Out of Memory" error.

chrisgwgreen, 2017-04-07:
Hi all, 

I've been knocking together an IoT device using Espruino and came across an issue whilst testing in the centre of London. Where I am sat, wifi.scan() is returning around 30 access points which is eating up approximately 26% of the 1700 js variables I have available on the ESP8266.

Obviously the memory used will vary based on location and therefore is a pretty serious point of failure in my project. 

Can I suggest an additional feature to limit the number of returned access points (i.e. Wifi.scan(callback, limit)) such that the first 0-limit elements are added to RAM?

Thanks for your help in advance.

Chris

MaBecker, 2017-07-21:
Hi @chrisgwgreen,

there is no way to tell the api call to limit the result for a search.

Check this ESP8266 SDK API Guide abstract:
````
Function:  
    Scan all available APs
Note:  
    Do not call this API in user_init. This API need to be called after system initializes and the ESP8266 station mode is enabled.
Prototype: 
    bool wifi_station_scan (struct scan_config *config, scan_done_cb_t cb);
Structure:  
    struct scan_config { 
    uint8 *ssid;
    uint8 *bssid;
    uint8 channel;
    uint8 show_hidden; //scan APs of which ssid is hidden. 
};

````
So this can be closed - what do you think?

MaBecker, 2017-07-29:
.

closed 2017-07-29 by MaBecker