espruino/Espruino #1133: Segmentation Faults

rwhitworth, 2017-04-11:
Hello,
I was using American Fuzzy Lop (afl-fuzz) to fuzz input to espruino on Linux and found a few crashing inputs. Is fixing these crashes something you're interested in? The input files can be found here: https://github.com/rwhitworth/espruino-fuzz/

The files are espruino scripts that can be executed as `espruino FILENAME` on Linux.  I have not tried these scripts in an embedded environment.

Please let me know if I can provide any additional details to help narrow down the issue.

gfwilliams, 2017-04-12:
That's great - thanks! I'm on holiday at the moment but I'll look into this when I get back - I'm up for fixing pretty much anything that can make it segfault :)

gfwilliams, 2017-04-12:
Actually just took a quick look at this, and found the problem - it seems to fix all segfaults you found so far.

I'm leaving this open because there was an assertion that got hit as well that I'd like to sort out. (If you compile with `DEBUG=1` it'll leave all the asserts in.

rwhitworth, 2017-04-14:
Should I fuzz with the asserts left in?

gfwilliams, 2017-04-18:
It'd be an idea, yes. Hopefully the asserts should never be reachable with normal code.

commit a535e100 (2017-04-24): Fix assert detected by fuzzing - fix #1133
closed 2017-04-24 by gfwilliams