espruino/Espruino #1136: ARM Firmware doesn't compile when using the GCC compiler installed as part of "scripts/provision.sh"

benSTMax, 2017-04-13:
Hi,
I tried to compile the firmware for a few different STM32 boards and I ran into the same issue anytime and ARM target had to be compiled.
I setup a Vagrant machine, however I didn't installed the ARM compiler as provided by Debian Jessie 64 and I opted to have the "**scripts/provision.sh**" take care of it.
The compiler installed with no issues, however errors were thrown at the linking stage.

`/home/vagrant/Espruino/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: cannot find crti.o: No such file or directory
/home/vagrant/Espruino/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: cannot find crtbegin.o: No such file or directory
/home/vagrant/Espruino/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: cannot find crt0.o: No such file or directory
collect2: error: ld returned 1 exit status
make/targets/ARM.make:2: recipe for target 'espruino_1v91.748_espruino_1r3.elf' failed
make: *** [espruino_1v91.748_espruino_1r3.elf] Error 1
`

It seems the newer versions of the GCC doesn't provide anymore the **crt*.o** objects.
The GCC version installed by the "**scripts/provision.sh**" is:

`gcc version 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512] (GNU Tools for ARM Embedded Processors 6-2017-q1-update)`

gfwilliams, 2017-04-18:
Strange. I wonder whether it's a path issue for you? provision should be used by the travis builds, whichh obviously work fine

benSTMax, 2017-04-18:
Hi. I tried today with an Ubuntu-based Vagrant box and it works fine. The "crt*.o" files are provided and the compilation is fine.
I think this was an issue on my Debian Jessie Vagrant box

closed 2017-04-18 by benSTMax