espruino/Espruino #1137: compile warnings when adding  'FLASHFS'

MaBecker, 2017-04-15:
@wilberforce  is there a way to fix this?

```
libs/filesystem/fat_sd/flash_diskio.c: In function 'disk_write':
libs/filesystem/fat_sd/flash_diskio.c:109:3: warning: passing argument 1 of 'jshFlashWrite' discards 'const' qualifier from pointer target type [enabled by default]
   jshFlashWrite( buff, addr,size);  
   ^
In file included from /Volumes/ESPTools/Espruino/ESP8266/MaBecker/src/jsinteractive.h:18:0,
                 from libs/filesystem/fat_sd/flash_diskio.c:18:
/Volumes/ESPTools/Espruino/ESP8266/MaBecker/src/jshardware.h:344:6: note: expected 'void *' but argument is of type 'const BYTE *'
 void jshFlashWrite(void *buf, uint32_t addr, uint32_t len);
      ^
libs/filesystem/fat_sd/flash_diskio.c: At top level:
cc1: warning: unrecognized command line option "-Wno-float-conversion" [enabled by default]
cc1: warning: unrecognized command line option "-Wno-discarded-qualifiers" [enabled by default]

```

hungryforcodes, 2017-04-16:
I noticed alot of compiler warnings when building it as well, but it works still (as far as I can tell). However, I have to comment out GRAPHICS in the board (that's the right way to do it, no?). However, if I comment out CRYPTO and uncomment GRAPHICS, I get a 

<PRE>
** user1.bin uses  481380 bytes of 479232 available
espruino_esp8266_user1.bin too big!
</PRE>

The above is generated by the following config: 

<PRE>
 'libraries' : [
     'NET',
     'TELNET',
     'GRAPHICS',
     #'CRYPTO',
     'NEOPIXEL',
     'FILESYSTEM',
     'FLASHFS'
   ]
</PRE>

Do you get the same thing? That's unfortunate as I had wanted to use the graphics and the neopixel at the same time, along with the new FS module. So lets say I comment also out the NEOPIXEL to get a smaller firmware size, like this: 

<PRE> 
'build' : {
   'libraries' : [
     'NET',
     'TELNET',
     'GRAPHICS',
     #'CRYPTO',
     #'NEOPIXEL',
     'FILESYSTEM',
     'FLASHFS'
   ]
</PRE>

I get a compile error: 

<PRE> 
CC targets/esp8266/jswrap_esp8266.o
targets/esp8266/jswrap_esp8266.c:35:29: fatal error: jswrap_neopixel.h: No such file or directory
 #include <jswrap_neopixel.h>
                             ^
compilation terminated.
make: *** [targets/esp8266/jswrap_esp8266.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** wait: No child processes.  Stop.
</PRE>

If I comment out the NEOPIXEL, it should not be looking for  jswrap_neopixel.h -- or am I missing something maybe? 

-hfc

MaBecker, 2017-04-16:
No, because I am working on #1110

add 'NEOPIXEL', it is very small

when you change lib setting run `make clean` before `make`

hungryforcodes, 2017-04-16:
OK thank you. :)

wilberforce, 2017-04-17:
@MaBecker these kind of warnings can be fixed by casting. Is it really worth worrying about?

closed 2017-04-17 by MaBecker