espruino/Espruino #1138: ESP32 - preparing for 1v92 release

wilberforce, 2017-04-15:
Is there anything that needs to be done to get the esp32 firmware files so they will appear here (when 1v92 is ready)?
http://www.espruino.com/files/espruino_1v91.zip

How do I make this file? It does not seem to be built by EspruinoDocs?

http://www.espruino.com/json/ESP32.json

The ESP32.md files are getting prepared now.

MaBecker, 2017-04-15:
> http://www.espruino.com/json/ESP32.json

    BOARD=ESP32  make boardjson

creates boards/ESP32.json

and then Gordon has to copy it to http://www.espruino.com/json/ESP32.json

wilberforce, 2017-04-15:
@gfwilliams 

https://github.com/wilberforce/Espruino/blob/master/boards/ESP32.json

Please can this be added to the main site:

http://www.espruino.com/json/ESP32.json

gfwilliams, 2017-04-19:
Done

closed 2017-04-19 by gfwilliams

wilberforce, 2017-05-29:
@MaBecker 

This needs to be done for:
http://www.espruino.com/json/ESP8266_4MB.json