espruino/Espruino #1140: Can't flash ESP8266 with espruino_1v91

nick-beznos, 2017-04-16:
README says: 
*** To flash a 4MB esp8266 (e.g. esp-12) using the serial port use:
esptool.py --port /dev/cu.wchusbserial1420 --baud 115200 write_flash \
  --flash_freq 80m --flash_mode qio --flash_size 32m \
  0x0000 "boot_v1.5.bin" 0x1000 espruino_esp8266_user1.bin \
  0x3FC000 esp_init_data_default.bin 0x37E000 blank.bin

I'm working on mac. I've done everything correctly even changed "32m" on "4MB", as terminal said, and "boot_v1.5.bin" on "boot_v1.6.bin". But its still not working. 

My Terminal says that there's an error:
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'boot_v1.6.bin'
 
Maybe previous version will work, but I hadn't find one.

I've searched in internet for a solution, but the fact that you're reading this text says that my search was unsuccessful. 
Help me please)

MaBecker, 2017-04-16:
Hi @nick-beznos,

did you download or build or own version?

nick-beznos, 2017-04-17:
I did download one from here: http://www.espruino.com/Download a few times and even tried to flash my Lolin NodeMcu v3 on windows. Same result.

wilberforce, 2017-04-17:
In zip file you should see a folder:
/espruino_1v91.zip/espruino_1v91_esp8266

and that has these files, including the boot.bin:
```
12/01/2017  10:58 PM             4,096 blank.bin
12/01/2017  10:58 PM             3,856 boot_v1.6.bin
12/01/2017  10:58 PM           460,068 espruino_esp8266_user1.bin
12/01/2017  10:58 PM           460,068 espruino_esp8266_user2.bin
12/01/2017  10:58 PM               128 esp_init_data_default.bin
12/01/2017  10:58 PM             1,007 README_flash.txt
12/01/2017  10:58 PM             2,961 wiflash.sh
```

nick-beznos, 2017-04-18:
At last I've done it!) I used an older version of esptool)

closed 2017-04-19 by gfwilliams

daveamit, 2017-06-09:
@nick-beznos I am having same issue, can you please list the version you used? for esptool and what version espruino did you flashed.

Thanks in advance.