espruino/Espruino #1141: analogRead not usable on ESP-WROOM-32

mrbar42, 2017-04-18:
ESP-WROOM-32 (no dev board) with espruino `1v91.529` rev `c59494`

regardless of the provided pin number, analogRead returns the same exact value `0.06298828125`.

I basically tried all pins that are ADC supported according to the [data sheet](https://www.google.co.il/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwiag-aLxK7TAhUU1mMKHYImBw0QFgghMAA&url=https%3A%2F%2Fwww.espressif.com%2Fsites%2Fdefault%2Ffiles%2Fdocumentation%2Fesp_wroom_32_datasheet_en.pdf&usg=AFQjCNEzYZYAle8cp3A93fQaLdEdzgyTNw&sig2=OfGgzAQDpbaHhBY8NovkPw)
and for each pin passed every possible value (0-32).
tried to set the pin mode manually to `input` without luck as well.

is it implemented? should it work?
according to [the docs](https://github.com/espruino/EspruinoDocs/blob/master/boards/ESP32.md#gpio-pins) it should, but they are somewhat ambiguous.

wilberforce, 2017-04-19:
I think true adc in on gpio35 and gpio36. @jumjum can cornfirm

wilberforce, 2017-04-20:
It seems that adc2 can't be used if you want I use wifi at the same time.

https://github.com/espressif/esp-idf/issues/461

And there is no ESP-idf driver at this stage..

jumjum123, 2017-04-20:
> espressif/esp-idf#461 <https://github.com/espressif/esp-idf/issues/461>

ADC2 is not supported yet.
To be honest, This status is some weeks old, but there haven't been any
commits around this since then.

mrbar42, 2017-04-20:
Let me verify that your description is indeed the case and i'll update the docs accordingly and close this issue

wilberforce, 2017-04-20:
I like your approach!

mrbar42, 2017-04-21:
I just checked, and on my board (again, plain - no dev board) i am able to read analog data from both pins IO35 AND IO36(aka SENSOR_VP) while Wifi is on and connected

Any remarks?

wilberforce, 2017-04-21:
The chip has two adc channels adc1 and adc2. Adc1 has been implemented, adc2 has not.
Adc1 is used for 35 and 36. 
And for the documention it would be good to mention the max input voltage for the pins.

closed 2017-04-27 by gfwilliams