espruino/Espruino #1149: Segmentation faults

rwhitworth, 2017-04-24:
Compiled commit a535e100 with DEBUG=1 and found additional inputs that cause segmentation faults.  These test cases look more like JavaScript than the previous ones so they should be easier to debug and test.  Many do contain non-ascii characters, fwiw.

Files available from https://github.com/rwhitworth/espruino-fuzz/tree/master/2017-04-24

gfwilliams, 2017-04-24:
Thanks - just fixed one or two, but it looks like there could be a mix of things here - looks like it's hitting a few of the built in functions here, which is great news.

There are a few `[[[[[[[[[[[[[[[[[[[[[[[[[[[[[[[[` type files which seem to just be filling the stack up. Good thing to know and it looks like I'll need to add some sanity checking - I believe there is stuff on ARM embedded targets to handle this, but on PC I tend to treat the stack as infinite :)

Out of interest, how long did it take to find those issues?

rwhitworth, 2017-04-24:
These were found on a previous version and took around a week on 4 CPU cores.  Coverage is estimated to be 24% so there is still quite a bit the fuzzer hasn't been able to trigger.  I need to find/create better starting input files to reach more functions efficiently.

gfwilliams, 2017-04-24:
Interesting - I imagine most of the uncovered stuff is for the built-in libraries - and stuff like HTTP would be particularly hard to fuzz. Still, just getting the base interpreter solid is an amazingly good start.

It doesn't make me feel too bad about those bugs if they took a week of solid hammering to find :)

I guess just triggering the functions from http://www.espruino.com/json/espruino.json with random input could give you an interesting start, and would increase your coverage a huge amount.

rwhitworth, 2017-04-25:
HTTP client code could be fuzzed, albeit slowly.  I think I'd have to deep dive into the code to fuzz the HTTP server or telnet server code.  I'll take a look at those.  In the meantime I'm doing some investigation into Date.parse, JSON.parse, new String, parseFloat, and url.parse.

Thanks for the JSON formatted file of functions.  I was unaware that existed and will try to parse with jq to turn into examples that can be used as input.

gfwilliams, 2017-05-05:
All segfaults now fixed, but there are one or two assertions I'd like to track down

gfwilliams, 2017-05-05:
Ok, and fixed the assert - that was a really good find - thanks! Turns out you could do `var a.b.c` for some weird reason - I think the code just got copy/pasted over by accident and never noticed.

When dumping it can still assert with `locks>15`, but that's not such a big deal - otherwise I think all the stuff you found is fixed now - so closing this bug to get some kind of closure. Feel free to submit more if you find them though :)

closed 2017-05-05 by gfwilliams