espruino/Espruino #1150: NRF.setServices advertise option and 128-bit UUIDs

joebowbeer, 2017-04-24:
The advertise option added in v1.92 is not working as advertised on Puck.js.

The following should work according to documentation:

```
NRF.setServices({
  "03B80E5A-EDE8-4B33-A751-6CE34EC4C700" : { // MIDI
    "7772E5DB-3868-4112-A1A9-F2669D106BF3": {
      readable: true,
      writable: true,
      notify: true,
      value : [0x80, 0x80, 0x00, 0x00, 0x00]
    }}
  }, {advertise: ["03B80E5A-EDE8-4B33-A751-6CE34EC4C700"]}
);
```

But actually this has no impact on the advertised data:

```
// Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
//    0x02,0x01,0x05,
// Complete Local Name: Puck.js 0000
//    0x0D,0x09,0x50,0x75,0x63,0x6B,0x2E,0x6A,0x73,
//    0x20,0x40,0x40,0x40,0x40,
// Nordic UART
//    0x11,0x07,0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,
//    0xE0,0x93,0xF3,0xA3,0xB5,0x01,0x00,0x40,0x6E
```

NUS is still advertised as the one and only service. Notice the *complete* list of services (`0x07`).

In v1.91, I was able to advertise the BLE MIDI service by calling `setAdvertising`:

```
NRF.setAdvertising([
    0x11,0x06,0x00,0xC7,0xC4,0x4E,0xE3,0x6C,0x51,
    0xA7,0x33,0x4B,0xE8,0xED,0x5A,0x0E,0xB8,0x03 // MIDI
  ]);
```

The resulting advertisement data (below) is incomplete and not well formed, but it works well enough for iOS to identify the Puck as a BLE MIDI peripheral:

```
// MIDI
//    0x11,0x06,0x00,0xC7,0xC4,0x4E,0xE3,0x6C,0x51,
//    0xA7,0x33,0x4B,0xE8,0xED,0x5A,0x0E,0xB8,0x03,
// Nordic UART
//    0x11,0x07,0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,
//    0xE0,0x93,0xF3,0xA3,0xB5,0x01,0x00,0x40,0x6E
```

Notice that the advertisement above is missing Flags and the Complete Local Name, and NUS is still advertised as the *complete* list of services, while MIDI is advertised as an incomplete listing.

This is the only advertisement. There is no rotating list of service advertisements.

The above hack still works in v1.92, but only if `setServices` is called without the `advertise` option.

Expected: Using the `advertise` option, what should be advertised is a rotating list of incomplete services:

```
// Advertisement #1:
// Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
//    0x02,0x01,0x05,
// Complete Local Name: Puck.js 0000
//    0x0D,0x09,0x50,0x75,0x63,0x6B,0x2E,0x6A,0x73,
//    0x20,0x40,0x40,0x40,0x40,
// Nordic UART (incomplete)
//    0x11,0x06,0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,
//    0xE0,0x93,0xF3,0xA3,0xB5,0x01,0x00,0x40,0x6E
//
// Advertisement #2:
// Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
//    0x02,0x01,0x05,
// Complete Local Name: Puck.js 0000
//    0x0D,0x09,0x50,0x75,0x63,0x6B,0x2E,0x6A,0x73,
//    0x20,0x40,0x40,0x40,0x40,
// MIDI (incomplete)
//    0x11,0x06,0x00,0xC7,0xC4,0x4E,0xE3,0x6C,0x51,
//    0xA7,0x33,0x4B,0xE8,0xED,0x5A,0x0E,0xB8,0x03,
```

See sample project at www.hackster.io/joebowbeer/ble-midi-button-puck-js-c74f0e

gfwilliams, 2017-04-25:
It could be possible that there are just too many services to advertise in the fixed advertising packet size?

What if you add `uart : false` in the second argument to setServices? Obviously you'll then not be able to use the BLE UART service until you reboot, but it might help to narrow down the problem?

You could also try shortening the device's name to just one character?

joebowbeer, 2017-04-25:
Yes, there are too many 128-bit services to advertise in a single packet, esp. if the flags and name is included. This is the situation that the incomplete list of services (0x06) was designed for, and why the advertisement in this case should rotate through the list of services (as is done on other devices, such as LightBlue Bean).

Even if they both fit in one packet, the Nordic UART service (NUS) itself shouldn't be encoded as type 0x07 (complete list).

gfwilliams, 2017-04-25:
If I automatically switched to multiple advertising packets then I'm pretty sure I'd just get another bug report from someone else about it not working as they expect.

Detecting it and fixing is also non-trivial as I just get a DATA_SIZE error returned from the softdevice stack. In the case where you specify 2x 128 bit UUID and 4x 16 bit UUIDs, how should they be split into different advertising packets? it turns into a minefield.

The best bet would just be to allow the advertising packets to be specified manually, and you can specify multiple advertising packets to be sent in rotation right now. It seems that a real blocking issue here is the list of UUIDs is always sent when perhaps there should be an option to disable it?

I've just fixed the original issue - when you called setAdvertising, after it completes and you disconnect the device got an internal error and rebooted, wiping out your changes (all the LEDs should have blinked?). It will now report an Exception so at least you know when something is wrong (just not immediately if you're connected by BLE, since advertising only gets updated when you disconnect).

joebowbeer, 2017-04-25:
In the implementation I'm most familiar with (Punch Through), everything (flags, name) is sent in each advertisement *except* the incomplete lists, which are divvied up among the rotating advertisements. That way there is a constant prefix size, and then the 16-bit and 128-bit UUID lists are handled in turn. Three rotating advertisements should be sufficient for the case you describe. The incomplete list types are designed for this, and service UUIDs are the only type for which this is supported:

https://www.bluetooth.com/specifications/assigned-numbers/generic-access-profile

If the 4x 16-bit UUIDs can be sent in one advertisement, that list would be encoded as type 0x03 (Complete List of 16-bit UUIDs).

If I could completely specify the rotating list of advertisements, and not have to disable UART, then I'd have what I need. But what I've found is that the NUS UUID is appended onto everything. And it is encoded as a Complete List, which is wrong in my case.

gfwilliams, 2017-04-25:
Ok, having looked into this a little more, what you've suggested isn't possible on any BLE device as far as I can see. The max advertising data size is 31 bytes, so the device's name would have to be smaller than you're showing to fit it all in to one advertising packet.

Currently the services are advertised as scan response data, but I'm not convinced that you can actually 'rotate' the scan response reliably since I'm not sure it's checked every time by the receiving device.

What you *can* do is re-order it. So the scan response contains the name:

```
NRF.setScanResponse([
   // Complete Local Name: Puck.js 0000
    0x0D,0x09,0x50,0x75,0x63,0x6B,0x2E,0x6A,0x73,
    0x20,0x40,0x40,0x40,0x40
]);
```

And the advertising alternates the advertised services:

```
NRF.setAdvertising([
// Advertisement #1:
// Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
[   0x02,0x01,0x05,
// Nordic UART (incomplete)
    0x11,0x06,0x9E,0xCA,0xDC,0x24,0x0E,0xE5,0xA9,
    0xE0,0x93,0xF3,0xA3,0xB5,0x01,0x00,0x40,0x6E],
//
// Advertisement #2:
// Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
[    0x02,0x01,0x05,
// MIDI (incomplete)
    0x11,0x06,0x00,0xC7,0xC4,0x4E,0xE3,0x6C,0x51,
    0xA7,0x33,0x4B,0xE8,0xED,0x5A,0x0E,0xB8,0x03 ]
]);
```

And that seems to work fine, although nRF connect doesn't seem to realise there are two services - it just alternates between showing one and the other.

Basically if you don't want Puck.js sending its own services, just do `NRF.setScanResponse([])`. I don't think  I can realistically start automatically alternating advertising packets though, especially since the default length of device name means it's not even as simple as alternating advertising packets

Closing as there seems to be a workaround, and I've also fixed the original issue with was the reboot caused by setServices with too big a UUID

closed 2017-04-25 by gfwilliams

joebowbeer, 2017-04-25:
Thanks for suggesting a workaround.

I notice that nRF Connect (Android) reports the default, raw advertisement data for my Puck.js as: 

```
0x0201050D095075636B2E6A73203666396111079ECADC240EE5A9E093F3A3B50100406E
```

That is 35 bytes.

So the scan response is already being employed in order to deliver the default advertisement?

gfwilliams, 2017-04-25:
Yes, absolutely - I didn't realise that until I looked into this bug, but the Services go into the scan response by default. With a small name you could fit one service in the main advertising body and one in the scan response if needed.

I was going to add some ability to disable the services but there's no need since `NRF.setScanResponse` just overwrites the scan response and does it for you.

Worth noting that I discovered a bug in 1v92 when dealing with arrays in setAdvertising (a double-free). I think it's ok (it triggered an assert in non-release builds but should have been fine otherwise), but if you get problems you can use a cutting-edge firmware on it.

joebowbeer, 2017-04-26:
I tried adding Nordic UART to advertising and removing it from the scan response, using the code above, but then Espruino IDE fails to find the Puck.js even though Nordic UART shows up in nRF Connect.

gfwilliams, 2017-04-26:
That's odd - is it possible that the lightblue bean does something different? Maybe you could try doing something there and seeing if it appears in the IDE?

joebowbeer, 2017-04-27:
The following works in v1.91 and v1.92:

```
NRF.setServices({
    "03B80E5A-EDE8-4B33-A751-6CE34EC4C700": { // MIDI
      "7772E5DB-3868-4112-A1A9-F2669D106BF3": {
        readable: true,
        writable: true,
        notify: true,
        value: [0x80, 0x80, 0x00, 0x00, 0x00]
      }}
});

NRF.setAdvertising([
    // Flags: LE Limited Discoverable Mode, BR/EDR Not Supported
    0x02,0x01,0x05,
    // Complete Local Name: PuckCC
    0x07,0x09,0x50,0x75,0x63,0x6B,0x43,0x43,
    // MIDI
    0x11,0x06,0x00,0xC7,0xC4,0x4E,0xE3,0x6C,0x51,
    0xA7,0x33,0x4B,0xE8,0xED,0x5A,0x0E,0xB8,0x03
  ]);
```

This shortens the local name to make room for the MIDI UUID in the advertisement, and leaves the scan response unchanged. Apparently the Espruino Web IDE is looking for the Nordic UART in the scan response.

gfwilliams, 2017-04-27:
Great! Thanks for the update. I think it might be a Web Bluetooth thing - I'll have a check with the Chrome team...