espruino/Espruino #1151: rendering i and j with setFontVector < 17 makes characters unreadable

jose1711, 2017-04-24:
i understand that with decreasing fontsize also the rendering quality decreases but i believe something still can be done to the characters containing a dot. here's a screenshot of 
```
g.setFontVector(16);
g.drawString("abcdefghijklm.:", 0, 5);
```
[image]
please note how "ij" part renders.

gfwilliams, 2017-04-25:
Thanks. However I would say 'unreadable' is a strong word for what's shown in that picture :)

Do you have any particular thoughts on what could be done? The vector font is automatically created from a source font, so I guess there are two options:

* Change the rendering somehow - but I'm not sure how
* Actually create a new font from scratch that will scale better (perhaps moving the dot further from the rest of the letter)

I'm not sure if you're aware, but for smaller fonts there are several different bitmap fonts you can use that will render much better: https://www.espruino.com/Fonts

jose1711, 2017-04-25:
Hello there,

the description may have sounded a bit harsh but OTOH if bitmap font does not suit your needs and you really want to be able to distinguish between i, I, | and !, well.. :-)

For the sake of completeness I rendered the example for font sizes going from 7 (which does not really make sense at all) all the way up to 19.

07
[image]
08
[image]
09
[image]
10
[image]
11
[image]
12
[image]
13
[image]
14
[image]
15
[image]
16
[image]
17
[image]
18
[image]
19
[image]

It is interesting to see how 15 looks acceptable (although it does fail when rendering an exclamation mark in another test), but 16 does not.

To be honest, I do not know what the best solution would be. I wonder though how the font is rendered - I could see the polygons defined for each letter but could not find the process which translates them to bitmaps.

opichals, 2017-04-25:
@gfwilliams Aside from creating a bitmap fonts of specific size I can only think of compiling FreeType library in which could be fine for board with plenty of RAM.

gfwilliams, 2017-04-25:
Since you can mix and match fonts, I'm still really struggling to see in what circumstances you absolutely need to use a small vector font but can't just use a bitmap, with a vector font for the larger characters? What's your use case?

I guess freetype would be better, but I don't think it's realistic? I just checked the size of the 'Free Sans' TTF font on Linux. It's 1.5MB! By comparison Espruino's vector font is 5kB.

Obviously the ideal would be to use grayscale and antialiasing, but the majority of devices will have a pure black and white display (and you can get grayscale by rendering at 2x size and sampling down if you need to). I just tried with Gimp and plenty of vector fonts on the PC render really badly in 1 bit at 8px high.

The character rendering code is here: https://github.com/espruino/Espruino/blob/master/libs/graphics/graphics.c#L373

It's pretty basic, but there's a huge pressure on keeping the flash and RAM needed to a minimum so quite a lot of work has gone in to both that and the font to make that happen. If you find a better way of rendering without massively increasing the font size then I'd be really interested though.

Perhaps one option would be to just render twice as large and subsample. The following code does do a better job (for a Pico with the Nokia LCD display on it):

```
A5.write(0); // GND
A7.write(1); // VCC

// Setup SPI
var spi = new SPI();
spi.setup({ sck:B1, mosi:B10 });
// Initialise the LCD
var g = require("PCD8544").connect(spi,B13,B14,B15, function() {
  // When it's initialised, clear it and write some text
  g.clear();
  g.setFontVector(10);
  g.drawString2("abcdefghijklmnopq",0,0);
  g.flip();
});

g.fontSize = 6;
g._setFontVector = g.setFontVector;
g.setFontVector = function(s) {
  this.fontSize = s;
  this._setFontVector(s);
};
g.drawString2 = function(txt,x,y) {
  var w = this.stringWidth(txt);
  var h = this.fontSize;
  var b = Graphics.createArrayBuffer(w*2,h*2,1);
  b.setFontVector(this.fontSize*2);
  b.drawString(txt);
  var x,y;
  for (y=0;y<h;y++)
    for (x=0;x<w;x++)
      if ((b.getPixel(x*2,y*2)+
           b.getPixel(x*2+1,y*2)+
           b.getPixel(x*2,y*2+1)+
           b.getPixel(x*2+1,y*2+1))>2)
        this.setPixel(x,y);
};
```

It's still far from ideal though - it's a hard problem to solve.

gfwilliams, 2017-12-15:
Closing this as a wontfix - I don't see any simple way around it, and playing around on desktop Linux and Gimp they appear to swap to a different line-based font under 17px high.

Realistically the best solution is just to use bitmap fonts, which are already supported and easy to use.

closed 2017-12-15 by gfwilliams

gfwilliams, 2018-06-01:
Minor update: I [fixed some rounding errors in fillPoly](ed6fad079cbcba40bb3a733f9250c7369b53e594) that improve the font rendering. However these still don't fix vector fonts <17 pixels.