espruino/Espruino #1152: Date.parse() segmentation faults

rwhitworth, 2017-04-25:
Similar in nature to the previously reported segmentation faults, but these input files must be fed into Date.parse().  An example script is included but the filename inside the script will need to be changed.  Input files available from https://github.com/rwhitworth/espruino-fuzz/tree/master/2017-04-25-Date-parse

Can an Espruino script access argv[1]?  I have not been able to find an easy way yet.

As always, let me know if I can provide help reproducing the issues.  Thanks.

gfwilliams, 2017-04-26:
Thanks! argv isn't exposed, but there is a flag that lets you execute an expression on the command-line, which could be an option?

gfwilliams, 2017-05-05:
These are all fixed with the latest commit 5240e7a8f6384cac3fa0c2e546b339ce196e95e1 - it seems they were all variants of the stack overflow caused by repeated recursion (so not specifically a bug, just not checking when the C stack was exhausted). Not even any asserts triggered.

closed 2017-05-05 by gfwilliams