espruino/Espruino #1153: Problem to save wifi settings

gomes-fdr, 2017-04-27:
Hello folks, I have two models of ESP8266, a model 01 and a node MCU.
When I do a config steps described in tutorials in wifi section, everything works well in node MCU. 

But unfortunately when I do the same steps with ESP8266-01, the settings doesnt keep in it, when I switch off I lost everything.

This is what I do:

var wifi = require("Wifi");
wifi.connect("xxx", {password:"xxx"}, function(ap){ console.log("connected"); });
wifi.save()

gomes-fdr, 2017-04-27:
The firmware version is 1.92

gomes-fdr, 2017-04-27:
And I flash the rigth version to both models

wilberforce, 2017-04-27:
You are doing the save before the connection has completed.

Change to:

```
var wifi = require("Wifi");
wifi.connect("xxx", {password:"xxx"}, function(ap){ 
   console.log("connected"); 
  wifi.save()
});
```

Please can you raise these kind of issues in the forum, as github for for development.
http://forum.espruino.com/microcosms/925/

closed 2017-04-27 by wilberforce

gomes-fdr, 2017-04-27:
No, I am waiting display the message "connected" show up, then I run the save command.
Even so I will test it.
Sorry about my mistake, next time I will ask in the right place.