espruino/Espruino #1155: ESP32 port has duplicated all Wifi functions in reference

gfwilliams, 2017-04-29:
As seen here:

http://www.espruino.com/Reference#Wifi

Ideally rather than just duplicating everything we'd move all Wifi JSON comments into one file `jswrap_wifi` and then keep the implementations separate for ESP32 and ESP8266 as they are currently.

We'd need to check that none of the documentation is separate for ESP8266/ESP32.

gfwilliams, 2017-04-29:
Fixed this, I hope. If someone could check (@wilberforce?) that there isn't something in the ESP32 port's documentation that got removed (eg, not added to `jswrap_wifi.c`) it'd be really handy. Both ESP32 and ESP8266 seem to compile fine though

closed 2017-04-29 by gfwilliams

wilberforce, 2017-04-30:
pulling:

```
>process.env
={
  "VERSION": "1v92.37",
```

It looks like all the functions are there.... hitting tab after `wifi.` yields:

```
>var wifi=require("Wifi");
=function () { [native code] }
>
=undefined
>
=undefined
>wifi.
connect             disconnect
getAPDetails        getAPIP
getDHCPHostname     getDetails
getHostByName       getHostname
getIP               getStatus
restore             save
scan                setConfig
setDHCPHostname     setHostname
startAP             stopAP
apply               bind
call                replaceWith
clone               emit
hasOwnProperty      length
on                  removeAllListeners
removeListener      toString
valueOf
```

gfwilliams, 2017-05-02:
Great - thanks for the test!

commit c09739f6 (2017-04-29): Remove duplicate JSON code and WiFi functions in reference, fix #1155