espruino/Espruino #1157: Bad Response on Serial in

denzelwamburu, 2017-05-02:
Am using [Sparkfun thing](https://www.sparkfun.com/products/13231), and i successfully flashed the firmware (v1.92), also i've followed all steps successfully. Though when i try to serial in, the response am getting is not even human readable. Changing the baud rate really doesn't help.
What could be the issue?
[image]

gfwilliams, 2017-05-02:
Please can you ask on the Espruino forums for ESP8266? http://forum.espruino.com/microcosms/925/

This kind of problem seems super-common.

closed 2017-05-02 by gfwilliams

denzelwamburu, 2017-05-02:
Okay. Thanks