espruino/Espruino #1158: WARNING: Module "fs" not found using the basic example on original espruino board

asez73, 2017-05-04:
Reviving my old original espruino board, I tried the example from the webide, code send from right pane to left pane:

`
var files = require("fs").readdirSync();
for (var i in files)
  console.log("Found file "+files[i]);
`

and get the error below

`
>
 _____                 _
|   __|___ ___ ___ _ _|_|___ ___
|   __|_ -| . |  _| | | |   | . |
|_____|___|  _|_| |___|_|_|_|___|
          |_| http://espruino.com
 1v92 Copyright 2016 G.Williams
>WARNING: Module "fs" not found
Uncaught Error: Field or method "readdirSync" does not already exist, and can't create it on undefined
 at line 1 col 26
var files = require("fs").readdirSync();
                         ^
=undefined
> 
`

Board information is 

> VERSION	1v92
BUILD_DATE	May  3 2017
BUILD_TIME	13:48:53
GIT_COMMIT	343d9cb226ce91a8a678ee43fa8a98366e1af92f
BOARD	ESPRUINOBOARD
CHIP	STM32F103RCT6
CHIP_FAMILY	STM32F1
FLASH	262144
RAM	49152
SERIAL	34ffdb05-42573838-24631743
CONSOLE	USB
EXPORTS	{"jsvLock":167949,"jsvLockAgainSafe":167933,"jsvUnLock":142425,"jsvSkipName": ...

And the console log says:

> SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":true,"MINIFICATION_Unreachable":true,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"SIMPLE_OPTIMIZATIONS","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":true,"SAVE_ON_SEND":false,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":true,"SHOW_CODE_LINK_ICON":true,"SHOW_RELOAD_ICON":true,"SHOW_WEBCAM_ICON":true,"STORE_LINE_NUMBERS":false,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"compact","projectEntry":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.STORE_LINE_NUMBERS => false
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":false,"MINIFICATION_Unreachable":true,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"SIMPLE_OPTIMIZATIONS","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":true,"SAVE_ON_SEND":false,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":true,"SHOW_CODE_LINK_ICON":true,"SHOW_RELOAD_ICON":true,"SHOW_WEBCAM_ICON":true,"STORE_LINE_NUMBERS":false,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"compact","projectEntry":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.MINIFICATION_Mangle => false
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":false,"MINIFICATION_Unreachable":false,"MINIFICATION_Unused":true,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"SIMPLE_OPTIMIZATIONS","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":true,"SAVE_ON_SEND":false,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":true,"SHOW_CODE_LINK_ICON":true,"SHOW_RELOAD_ICON":true,"SHOW_WEBCAM_ICON":true,"STORE_LINE_NUMBERS":false,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"compact","projectEntry":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.MINIFICATION_Unreachable => false
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":false,"MINIFICATION_Unreachable":false,"MINIFICATION_Unused":false,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"SIMPLE_OPTIMIZATIONS","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":true,"SAVE_ON_SEND":false,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":true,"SHOW_CODE_LINK_ICON":true,"SHOW_RELOAD_ICON":true,"SHOW_WEBCAM_ICON":true,"STORE_LINE_NUMBERS":false,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"compact","projectEntry":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.MINIFICATION_Unused => false
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":false,"MINIFICATION_Unreachable":false,"MINIFICATION_Unused":false,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"","MODULE_URL":"http://www.espruino.com/modules","NPM_MODULES":false,"RESET_BEFORE_SEND":true,"SAVE_ON_SEND":false,"SERIAL_AUDIO":"0","SERIAL_TCPIP":"","SERIAL_THROTTLE_SEND":true,"SET_TIME_ON_WRITE":true,"SHOW_CODE_LINK_ICON":true,"SHOW_RELOAD_ICON":true,"SHOW_WEBCAM_ICON":true,"STORE_LINE_NUMBERS":false,"Sound_Error":"","Sound_Info":"","Sound_Success":"","Sound_Warning":"","Speak_Error":false,"Speak_Info":false,"Speak_Success":false,"Speak_Warning":false,"UI_MODE":"compact","projectEntry":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.MODULE_MINIFICATION_LEVEL => 
SET chrome.storage.sync = {"AUTO_SAVE_CODE":true,"BAUD_RATE":9600,"BLOCKLY_EXTENSIONS":"|bluetooth|robot|","BLOCKLY_LANGUAGE":"en","BLOCKLY_TO_JS":false,"BOARD_JSON_URL":"http://www.espruino.com/json","CODE":"var files = require(\"fs\").readdirSync();\nfor (var i in files)\n  console.log(\"Found file \"+files[i]);","COMPILATION":true,"COMPILATION_URL":"http://www.espruino.com:32766","ENABLE_Testing":true,"ENV_ON_CONNECT":true,"FONT_SIZE":12,"FREQUENCY_Testing":1000,"MINIFICATION_DeadCode":true,"MINIFICATION_LEVEL":"","MINIFICATION_Literal":true,"MINIFICATION_Mangle":true,"MINIFICATION_Unreachable":false,"MINIFICATION_Unused":false,"MODULE_AS_FUNCTION":false,"MODULE_EXTENSIONS":".min.js|.js","MODULE_MINIFICATION_LEVEL":"","MODULE_URL":"http://www.espruino.com/modules [... cut ...] y":"CEDBBAA75AEB7325E0605429CC5E0219:Original Espruino"}
Config.ENABLE_Testing => false
loadModule(fs)
ERROR: Not Found
ERROR: Not Found
WARNING: [notify_warn] Module fs not found
Found a prompt... great!
>>> Sending...
---> "\u0010reset();\n\u0010setTime(1493912312.961);\n\u0010Modules.removeAllCached();\n\u0010var files = require(\"fs\").readdirSync();\n\u0010for (var i in files)\u001b\n  console.log(\"Found file \"+files[i]);\n\n"
Splitting at "reset();\n", delay 250
>>> Sent

Any suggestion?

gfwilliams, 2017-05-05:
Wow, sorry about this. Thanks for all the info - I just checked and you're totally right. It looks like a regression in the build caused the filesystem libraries not to be built in to 1v92.

If you do an 'advanced flash' with the latest binary from: http://www.espruino.com/binaries/travis/1759f510793dc16b557187333d4e2181f5d8a76a/

It should now be fixed though. I'll try and push a proper release soon to fix the website version.

(fixed with 367507571324dae197aacaf34436bdd23e53aef5)

closed 2017-05-05 by gfwilliams

asez73, 2017-05-05:
Thanks that worked as expected after flashing 1v92.45...