espruino/Espruino #1159: Release process needs to include ESP32 bin tgz folder

wilberforce, 2017-05-04:
issue raised here:
https://gitter.im/espruino/Espruino?at=590b7056bedc54b621ca2c71

For the ESP8266 as part of the build process, as well as the firmware.bin, an archive is built containing all of the firmware required for a first time flash.

For the esp32 this is made as part of the travis builds:
https://www.espruino.com/binaries/travis/master/espruino_1v92.43_esp32.tgz

a. This needs to be added as part of the scripting process for releases so that the archive gets added to the release zip.

b. Ideally the firmware should be called espurino_esp32.bin in this archive, so that the scripts to flash don't require a version number.
i.e built with the name espruino_esp32.bin rather than espruino_1v92.43_esp32.bin

This keeps it in line with the ESP8266 build.

gfwilliams, 2017-05-05:
the zip build is done with `scripts/create_zip.sh` so you could get whatever you need set up via that (for instance I think it unpacks the tgz file for ESP8266?).

If you want to keep the filenames the same it'll have to be in a versioned directory if you want the files to be put on the site as part of espruino.com/binaries though

wilberforce, 2017-05-08:
@gfwilliams 

Please see https://github.com/espruino/Espruino/pull/1162

closed 2017-05-13 by wilberforce