espruino/Espruino #1166: seed Math.random() during boot

drandreas, 2017-05-14:
I've noticed that all my Pucks produce the same sequence of random numbers. I think the RNG is not properly seeded during boot.
```
Connected
>
>Math.random();
=0.93255273901
>Math.random();
=0.70772355874
>Math.random();
=0.25562629208
>Math.random();
=0.68512640366
>Math.random();
=0.65661093343
>
Disconnected
Connected
>
>Math.random();
=0.93255273901
>Math.random();
=0.70772355874
>Math.random();
=0.25562629208
>Math.random();
=0.68512640366
>Math.random();
=0.65661093343
>Math.random();
=0.43274184848
```

drandreas, 2017-05-14:
`E.srand(v)` I've no idea why I haven't found it in the first place....

gfwilliams, 2017-05-22:
Thanks - sorry for the delay - I was on holiday last week.

I'll reopen this as I'd like to properly seed the RNG on boot anyway. It's what I do on other platforms.

commit a715da0c (2017-06-01): nRF52: Seed random number generator at boot (fix #1166)
closed 2017-06-01 by gfwilliams