espruino/Espruino #1168: ESP32: Wifi connects successfully only every second time

chevdor, 2017-05-22:
See related https://github.com/espressif/arduino-esp32/issues/234

I do have a very simple piece of code to connect to a Wifi network.
The Wifi network is fine and I connect to it 100% of the time successfully with various devices incl. and ESP8266.

With the ESP32 this is another story though. It fails/works always, every second (hard) restart.
If I do an `ESP32.reboot()` however, it connects 100% of the time.

I tried various workaround, none being really good.
So far the best workaround I have considers that my project needs Wifi 100% of the time and simply do a soft reset when the Wifi is seen as disconnected (which seems to happen and the connection fails):

```
wifi.on('disconnected', function() {
    console.log('Lost WIFI! Disconnected.');
    ESP32.reboot(); 
})
```

A real fix would be much better, no doubt :)

wilberforce, 2017-05-25:
Looks like need to add:

https://github.com/espressif/arduino-esp32/commit/fcbb3e7987bfc0df27b2dd24e2a5f20f66276f3b

+        } else if(reason == WIFI_REASON_AUTH_EXPIRE) {
 +            WiFi.begin();

So add the auth expire to the switch block to reconnect..

wilberforce, 2017-05-28:
@chevdor 
Can you set up your javascript to trap this:

http://www.espruino.com/Reference#l_Wifi_disconnected

`wifi.on('disconnected', function(details) { ... });`

this should be output:

jsWarn( "wifiReasonToString: Unknown reasonL %d", reason);

Looking here - would espect the reason number to be 2:

https://github.com/espressif/esp-idf/blob/d8c0b505f686db8c0f5b5de864918272c1d30836/components/esp32/include/esp_wifi_types.h#L62

In the js callback, you could reconnect rather than rebooting...

If you do this and confirm the error code, then we know if the fix would work

wilberforce, 2017-05-30:
@chevdor 

Please paste your code so we have a test case for this.

Thanks.

chevdor, 2017-05-30:
@wilberforce I am using ES6 and Thingssdk which does not make things simpler.
So I re-wrote a piece of code for the IDE (see below).
Unfortunately, I don´t reproduce the issue atm.

I have even tried running again the code from a few days ago and again, I no longer see the issue.
I notice something different though, I get a new IP at each reboot whereas I think I previously always got the same (no change in the config of my router). Getting a new IP is not really an issue in my case.

```
var WIFI_NAME = "...";
var WIFI_PASSWORD = "...";

var wifi = require('Wifi');

const connectWifi = function(networkName, options) {
    console.log(`Connecting to network ${ networkName }`);
    return new Promise(function(resolve, reject) {
        wifi.connect(networkName, options, function(error) {
            const ipAddress = wifi.getIP().ip;
            console.log(`IP: ${ipAddress}`);
            if (error) {
                reject(error);
            } else {
                wifi.stopAP();
                resolve(ipAddress);
            }
        });
    });
};

E.on('init', function() {
  connectWifi(WIFI_NAME, { password: WIFI_PASSWORD })
      .then(function(ip) {
          console.log('Yeah! ', ip);
      })
      .catch(function(err) {
          console.log('Bummer! ', err);
      });
});
```

If you have a lead, we could keep the ticket open, otherwise, since I don´t have better, I can propose to close it until the issue shows up again.

wilberforce, 2017-05-31:
@chevdor 
Ok - we'll close the issue.

Do you know about `wifi.save()` ? Once you have connected the current connection is saved in flash and connects on the next reboot - you don't need any `onInit()` code.

chevdor, 2017-06-12:
Hello @wilberforce 
The issue is still there and I can now see it again using 1.92.89

Here is the test code I use (just provide SSID and SECRET):

```
 var wifi = require('Wifi');

var SSID= "...provide your own...";
var SECRET = "...provide your own...";

function connectWifi(networkName, options) {
    console.log('Connecting to network '+networkName);
    return new Promise(function(resolve, reject) {
        wifi.connect(networkName, options, function(error) {
            const ipAddress = wifi.getIP().ip;
            console.log('IP: '+ipAddress);
            if (error) {
                reject(error);
            } else {
                wifi.stopAP();
                resolve(ipAddress);
            }
        });
    });
}

wifi.on('disconnected', function(details) {
  console.log(details);
});

E.on('init', function() {
  console.log('init, connecting to WIFI');
  digitalWrite([25,26], 0);
  connectWifi(SSID, { password: SECRET })
      .then(function(ip) {
          console.log('Yeah! ', ip);
          digitalWrite(25, 1);
      })
      .catch(function(err) {
          console.log('Bummer! ', err);
          digitalWrite(26, 1);
      });
});

```

As you see, I have included an event handler in case of a disconnect.

Here is what happen:
 - when calling `ESP32.reboot()` the subsequent connections are ALWAYS fine
 - when clicking the RST button / Unplug-resplug => the connection works every second time

In the case when the connection does not succeed, there is no disconnect event so nothing additional I can report, just nothing happen.

Can you please re-open this issue?

gfwilliams, 2017-07-18:
Closing https://github.com/espruino/Espruino/issues/1175#issuecomment-316122535

wilberforce, 2017-08-18:
The esp-idf has been updated to 2.1 - there was something in here about joining access points. Please re-check now

chevdor, 2017-08-25:
I tested today with the cuttingEdge 2017-08-24 and I consider the issue fixed now.
Thanks for the heads up @wilberforce 

I tested with > x20 resets and got the ESP32 connected properly 19 times.
I put the credit of the 1 failure to my router, probably annoyed to get so man subsequent requests so fast.

closed 2017-08-25 by chevdor