espruino/Espruino #1169: long history causes, out of memory

drandreas, 2017-05-22:
The history payload is garbage collected but the history vector is not shrunk.

After doing extensive WebBLE communication `global["\xFF"].history.length` shows several hundred entries. Looking at the history, all except the last few point to `undefined` .i.e.:
`global["\xFF"].history = [ undefined, undefined,  undefined, ... , "process.memory()" ]`

Continuing WebBLE communication after this state crashes the puck.
running `global["\xFF"].history = []` now and the immediates this issue.

gfwilliams, 2017-05-22:
That shouldn't be a problem, since arrays are 'sparse' in JavaScript - so you can have a million element array full of undefined that only uses up the same amount of space as a zero element array.

This method of handling history has been around for years, for all kinds of different devices - if it was a problem I think it would have been noticed by now.

Are you using up to date firmware? Were you executing some specific code when it crashed? It could be related to that.

drandreas, 2017-05-22:
Nevermind I musst have crashed the Puck by some other means.

I've tried to reproduce it by putting 1000+ items in the history but it didn't ran out of memory now.

closed 2017-05-22 by drandreas