espruino/Espruino #1172: ESP32: Add support for setSNTP

chevdor, 2017-05-26:
While wifi.setSNTP(...) is implemented for ESP8266, this does not seem to be the case yet for the ESP32.

wilberforce, 2017-06-02:
Here is a work around.

http requests contain the current time, so this can be used to set the Espruino clock:

```
  var http = require("http");
  http.get("http://www.espruino.com", function(res) {
    setTime(Date.parse(res.headers.Date)/1000);
    d= new Date();
    console.log(d,d.toString());
    res.on('data', function(data) {
    });

  });
```

chevdor, 2017-06-09:
AFAIK, this does not set the system date so you get the date NOW but cannot use something similar to getDate() at any time later.

wilberforce, 2017-06-10:
Huh? If you reboot and put this in the left side of the IDE:

```
  d= new Date();
=Date { "ms": 5724.567 }
>    console.log(d,d.toString());
Date { "ms": 5724.567 } Thu Jan 1 1970 00:00:05 GMT+0000
```

Then:
```
var r;
  var http = require("http");
  http.get("http://www.espruino.com", function(res) {
      console.log(res);
    r=res;
    setTime(Date.parse(r.headers.Date)/1000);
    d= new Date();
    console.log(d,d.toString());
    res.on('data', function(data) {
    });

  });
```

and now the time is:

```
Date { "ms": 1497138924000.80395507812 } Sat Jun 10 2017 23:55:24 GMT+0000
>
```

So the date and time is set.

gfwilliams, 2017-07-18:
Closing https://github.com/espruino/Espruino/issues/1175#issuecomment-316122535

closed 2017-07-18 by gfwilliams

gfwilliams, 2017-07-19:
I think this could be addressed with https://github.com/espruino/Espruino/issues/624 - no need for special per-platform code, we could do this with a JS module that worked on all platforms.