espruino/Espruino #1176: ESP32: setWatch results in a crash

chevdor, 2017-05-31:
I am using the firmware 1.92.73 in the ESP32.

I have tested the following code with an ESP8266:
```
var pwmIn = 15;
[dump.txt](https://github.com/espruino/Espruino/files/1041521/dump.txt)

var pwmOut = 14;
var speedi = 0,
    speedo = 0;

function startPinWatch() {
    digitalWrite(pwmOut, 0);
    speedi = 0;
    throttle = true;

    var pinWatchId = setWatch(function(interrupt) {
      speedi = (interrupt.time - interrupt.lastTime) * 1000;
    }, pwmIn, { repeat: 1, edge: 'both' });

    return pinWatchId;
}

function monitor() {
    console.log(speedi, speedo, process.memory());
}

setInterval(monitor, 1000);
startPinWatch();
```
In the ESP8266, it works all fine.

Similar code (only the GPIO pins are different):

```
var pwmIn = 26;
var pwmOut = 27;
var speedi = 0,
    speedo = 0;

function startPinWatch() {
    digitalWrite(pwmOut, 0);
    speedi = 0;
    throttle = true;

    var pinWatchId = setWatch(function(interrupt) {
      speedi = (interrupt.time - interrupt.lastTime) * 1000;
    }, pwmIn, { repeat: 1, edge: 'both' });

    return pinWatchId;
}

function monitor() {
    console.log(speedi, speedo, process.memory());
}

setInterval(monitor, 1000);
startPinWatch();
```

In the ESP32, as long as no signal comes to the watched pin, everything is OK.
However, as soon as a PWM signal is presented to the GPIO called `pwmIn` above, the ESP32 crashes with the following trace (See attached file).

wilberforce, 2017-06-01:
fails in this simple case:
```
function button_down() {
  print('button');
}

setWatch(button_down,D0, {repeat:true, edge:"rising"});
```

Pressing the `flash` button about 20 times causes a core dump

wilberforce, 2017-06-01:
make clean && RELEASE=1 BOARD=ESP32 make
export IDF_PATH=$ESP_IDF_PATH
#flash firmware
cat > coredump.dat

esp-idf/components/espcoredump/espcoredump.py dbg_corefile -t b64 -c coredump.dat

wilberforce, 2017-06-01:
Can't debug, getting:

esp-idf/components/espcoredump/espcoredump.py info_corefile -t b64 -c coredump.dat espruino_1v92.54_esp32.elf
espcoredump.py v0.1-dev

A fatal error occurred: Can not add overlapping region [3ffcea2c..3ffceb9f] to ELF file. Conflict with existing [3ffb0470..3ffd55b3].

chevdor, 2017-06-12:
Still seeing the issue in 1.92.89

gfwilliams, 2017-07-18:
Closing https://github.com/espruino/Espruino/issues/1175#issuecomment-316122535

wilberforce, 2017-08-15:
with esp-idf 2.1 and v1.93 get the following core dump after on 4 button presses:

` ../esp-idf/components/espcoredump/espcoredump.py  info_corefile -t b64 -c coredump.dat  `

```
espruino_1v93.168_esp32.elf
espcoredump.py v0.1-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x40127d18       0x40127d18 <esp_vApplicationIdleHook+40>
lbeg           0x0      0
lend           0x0      0
lcount         0x0      0
sar            0x0      0
ps             0x60820  395296
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40085c38       1074289720
a1             0x3ffcd340       1073533760
a2             0x8      8
a3             0x1      1
a4             0x60023  393251
a5             0x3ffafd18       1073413400
a6             0x1      1
a7             0x0      0
a8             0x3ffb33d8       1073427416
a9             0x3ffcd320       1073533728
a10            0x0      0
a11            0x60820  395296
a12            0x60021  393249
a13            0x3ffafab0       1073412784
a14            0x3ffb0824       1073416228
a15            0x60720  395040

==================== CURRENT THREAD STACK =====================
#0  0x40127d18 in esp_vApplicationIdleHook () at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/esp32/./freertos_hooks.c:52
#1  0x40085c38 in prvIdleTask (pvParameters=0x0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./tasks.c:3322

======================== THREADS INFO =========================
  Id   Target Id         Frame
  9    process 8         xQueueGenericReceive (xQueue=0x3ffae5bc, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  8    process 7         xQueueGenericReceive (xQueue=0x3ffd5150, pvBuffer=0x3ffd6750, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  7    process 6         xQueueGenericReceive (xQueue=0x3ffdbdc4, pvBuffer=0x3ffdcbd0, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  6    process 5         xQueueGenericReceive (xQueue=0x3ffd40a0, pvBuffer=0x3ffd4e40, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  5    process 4         xQueueGenericReceive (xQueue=0x3ffdb0b8, pvBuffer=0x3ffdbb20, xTicksToWait=4294967295, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  4    process 3         prvTimerTask (pvParameters=0x0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./timers.c:445
  3    process 2         xQueueGenericReceive (xQueue=0x3ffdee90, pvBuffer=0x0, xTicksToWait=100, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
  2    process 1         xQueueGenericReceive (xQueue=0x3ffce3b0, pvBuffer=0x3ffcee40, xTicksToWait=100, xJustPeeking=0) at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/freertos/./queue.c:1452
* 1    <main task>       0x40127d18 in esp_vApplicationIdleHook () at /mnt/c/Users/rhys/esp32/edge/EspruinoBuildTools/esp32/build/esp-idf/components/esp32/./freertos_hooks.c:52

======================= ALL MEMORY REGIONS ========================
Name   Address   Size   Attrs
.rtc.text 0x400c0000 0x0 RW
.iram0.vectors 0x40080000 0x400 R XA
.iram0.text 0x40080400 0x14014 R XA
.dram0.data 0x3ffb0000 0x2614 RW A
.flash.rodata 0x3f400010 0x1e83c RW A
.flash.text 0x400d0018 0x95308 R XA
.coredump.tasks 0x3ffafd18 0x174 RW
.coredump.tasks 0x3ffcd280 0x164 RW
.coredump.tasks 0x3ffceee0 0x174 RW
.coredump.tasks 0x3ffcecf0 0x1e4 RW
.coredump.tasks 0x3ffde694 0x174 RW
.coredump.tasks 0x3ffde440 0x248 RW
.coredump.tasks 0x3ffcdbf8 0x174 RW
.coredump.tasks 0x3ffcda90 0x15c RW
.coredump.tasks 0x3ffdbbc8 0x174 RW
.coredump.tasks 0x3ffdba20 0x19c RW
.coredump.tasks 0x3ffd4efc 0x174 RW
.coredump.tasks 0x3ffd4d40 0x1b0 RW
.coredump.tasks 0x3ffdcc70 0x174 RW
.coredump.tasks 0x3ffdcad0 0x194 RW
.coredump.tasks 0x3ffd67ec 0x174 RW
.coredump.tasks 0x3ffd6640 0x1a0 RW
.coredump.tasks 0x3ffaea18 0x174 RW
.coredump.tasks 0x3ffae880 0x18c RW

===================== ESP32 CORE DUMP END =====================
```

wilberforce, 2017-12-14:
Fixed with https://github.com/espruino/Espruino/pull/1285

```
var count=0;
function button_down() {
  print('button',count++);
}

setWatch(button_down,D0, {repeat:true, edge:"rising"});
```

Working now:

```
....
button 186
button 187
button 188
button 189
button 190
button 191
button 192
````

closed 2017-12-14 by wilberforce