espruino/Espruino #1177: Unstable console

opichals, 2017-06-01:
I have gotten the [DOIT ESP32 board]( https://www.aliexpress.com/item/ESP-32S-Development-Board-Esp32-Module-2-4-GHz-dual-mode-Wi-Fi-and-Bluetooth-chips/32803195774.html) and have issues with basic espruino stuff like the following keyboard presses get corrupted on the way it seems (e.g. see the mangled `process` words below).

Any ideas?

```
$ npm run espruino

> d1-dht11@1.0.0 espruino /Users/standa/Dev/private/d1-dht11
> espruino -b 115200

Espruino Command-line Tool 0.0.30
-----------------------------------

Using first port, {"path":"/dev/cu.SLAB_USBtoUART","usb":[4292,60000],"description":"Silicon Labs"}
Connecting to '/dev/cu.SLAB_USBtoUART'
Connected
prûcÙss.memory()
Uncaught SyntaxError: Got ?[251] expected EOF
 at line 1 col 3
prûcÙss.memory()
  ^
>process.memory()
={ "free": 4963, "usage": 37, "total": 5000, "history": 10 }
>procesî.ÛemorÞ()
Uncaught SyntaxError: Got ?[238] expected EOF
 at line 1 col 7
procesî.ÛemorÞ()
      ^
>process.memory()
={ "free": 4963, "usage": 37, "total": 5000, "history": 16 }
>process.memory()
={ "free": 4963, "usage": 37, "total": 5000, "history": 16 }
>process.env
={ 
  "VERSION": "1v92.73", 
  "BUILD_DATE": "May 30 2017", 
  "BUILD_TIME": "08:14:01", 
  "GIT_COMMIT": "3206eecda556ce9c96a781e2811b2043f2831eae", 
  "BOARD": "ESP32", 
  "CHIP": "ESP32", 
  "CHIP_FAMILY": "ESP32", 
  "FLASH": 0, "RAM": 524288, 
  "SERIAL": "30aea400-fe90", 
  "CONSOLE": "Serial1", 
  "EXPORTS": { "jsvLock": 1074775772, "jsvLockAgainSafe": 1075142760, "jsvUnLock": 1074777712, "jsvSkipName": 1074792772, 
    "jsvMathsOp": 1074802380, "jsvMathsOpSkipNames": 1074805880, "jsvNewFromFloat": 1074787948, "jsvNewFromInteger": 1074787860, "jsvNewFromString": 1074786876, 
    "jsvNewFromBool": 1074787920, "jsvGetFloat": 1074796724, "jsvGetInteger": 1074791156, "jsvGetBool": 1074797472, "jspeiFindInScopes": 1074815288, 
    "jspReplaceWith": 1074815056, "jspeFunctionCall": 1074824480, "jspGetNamedVariable": 1074816920, "jspGetNamedField": 1074817304, "jspGetVarNamedField": 1074817412, 
    "jsvNewWithFlags": 1074786604 }
 }
```

opichals, 2017-06-01:
Could also be relate to other stuff I observe like https://github.com/espressif/arduino-esp32/issues/409

wilberforce, 2017-06-01:
First I have heard if anything like this. What power supply is used?

If characters are getting mangled, I'm surprised the flashing works.

wilberforce, 2017-06-01:
Dos it work from the web IDE?

opichals, 2017-06-01:
I have it connected to a MacBook Pro USB port only. Will try the web IDE and write more later. Thanks!

opichals, 2017-06-05:
Same thing in the IDE.

opichals, 2017-06-08:
No such issue with [WeMos LoLin32 board](https://www.aliexpress.com/item/WEMOS-LOLIN32-V1-0-0-wifi-bluetooth-board-based-ESP-32-4MB-FLASH/32808551116.html) observed

wilberforce, 2017-06-08:
I think it is related to USB to serial chip used, and the driver for that chipset and the OS.

All of the boards seem to use a different chipset, so unless we document the board and the chip and OS, it will be hard to track. At least some patterns could be observed.

wilberforce, 2017-08-15:
@opichals 
Want to leave this open? Do you think it was the board?

opichals, 2017-08-15:
I tested [MH-ET LIVE MiniKit for ESP32](https://www.aliexpress.com/item/MH-ET-LIVE-D1-mini-ESP32-ESP-32-WiFi-Bluetooth-Internet-of-Things-development-board-based/32815530502.html?spm=a2g0s.9042311.0.0.3jlEVz) and I didn't observe anything like this there.

I think it was a faulty board.

closed 2017-08-15 by opichals