espruino/Espruino #1178: Pre-tokenisation

gfwilliams, 2017-06-08:
While Espruino executes from source, nothing stops us from 'compressing' that source. Espruino currently only copes with character codes below 128 in sourcecode and nobody's complained so far.

Tokens such as `function` could then be stored internally as a single character >=128.

Not to mention whitespace of the form `"\n    "` which is all over the place. It'd save memory while also speeding up execution.

gfwilliams, 2017-06-08:
If anyone's reading this, it's be great if you could let me know if there are any cases in valid JS where you'd get:

* `Identifier Number` eg. `hello 1234`
* `Number Identifier`
* `Identifier Identifier`
* `Number Number`

I'm wondering if I actually need to bother inserting whitespace between tokens at all. If not I could have this as a kind of super minification pass - it seems to at least halve the amount of characters in most code.

opichals, 2017-06-08:
Ecma identifiers https://www.ecma-international.org/ecma-262/5.1/#sec-7.6 do not contain spaces nor start with a number. But Object keys can.

gfwilliams, 2017-06-08:
The issue is really about stripping whitespace... If you've tokenised a string, can you just push everything back together without any spaces?

For instance:

```
for (var i=0;i<10;i++) print("Hello")
// goes to
[for]([var]i=0;i<10;i[++])print("Hello")

// but if it could were valid js to have one Identifier after another
hello world
// would go to
helloworld
```

which is obviously not the same thing

wilberforce, 2017-06-08:
> Tokens such as function could then be stored internally as a single character >=128.

I had a bit of a chuckle when I read this. Back in the day, Microsoft basic on the Z80 tokenised the inbuild basic functions this way. It's funny how things go full circle!

opichals, 2017-06-09:
I would not be sure this kind of optimization is worth the effort. Rather I would propose going the standard way. The ES6 arrow functions and common minification libs should give us good enough results (once we fix the minification - https://github.com/espruino/EspruinoTools/issues/64).

gfwilliams, 2017-06-09:
Yeah, I guess the uCs have similar amounts of RAM - I'd heard about the tokenisation before, but I could probably do worse than looking through some old interpreters and figuring out what else I could 'borrow' :)

@opichals I wouldn't spend too much time on it, but the weird thing is that it 'just works'. I have a branch that implements it in less than 50 lines, changing only `jslex.c`. It's not just the `function` keyword that can be compressed either.

IMO the people who specced ES6 weren't really thinking about the mechanics of the parser at all, so arrow functions are horrible and slow to parse unless they're going into a parse tree first. If you think about it, you can figure out what most JS code does almost immediately by reading through it character by character. Arrow functions are nasty because the code `(a,b,c,d)` is totally valid all by itself, but add a `=>` after it and now it's something totally different :(

gfwilliams, 2017-06-09:
Ok, check out this branch: https://github.com/espruino/Espruino/tree/pre_tokenized

I just uploaded some simple stuff - the Pico 'Hello world LCD' example, plus `EspruinoWiFi` and `ws` modules - before: 949 blocks used, after: 847 blocks used.

Bear in mind that the libraries are pre-minified already, so that's a big win. It's not just `function`, but things like `this` that get squished down to a single character.

The only gotcha is that while this should improve execution speed, it'll be slower to define functions (including functions inside functions) because it involves running the lexer again - but maybe that could be skipped if we're executing from a function that we know has been squished already.

opichals, 2017-06-09:
Nice savings! :) 

wrt to ES6, I guess the spec people had a lot of other stuff to figure out and streamlining parsing effort wasn't really one of them. :)

It is just that from my perspective it is a non-standard thing from a JavaScript guy's point of view which needs to be dealt with during debugging. Therefore I would argue to make this kind of optimization optional (by default off). So that the dev workflow is: Have everything running, now let's try to make it smaller by enabling this kind of '-Os'-like stuff during preprocessing. Or I'am getting short of memory so I try size optimization. But in either case the decisions is conscious.

gfwilliams, 2017-06-09:
Yes, I totally agree. It should be switchable - but actually doing that isn't hard to do at all, since it's just a matter of calling the new function or the old one when parsing.

I guess the only question then is how to do that - there have been a few things recently where I guess having a set of system-wide flags would be handy - stuff like whether to block on USB transmission or chuck characters.

opichals, 2017-06-09:
System wide flags: The ESP's wifi.save() method uses a full flash page already IIRC. What about generalizing this to a simple 'system properties' key-value storage?

wilberforce, 2017-06-10:
system properties persisted would be good, it means setting like telnet on/off could be stored, perhaps the Dataview classes could be used to keep the data in binary.

opichals, 2017-06-11:
@wilberforce Sounds like a good idea. That would save bytes. But also generic key-value would be nice for application settings and similar.

gfwilliams, 2017-06-12:
I'd been vaguely considering building in something like `FlashEEPROM` - we already have a hacked up solution for storing raw JS code and/or a compressed RAM image - so it'd be nice to extend that to allow other data to be stored as well.

opichals, 2017-06-12:
@gfwilliams Reusing the JS code storing sounds like a nobrainer.

MaBecker, 2017-06-12:
@opichals yes something like a `SimpleFlashStore` as a mix of `FlashEEPROM` and `Flash storage Modul` from (http://forum.espruino.com/conversations/283045/) including a 4byte fix to work an ESP as well.

gfwilliams, 2017-07-07:
Now merged with master, default is off. Added `E.set/getFlags` - it saves to a JsVar, which seems to work out fine.

To turn it on just type `E.setFlags({pretokenise:1})` and then upload your code. It persists after a `reset()` so it should apply even when modules are loaded in. Should help a lot on systems without much memory.

commit 5ba47c02 (2017-07-07): Add  to dynamically turn on pretokenisation (fix #1178)
closed 2017-07-07 by gfwilliams