espruino/Espruino #1183: Suggestion: Add STATIC_ASSERT for enum types

nine-fox, 2017-06-15:
When moving source code into MDK compiler, I just found the following code acts differently under MDK. The size of this enum is 4 bytes instead of 2 bytes

typedef enum {
  JSWAT_FINISH = 0, // no argument
  JSWAT_VOID = 0, // Only for return values
  JSWAT_JSVAR, // standard variable
  JSWAT_ARGUMENT_ARRAY, // a JsVar array containing all subsequent arguments
  JSWAT_BOOL, // boolean
  JSWAT_INT32, // 32 bit int
  JSWAT_PIN, // A pin
  JSWAT_JSVARFLOAT, // 64 bit float
  JSWAT__LAST = JSWAT_JSVARFLOAT,
  JSWAT_MASK = NEXT_POWER_2(JSWAT__LAST)-1,

  // should this just be executed right away and the value returned? Used to encode constants in the symbol table
  // We encode this by setting all bits in the last argument, but leaving the second-last argument as zero
  JSWAT_EXECUTE_IMMEDIATELY = 0x7000,
  JSWAT_EXECUTE_IMMEDIATELY_MASK = 0x7E00,

  JSWAT_THIS_ARG    = 0x8000, // whether a 'this' argument should be tacked onto the start
  JSWAT_ARGUMENTS_MASK = ~(JSWAT_MASK | JSWAT_THIS_ARG)
} JsnArgumentType;

After investigation, I found the problem is this code: ~(JSWAT_MASK | JSWAT_THIS_ARG), its size is 4 bytes.

Since it would be hard to find the root cause, I would suggest to add the STATIC_ASSERT like this for all enum types:
STATIC_ASSERT( sizeof(JsnArgumentType) == 2);

The STATIC_ASSERT is included in NRF SDK.

Thanks,
Vincent

nine-fox, 2017-06-15:
BTW, for work around, probably the fix is like this:
JSWAT_ARGUMENTS_MASK = (uint16_t)(~(JSWAT_MASK | JSWAT_THIS_ARG))

Not tested yet :)

gfwilliams, 2017-06-15:
Thanks for reporting that!

Yes, that sounds like a good plan (however we couldn't use STATIC_ASSERT directly since Espruino targets a bunch of other platforms that don't have it). For now I'll add a tweak for the enum, and will add something to the jsnative sanity test (which is added to debug builds)

commit 8ff54459 (2017-06-15): fix compile issue onMDK compiler and add warning, fix #1183
closed 2017-06-15 by gfwilliams