espruino/Espruino #1194: Reset when uploading code

nelsonec87, 2017-06-29:
Hi, first of all, great project!

I'm trying to organize my code writing TypeScript, transpiling to JS and then uploading to Espruino (running on NodeMCU/Esp8266). To try the setup, I created:

```
class A{
    public a = 'a';
}

class B extends A{
    public b = 'b';
}

function onInit(){
   let myB = new B;
   console.log("Here is my B", b.a, b.b);
}
```

which was turned into:

```
var __extends = (this && this.__extends) || (function () {
    var extendStatics = Object.setPrototypeOf ||
        ({ __proto__: [] } instanceof Array && function (d, b) { d.__proto__ = b; }) ||
        function (d, b) { for (var p in b) if (b.hasOwnProperty(p)) d[p] = b[p]; };
    return function (d, b) {
        extendStatics(d, b);
        function __() { this.constructor = d; }
        d.prototype = b === null ? Object.create(b) : (__.prototype = b.prototype, new __());
    };
})();
var A = (function () {
    function A() {
        this.a = 'a';
    }
    return A;
}());
var B = (function (_super) {
    __extends(B, _super);
    function B() {
        var _this = _super !== null && _super.apply(this, arguments) || this;
        _this.b = 'b';
        return _this;
    }
    return B;
}(A));
function onInit() {
    var myB = new B;
    console.log("Here is my B", b.a, b.b);
}
```
This code runs just fine in node and in the browser, but when I try uploading...

```
:
:    return B;
:
:}(A));

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user1 @ 1000
```

If I take the "extends" off the picture it works, so I believe it has something to do with the way it handles prototypes, setPrototypeOf, hasOwnProperty, or something like that.

I tried with 1v91 e 1v92 of Espruino, same results.

Ideas?

closed 2017-06-30 by gfwilliams

gfwilliams, 2017-06-30:
The actual issue here is:

```
var B = (function (_super) {
    __extends(B, _super);
    function B() {
        var _this = _super !== null && _super.apply(this, arguments) || this;
        _this.b = 'b';
        return _this;
    }
    return B;
}(A));
``` 

Because normal JS interpreters pre-parse everything beforehand, you get 'function hoisting'. In Espruino you don't (it could do it, but it'd make functions take a lot longer to execute so it's been left out).

The transpiler you're using seems to rely on it, because it calls `extend` on `B` before the function is even defined. IMO it's bad practice from a readability point of view, and something that'd be trivial to change in the transpiler.

So that's why it isn't working, however you inadvertently stumbled on another issue which is that if you call setPrototypeOf with `undefined` it crashes (it was assuming it got called on an object) - so I've fixed that. Now, you'll get an exception rather than a crash - but it still won't work I'm afraid,

gfwilliams, 2017-06-30:
Just to add that it would be pretty trivial to write a script with `acorn` that parsed the JS file, detected this particular pattern (`__extends` the `function`) and flipped them.

commit 91d26d34 (2017-06-30): Fix Object.setPrototypeOf when called on undefined (fix #1194)