espruino/Espruino #1199: Can't get data from UART (Serial)

Sparow199, 2017-07-19:
Hello!
I tried to get data from UART, i had the data displayed on console but the function in the handler doesn't work, so i can't get the to manipulate it.
**Code:**
Serial1.setup(115200);
Serial1.on('data', function (data) { print("<Serial1> "+data); }); // doesn't work 
Serial1.print("Hello World");

**Console:**
`{"MQ131":["O3"],"MQ135":["NH3","CO2"],"MULTI":["NH3","CO","NO2","C3H8","C4H10","CH4","H2","C2H5OH"]}`

**Expected:**
`<Serial1>{"MQ131":["O3"],"MQ135":["NH3","CO2"],"MULTI":["NH3","CO","NO2","C3H8","C4H10","CH4","H2","C2H5OH"]}`

I'm using ESP8266 NodeMCU and i tried all of official [examples](https://www.espruino.com/USART).

gfwilliams, 2017-07-19:
This is standard Espruino stuff I think - check out what it says under https://www.espruino.com/USART#console-over-serial

Try:

```
var LED = whatever_a_led_is_connected_to;
Serial1.setup(115200);
Serial1.on('data', function (data) { LED.toggle(); });
Serial1.print("Hello World");
LoopbackA.setConsole();
// you won't be able to control/reprogram the ESP8266 now until you reset - but your code will work
```

`LoopbackA.setConsole()` is the important thing, but what you're trying to do won't work. There is only one serial port on ESP8266 - you can use it for the console and can print stuff to it/upload code, but if you want to use it to read/write data, you have nothing available for the console so `print` won't work.

Sparow199, 2017-07-19:
Thank you for reply,
I'm trying to connect ESP8266 (Espruino.js) to Arduino Uno , i'm testing a serial communication but i know that's it's not a good idea, SPI may be a best way but i'm not sure that the implementation of SPI on Espruino.js is compatible with Arduino ?

gfwilliams, 2017-07-20:
Serial will work well enough as long as you don't need the USB as well. I guess the code above fixed your problem - assuming you disconnected the wire used for USB TX?

If the Arduino will do SPI slave (and is a 3.3v part rather than 5v) then yes, everything will work fine with SPI. Espruino won't do SPI slave though - only master.

If you want a proper microcontroller with peripherals, and Wifi, please check out http://www.espruino.com/WiFi - it'll be way easier to use, is more powerful, and you won't have these problems!

closed 2017-07-20 by gfwilliams